OpenIntelWireless/IntelBluetoothFirmware #1: 驱动加载后未发现设备，应该怎么处理？

laozhiang, 2019-12-05:
你好：
我的机器蓝牙是intel AC9560的，蓝牙设备信息如下  
_Apple 蓝牙软件版本：	6.0.14d3
  供应商 ID：	0x8087
  产品 ID：	0x0AAA_

我修改了代码的info.plist适配我的vendorID和productID
		_<key>IntelBluetoothFirmware_common</key>
		<dict>
			<key>CFBundleIdentifier</key>
			<string>com.zxystd.IntelBluetoothFirmware</string>
			<key>IOClass</key>
			<string>IntelBluetoothFirmware</string>
			<key>IOProviderClass</key>
			<string>IOUSBHostDevice</string>
			<key>idProduct</key>
			<integer>2730</integer>
			<key>idVendor</key>
			<integer>32903</integer>
		</dict>_
我编译好之后，通过kextload加载，内核日志里出现如下信息
_kernel: Notice - new kext com.apple.driver.usb.AppleUSBHostPlatformProperties, v1.2 matches prelinked kext but can't determine if executables are the same (no UUIDs)._
请问应该如何处理？

另：该驱动我可以进行测试。如果需要测试请联系QQ2747055866。

zxystd, 2019-12-05:
1. 未发现设备是因为设备被其他驱动占用了，sudo kextunload -b com.apple.iokit.IOBluetoothHostControllerUSBTransport 之后就可以加载本kext。
2. 本项目目前还未完成，目前在发送固件到设备的过程还有些问题没有验证。
3. 本项目分两个阶段，第一阶段是适配初期版本设备
0x8087, 0x0a2a	3165
0x8087, 0x07dc	7260
0x8087, 0x0aa7	3168
第二阶段为新设备
0x8087, 0x0025	9260ac
0x8087, 0x0aaa	9560ac
0x8087, 0x0026	??
0x8087, 0x0029       ??
0x8087, 0x0a2b	8265/8275，
目前我手里只有3165，所以先适配第一阶段。
4.期待这个项目能完成吧~

laozhiang, 2019-12-05:
谢谢指导，已经启动加载，并看到日志信息。
目前很多人使用虚拟机驱动加载固件来驱动蓝牙，相信通过这个项目能造福大家。
我会持续关注该项目。感谢您的付出。

closed 2019-12-05 by laozhiang

ISPP2056, 2019-12-17:
加油加油，一群人在关注着

zxystd, 2019-12-17:
谢谢，这几天淘个8260开始尝试新版网卡的固件上传

zxystd, 2019-12-17:
> 加油加油，一群人在关注着

谢谢关注，年底了比较忙。这几天淘个8260开始尝试新版网卡的固件上传

laozhiang, 2019-12-28:
请问：AC9560固件上传成功了么

zxystd, 2019-12-28:
> 请问：AC9560固件上传成功了么

应该快了，在进行ax200的上传，还有点问题，这个成功了，9560这一类应该也可以了

laozhiang, 2019-12-28:
谢谢

zxystd, 2019-12-31:
> 谢谢

ax200已经可以上传固件了，你可以试试换成你的pid跟vid，然后放到clover或者oc目录进去，看看能不能驱动。
目前存在关机无法断电的问题，待修复

zxystd, 2020-01-01:
关机不断电的问题已修复

yigendabaicai, 2020-11-11:
[image]
我的把驱动放到oc的kext文件夹之后，还是没法用，直接没有蓝牙地址了…请问大神知道这个是什么原因不呀？