OpenIntelWireless/IntelBluetoothFirmware #2: 怎么使用这个驱动

zhoubin0422, 2019-12-30:
这个驱动要怎么使用呢？我将编译好的 kext 放入 OC 下的 kernel 目录，并配置了 config.split ，但是重启系统后，蓝牙固件依然没有加载；（这边下载了这个驱动后，虽然 xcode 显示编译成功了，但是报了很多的 warning，这个有影响没？）
我使用虚拟机加载蓝牙时提示的版本是 ibt-17-16-1.ddc；系统信息中显示的版本为：9.256，

zxystd, 2019-12-30:
> 这个驱动要怎么使用呢？我将编译好的 kext 放入 OC 下的 kernel 目录，并配置了 config.split ，但是重启系统后，蓝牙固件依然没有加载；（这边下载了这个驱动后，虽然 xcode 显示编译成功了，但是报了很多的 warning，这个有影响没？）
> 我使用虚拟机加载蓝牙时提示的版本是 ibt-17-16-1.ddc；系统信息中显示的版本为：9.256，

1.是编译的昨天晚上的提交版本吗？系统是哪个版本？可以提供下log吗， log show --last 1d | grep "IntelFir"
2.编译提示warning没关系
3.这个驱动快完成了，还剩下一点小问题这几天会修复，目前我手里也没有9系的卡

zhoubin0422, 2019-12-30:
对的，是最新提交的版本编译的，系统是 Catalina 10.15.2， 日志内容我用你这个命令找不到任何输出；

zxystd, 2019-12-30:
> 对的，是最新提交的版本编译的，系统是 Catalina 10.15.2， 日志内容我用你这个命令找不到任何输出；

pid跟vid配置了吗？是多少？等我把这个完成以后会编译发布出来的

zhoubin0422, 2019-12-30:
没有配置哦，pid 和 vid 配置在编译好的驱动中 info.plist 文件中？我在系统信息中看到 pid 和 vid 是0x8087 以及 0x0AAA

zhoubin0422, 2019-12-30:
HI, 配置完 pid 和 vid 后蓝牙挂了，日志显示如下：

2019-12-30 12:46:32.008614+0800 0x155      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: recv event=0x000e, length=4
2019-12-30 12:46:32.008638+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: boot_param=0x40800
2019-12-30 12:46:32.008811+0800 0x155      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: recv event=0x000e, length=4
2019-12-30 12:46:32.008827+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware done
2019-12-30 12:46:32.008829+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2019-12-30 12:46:32.008832+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_RESET
2019-12-30 12:46:32.008835+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: opCode=0xfc01, paramLen=8
2019-12-30 12:46:32.008838+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: check=00
2019-12-30 12:46:32.009000+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Intel reset succeed
2019-12-30 12:46:33.009043+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2019-12-30 12:46:33.009046+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2019-12-30 12:46:33.009048+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: opCode=0xfc52, paramLen=8
2019-12-30 12:46:33.009051+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: check=87
2019-12-30 12:46:33.009202+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2019-12-30 12:46:33.009720+0800 0x155      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: recv event=0x00ff, length=5
2019-12-30 12:46:33.009734+0800 0x155      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2019-12-30 12:46:33.009738+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2019-12-30 12:46:33.009740+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2019-12-30 12:46:33.009742+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: opCode=0xfc52, paramLen=8
2019-12-30 12:46:33.009744+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: check=87
2019-12-30 12:46:33.009887+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2019-12-30 12:46:33.010715+0800 0x155      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: recv event=0x00ff, length=7
2019-12-30 12:46:33.010725+0800 0x155      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2019-12-30 12:46:33.010729+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2019-12-30 12:46:33.010731+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2019-12-30 12:46:33.010733+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: opCode=0xfc52, paramLen=8
2019-12-30 12:46:33.010734+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: check=87
2019-12-30 12:46:33.010877+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2019-12-30 12:46:33.011707+0800 0x155      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: recv event=0x000e, length=4
2019-12-30 12:46:33.011716+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2019-12-30 12:46:33.011718+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2019-12-30 12:46:33.011740+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...
2019-12-30 12:46:33.011800+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
2019-12-30 12:46:33.011802+0800 0x183      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...

zxystd, 2019-12-30:
> HI, 配置完 pid 和 vid 后蓝牙挂了，日志显示如下：
>
> 2019-12-30 12:46:32.008614+0800 0x155 Default 0x0 0 0 kernel: (IntelBluetoothFirmware) IntelFirmware: recv event=0x000e, length=4
> 2019-12-30 12:46:32.008638+0800 [...]

从日志来看，固件已经成功写入，蓝牙挂是因为这就是没有处理完的问题之一，待解决以后就可以正常加载。另外请试试加载以后关机是否正常

zhoubin0422, 2019-12-30:
加载驱动后，关机不正常，点关机后会自动重启系统，并且出现禁止标识符，然后才会再次重启进入系统

zxystd, 2019-12-30:
> 加载驱动后，关机不正常，点关机后会自动重启系统，并且出现禁止标识符，然后才会再次重启进入系统

这两个问题为待解决，谢谢测试

zhoubin0422, 2019-12-30:
别的帮不上忙，只能帮忙测试了

zxystd, 2019-12-31:
> 别的帮不上忙，只能帮忙测试了

蓝牙挂掉的问题昨晚已经修复了，关机问题还待修复

zhoubin0422, 2019-12-31:
OK，我测试一下

zhoubin0422, 2019-12-31:
> 蓝牙挂掉的问题昨晚已经修复了，关机问题还待修复

已经测试了，如你说说，蓝牙确实可以冷启动加载了，只剩下关机问题了，厉害！

zxystd, 2019-12-31:
> > 蓝牙挂掉的问题昨晚已经修复了，关机问题还待修复
> 
> 已经测试了，如你说说，蓝牙确实可以冷启动加载了，只剩下关机问题了，厉害！

好的，另外你的蓝牙信息里传输方式是显示的UART还是USB？加载brcmbluetoothcontroller能不能显示蓝牙开关？我手里没有9系的网卡，帮忙截个图就更好了。

zhoubin0422, 2019-12-31:
> 好的，另外你的蓝牙信息里传输方式是显示的UART还是USB？加载brcmbluetoothcontroller能不能显示蓝牙开关？我手里没有9系的网卡，帮忙截个图就更好了。

蓝牙信息中传输方式显示的是 USB，加载brcmbluetoothcontroller 显示的蓝牙开关无法关闭；
[image]
[image]

zxystd, 2019-12-31:
> > 好的，另外你的蓝牙信息里传输方式是显示的UART还是USB？加载brcmbluetoothcontroller能不能显示蓝牙开关？我手里没有9系的网卡，帮忙截个图就更好了。
> 
> 蓝牙信息中传输方式显示的是 USB，加载brcmbluetoothcontroller 显示的蓝牙开关无法关闭；
> [image]
> [image]

好的，谢谢

laozhiang, 2020-01-01:
眼看就要成功了。期待第一个release版本

zxystd, 2020-01-01:
关机不断电问题已修复，可查看release下载测试。

closed 2020-01-01 by zxystd