OpenIntelWireless/IntelBluetoothFirmware #3: magicbookpro Catalina10.15.2 Intel9560AC 测试成功

Takagi-sang, 2020-01-01:
两个驱动（冷启动和关闭蓝牙）都有效。

wwwsky, 2020-01-01:
Intel9565AC 能用？？

lqbhhm, 2020-01-01:
IntelAc9560测试成功

wwwsky, 2020-01-01:
Intel9565AC 失败
[image]

zxystd, 2020-01-01:
> Intel9565AC 失败
> [image]

1.请提供一下你的productid
2.另外如果你是用了Injector那个驱动的话，你先只用firmware那一个驱动
3.查看log的命令 log show --last 1d | grep IntelFir

wwwsky, 2020-01-01:
1. DELL .  10.15.2 .       productid    0x8087, 0x0aaa
2.OK 现象目前一致
3. log show --last 1d | grep IntelFir 这条没抓到

wwwsky, 2020-01-01:
1. DELL .  10.15.2 .       productid    0x8087, 0x0aaa
2.OK 现象目前一致
3. log show --last 1d | grep IntelFir 这条没抓到

> > Intel9565AC失败
> > [image]
> 
> 1.请提供一下你的productid 
> 2.另外如果你是用了Injector那个驱动的话，你先只用firmware那一个驱动
> 3.查看log的命令log show --last 1d | grep IntelFir

1. DELL .  10.15.2 .       productid    0x8087, 0x0aaa
2.OK 现象目前一致
3. log show --last 1d | grep IntelFir 这条没抓到

zxystd, 2020-01-01:
> 1. DELL .  10.15.2 .       productid    0x8087, 0x0aaa
>    2.OK 现象目前一致
> 2. log show --last 1d | grep IntelFir 这条没抓到
> 
> > > Intel9565AC失败
> > > [image]
> > 
> > 
> > 1.请提供一下你的productid
> > 2.另外如果你是用了Injector那个驱动的话，你先只用firmware那一个驱动
> > 3.查看log的命令log show --last 1d | grep IntelFir
> 
> 1. DELL .  10.15.2 .       productid    0x8087, 0x0aaa
>    2.OK 现象目前一致
> 2. log show --last 1d | grep IntelFir 这条没抓到

没有日志说明驱动根本就没有找到硬件，先查看一下放的路径是不是对的？其次系统信息里可以看到蓝牙信息么？最好提供多一些信息，方便定位问题

wwwsky, 2020-01-01:
OK  THANKS

zxystd, 2020-01-01:
> OK THANKS

系统里不显示蓝牙的话，查一下usb定制是否正常

javasouls, 2020-01-01:
我的蓝牙和你的id是一样的，我是主板（z390 aorus pro wifi）自带的，可以驱动，你检查一下kexts里面是否包含USBInjectAll.kext，我没放这个进去的时候是不行的，放进去之后就ok了。另外检查下clover里面的system parameters里面的参数 injects kexts 是否为detect，我不太确定这个是否有影响，你可以试试。

总之，非常感谢 zxystd！！！

AlikWu, 2020-01-02:
> Intel9565AC 能用？？

> Intel9565AC 失败
> [image]

> Intel9565AC 失败
> [image]

还有 Intel 9565AC ？ 我一直以为只有 9560AC

closed 2020-07-13 by williambj1