OpenIntelWireless/IntelBluetoothFirmware #10: 感谢大神，小米air13.3 i5 7200u驱动成功，但是无法连接android手机，音响能正常连接

FanZhengTao, 2020-01-02:
首先感谢大神
我这款笔记本网卡应该是intel ac8265
系统 10.15.2
连接手机提示配对，配对后2秒左右断开连接，我测试了小米mix2，华为mate30 pro两款手机，明天可以测试一下iphone11，今天iphone11放公司了，连接小米方盒子蓝牙音响没问题，如果需要测试什么我也可以尽力帮忙

williambj1, 2020-01-02:
我手上的是 9560ac，华为手机匹配上去过几秒就断开了，匹配 iPad 没问题，不用驱动 Windows 热启动进入 macOS 是一样的表现，应该和驱动没什么关系

zxystd, 2020-01-02:
断开连接是正常的，你可以试试配对安卓手机，然后打开安卓手机里的蓝牙网络分享，然后在选择连接网络，你会发现可以连上网了。mac系统只要不是实时传输数据的连接基本上都不会一直维持连接。
[image]

FanZhengTao, 2020-01-02:
> 断开连接是正常的，你可以试试配对安卓手机，然后打开安卓手机里的蓝牙网络分享，然后在选择连接网络，你会发现可以连上网了。mac系统只要不是实时传输数据的连接基本上都不会一直维持连接。
> [image]

多谢回复，确实是可用的，但是我又发现了一个新的小问题，
[image]
在设置界面右键接入网络是连不上的，但是在顶部状态栏的蓝牙小图标里面选接入网络可以，虽然不怎么影响使用，无所谓了，有空你可以看看，谢谢

closed 2020-03-10 by tg567