OpenIntelWireless/IntelBluetoothFirmware #7: 部分神舟笔记本机型测试反馈

kirainmoe, 2020-01-02:
# 网卡型号

- Intel AC9462 (8087, 0aaa)
- Intel AC9560 

# 测试情况

- macOS Catalina 10.15.x 可用：冷启动正常，正常开关，睡眠正常。
- macOS Mojave 10.14.x & High Sierra 10.13.x ：蓝牙显示不可用

感谢大佬的付出！

closed 2020-01-02 by kirainmoe

zxystd, 2020-01-02:
> # 网卡型号
> * Intel AC9462 (8087, 0aaa)
> * Intel AC9560
> 
> # 测试情况
> * macOS Catalina 10.15.x 可用：冷启动正常，正常开关，睡眠正常。
> * macOS Mojave 10.14.x & High Sierra 10.13.x ：蓝牙显示不可用
> 
> 感谢大佬的付出！

感谢测试，release的产物是使用10.15的sdk编译的，所以在低版本无法使用，你可以尝试在低版本自行使用xcode编译即可。

kirainmoe, 2020-01-02:
> > # 网卡型号
> > 
> > * Intel AC9462 (8087, 0aaa)
> > * Intel AC9560
> > 
> > # 测试情况
> > 
> > * macOS Catalina 10.15.x 可用：冷启动正常，正常开关，睡眠正常。
> > * macOS Mojave 10.14.x & High Sierra 10.13.x ：蓝牙显示不可用
> > 
> > 感谢大佬的付出！
> 
> 感谢测试，release的产物是使用10.15的sdk编译的，所以在低版本无法使用，你可以尝试在低版本自行使用xcode编译即可。

明白了，我尝试一下。

kirainmoe, 2020-01-03:
我尝试了：

- 在 10.15 下使用 Xcode 11.3 编译，把 Deployment target 改成 10.14
- 在 10.14.6 下使用 Xcode 11.3 编译

都没有成功在 10.14 下作用。想问这个 kext 是否依赖 10.15 的 API，或者我这样操作是有问题的？

zxystd, 2020-01-03:
> 我尝试了：
> 
> * 在 10.15 下使用 Xcode 11.3 编译，把 Deployment target 改成 10.14
> * 在 10.14.6 下使用 Xcode 11.3 编译
> 
> 都没有成功在 10.14 下作用。想问这个 kext 是否依赖 10.15 的 API，或者我这样操作是有问题的？

没有依赖10.15，但是info.plist的依赖项版本是10.15的，改低一点就好了，改成跟你系统一样或者更低

[image]

kirainmoe, 2020-01-03:
好的，我再尝试一下

kirainmoe, 2020-01-03:
我在 10.14.6 下编译了 kext，已经成功加载并且由其他 10.14.6 用户测试可用了，其他 10.14 版本没有测试，如果各位 10.14 需要可以测试一下。

感谢。

[IntelBluetoothFirmware.kext 10.14.zip](https://github.com/zxystd/IntelBluetoothFirmware/files/4018076/IntelBluetoothFirmware.kext.10.14.zip)

kirainmoe, 2020-01-03:
对了，再多问一下，因为现在这个 kext 体积比较大，我想自己编译一个适用于特定机型的 kext，只保留 9462/9560(0x0aaa) 和 ax200 的固件，请问 fw 文件夹里哪些头文件对应着这些型号的网卡呢

zxystd, 2020-01-03:
> 对了，再多问一下，因为现在这个 kext 体积比较大，我想自己编译一个适用于特定机型的 kext，只保留 9462/9560(0x0aaa) 和 ax200 的固件，请问 fw 文件夹里哪些头文件对应着这些型号的网卡呢

9560是ibt-17-16-1好像
ax200是ibt-20-1-3好像，我使用的就是ax200

kirainmoe, 2020-01-03:
看起来应该是： 17 开头的是 9560 系列的，20 开头的是 AX200 系列的，在 linux 内核的 git 固件里有说明，谢谢

ref: https://github.com/wkennington/linux-firmware/tree/master/intel

JayDragon0523, 2021-05-11:
小米air13.3 i5 8265u一直没有蓝牙图标，系统是10.14.6，没有usbport，用的usballinject