OpenIntelWireless/IntelBluetoothFirmware #8: 蓝牙上传问题

javasouls, 2020-01-02:
目前的设备是
[image]
蓝牙连接音响，耳机听音乐都没问题，判断下行是正常的
但是使用siri和隔空投送均无法使用，判断是上行存在问题。
不知道问题出在什么地方

javasouls, 2020-01-02:
> 目前的设备是 蓝牙连接音响，耳机听音乐都没问题，判断下行是正常的 但是使用siri和隔空投送均无法使用，判断是上行存在问题。 不知道问题出在什么地方
> [image]

重启后正

closed 2020-01-02 by javasouls

zxystd, 2020-01-02:
> 目前的设备是
> 双系统，分别在不同的ssd，efi存在黑苹果所在ssd，windows10直接重启切换到黑苹果的时候有时候会报错卡禁止图标，不知道为什么

我测试了几次好像没出现这个问题哦，观察看看有没有什么规律？