OpenIntelWireless/IntelBluetoothFirmware #11: 小米游戏本 0x8087, 0x0aaa 没有成功

chumen77, 2020-01-03:
[image]
系统 14.6的  只用了主驱动  安装另一个inject驱动后 会不显示蓝牙设备 只用主的 可以显示了 但是无法驱动  还是没办法使用。

zxystd, 2020-01-03:
> [image]
> 系统 14.6的 只用了主驱动 安装另一个inject驱动后 会不显示蓝牙设备 只用主的 可以显示了 但是无法驱动 还是没办法使用。

没有上传固件直接使用inject会导致蓝牙显示不可用。
因为是用的10.15的api编译的，暂时不支持10.15以下的系统，晚点我会编译一个，diy能力强的也可以自行改编译版本和info.plist的依赖库版本自行编译

zxystd, 2020-01-03:
> [image]
> 系统 14.6的 只用了主驱动 安装另一个inject驱动后 会不显示蓝牙设备 只用主的 可以显示了 但是无法驱动 还是没办法使用。

这里有位朋友编译好了10.14可用的，你可以试试
https://github.com/zxystd/IntelBluetoothFirmware/issues/7#issuecomment-570451673

chumen77, 2020-01-10:
> > [image]
> > 系统 14.6的 只用了主驱动 安装另一个inject驱动后 会不显示蓝牙设备 只用主的 可以显示了 但是无法驱动 还是没办法使用。
> 
> 这里有位朋友编译好了10.14可用的，你可以试试
> [#7 (comment)](https://github.com/zxystd/IntelBluetoothFirmware/issues/7#issuecomment-570451673)

十分感谢 刚刚测试 已经可以了 谢谢你为之付出的努力

closed 2020-01-10 by zxystd