OpenIntelWireless/IntelBluetoothFirmware #14: thinkpad x390,i5 8265U测试失败

icedifferent, 2020-01-03:
USB\VID_8087&PID_0AAA\5&1164C3F9&0&10
没有加-v
系统10.15.2
两个kext直接放到other里面，系统报告里面蓝牙信息读不到，设置里面没有蓝牙。

zxystd, 2020-01-03:
> USB\VID_8087&PID_0AAA\5&1164C3F9&0&10
> 没有加-v
> 系统10.15.2
> 两个kext直接放到other里面，系统报告里面蓝牙信息读不到，设置里面没有蓝牙。

看看usb是否正常，加入usbinjectall.kext试试

icedifferent, 2020-01-03:
尴尬，加入了usbinjectall.kext，但是并没有效果

qwe123okd, 2020-01-03:
nuc8i5bek 10.15.2 一样的状况..

zxystd, 2020-01-03:
> nuc8i5bek 10.15.2 一样的状况..

> 尴尬，加入了usbinjectall.kext，但是并没有效果

试试只加Firmware那个驱动，另外在开机之后输入 log show --last 1d | grep IntelFir 命令提取日志

wwwsky, 2020-01-03:
同样情况，usbinjectall.kext，无反应

zxystd, 2020-01-03:
> 同样情况，usbinjectall.kext，无反应

什么都不放的时候是不是在系统里面也看不到蓝牙？是不是被屏蔽掉了？检查一下usb是否正常，如果系统都没挂载这个设备的话，那就不可能会去加载驱动的。

qwe123okd, 2020-01-03:
我的成功了 还是usb定制的问题了 把原来的usbport删了 重新定制了 感谢大佬

icedifferent, 2020-01-04:
> 我的成功了 还是usb定制的问题了 把原来的usbport删了 重新定制了 感谢大佬

你好，请问usb定制的相关教程有没有？可以分享一下吗？

icedifferent, 2020-01-04:
> log show --last 1d | grep IntelFir
好像没有任何显示，，，是usb不正常导致的吗？
ice@icedeMacBook-Pro ~ % log show --last 1d | grep IntelFir
ice@icedeMacBook-Pro ~ %

wwwsky, 2020-01-04:
> 把原来的usbport删了重新定制

把原来的usbport删了重新定制,还没反应，怎么做？

sz-macfei, 2020-01-14:
S1 PRO Intel ac9560 8087＆PID_0AAA，冷启还是不带蓝牙地址，但是热启现在带地址了。还有没有希望再好点哈哈

zxystd, 2020-01-14:
看你系统版本了，10.14的话用这个 https://github.com/zxystd/IntelBluetoothFirmware/issues/7#issuecomment-570451673 试试

sz-macfei, 2020-01-14:
就是10.15呢，说错了。冷启不出蓝牙图标的

zxystd, 2020-01-15:
输入 log show --last 1d | grep IntelFir 命令提取日志看看

sz-macfei, 2020-01-15:
2020-01-14 17:03:52.241480+0800 0x14b      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: recv event=0x000e, length=4
大部分是上面这个，应该是冷起吧。
020-01-14 16:52:20.534955+0800 0x579cd    Default     0x0                  0      0    kernel: (kernel) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0AAA
2020-01-14 16:52:20.535046+0800 0x579cd    Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Start()
2020-01-14 16:52:20.535137+0800 0x579b4    Default     0x0                  0      0    kernel: (kernel) IntelFirmware: setPowerState powerStateOrdinal=1
2020-01-14 16:52:22.035252+0800 0x579cd    Default     0x0                  0      0    kernel: (kernel) IntelFirmware: set configuration to 1
2020-01-14 16:52:22.035945+0800 0x579cd    Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-01-14 16:52:22.035950+0800 0x579cd    Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-01-14 16:52:22.035953+0800 0x579cd    Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-01-14 16:52:22.035956+0800 0x579cd    Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-01-14 16:52:22.035959+0800 0x579cd    Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了只找到了！！！
2020-01-14 16:52:22.035984+0800 0x579cd    Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到中断端点了！！！
2020-01-14 16:52:22.036210+0800 0x579cd    Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到Bulk输出端点了！！！
2020-01-14 16:52:22.036476+0800 0x579cd    Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到Bulk输入端点了！！！
热启倒是很稳，声卡也是需要热启，有什么关联吗

williambj1, 2020-07-13:
请尝试新版本，谢谢

closed 2020-07-13 by williambj1