OpenIntelWireless/IntelBluetoothFirmware #21: GPD P2 Max 0x8087/0x0a2a 驱动成功，但罗技键盘鼠标无法连接

andot, 2020-01-08:
可以正常连接小米手机，PS4 手柄。
但是连接罗技 M590 鼠标和 K780 键盘时，无法正常连接。

```
ce-1b-e9-a1-5d-10：
  地址：	CE-1B-E9-A1-5D-10
  随机地址：	是
  可解析地址：	否
  服务：	
  已配对：	否
  已配置：	否
  已连接：	是
  生产企业：	Nordic Semiconductor (0x7, 0x5A)
  蓝牙核心规范：	4.1
  设备类：	电量不足
  AFH：	打开
  AFH映射：	d50700801f
  RSSI：	-54
  角色：	中央设备
  连接模式：	活跃模式
  时间间隔：	0 ms
```

上面是连接鼠标时，在系统信息里面显示的信息。在 Windows 的下蓝牙鼠标和键盘是可以正常连接的。

langsam1998, 2020-01-08:
我也是，罗技M720和小米蓝牙鼠标都连不上

zxystd, 2020-01-08:
> 可以正常连接小米手机，PS4 手柄。
> 但是连接罗技 M590 鼠标和 K780 键盘时，无法正常连接。
>
> ```
> ce-1b-e9-a1-5d-10：
>   地址：	CE-1B-E9-A1-5D-10
>   随机地址：	是
>   可解析地址：	否
>   服务：
>   已配对：	否
>   已配置：	否
>   已连接：	是
>   生产企业：	Nordic Semiconductor (0x7, 0x5A) [...]

> 我也是，罗技M720和小米蓝牙鼠标都连不上

大概是罗技的对mac兼容性不强吧，这个不属于蓝牙固件影响的范围了，属于系统蓝牙驱动的兼容性问题

836864734, 2020-01-09:
我的也是这样，之前用usb蓝牙的时候，调系统自带驱动是正常的

zxystd, 2020-01-10:
这个属于蓝牙驱动的问题，应该是苹果自带的驱动与intel的指令集不一致造成的，以后有时间的话看看能不能出个fix驱动去解决一些这类兼容问题

closed 2020-01-10 by zxystd

huhu6608, 2022-01-29:
```
Apple 蓝牙软件版本：	6.0.14d6
硬件、功能和设置：
名称：	
地址：	A4-34-D9-0A-5F-9A
支持低功耗蓝牙：	是
支持“接力”：	否
支持智能热点：	否
制造商：	Intel
传输：	USB
固件版本：	8.256 (8.256)
蓝牙电源：	关闭
可被发现：	打开
可连接：	是
自动寻找点：	打开
远程唤醒：	打开
供应商 ID：	0x8087
产品 ID：	0x0A2B
HCI 版本：	4.2 (0x8)
HCI 修正版：	0x100
LMP 版本：	4.2 (0x8)
LMP 子版本：	0x100
设备类型（主要）：	Computer
设备类型（完整）：	Mac Portable
复合类设备：	0x38010C
设备类（主要）：	0x01
设备类（次要）：	0x03
服务类：	0x1C0
自动寻找键盘：	打开
```
我这个显示是0x0A2B ，不在支持列表中，竟然可以跑。音乐播放可以用，但是断断续续的，无线鼠标也同样连不上。 XD