OpenIntelWireless/IntelBluetoothFirmware #23: 睡眠后自动重启(崩溃) 0x0a2b

Menchen, 2020-01-08:
测试了大概3次,3天连着崩溃, 都是刚刚装完 Kext 后才发生的, 就是当长时间睡眠(sleep)的话就提示出错自动重启了.... 今天的话 BT 直接开不了了.. 显示是关闭的而且按开也没反应.

Mac 自带的 Problem Report:
```
panic(cpu 0 caller 0xffffff80208a31fc): Wake transition timed out after 180 seconds while calling power state change callbacks. Suspected bundle: com.apple.iokit.IOUSBHostFamily. Thread 0x3a9bb.
Failure code:: 0x00000004 00000027

Backtracing specified thread
Backtrace (CPU 0), Frame : Return Address
0xffffff9237adb900 : 0xffffff8020263848 
0xffffff92370a3960 : 0xffffff8020161334 
0xffffff92370a39d0 : 0xffffff802015fb1f 
0xffffff92370a3a20 : 0xffffff8020150a63 
0xffffff92370a3a60 : 0xffffff802084039f 
0xffffff92370a3aa0 : 0xffffff80208414a3 
0xffffff92370a3ae0 : 0xffffff7fa0bc3f10 
0xffffff92370a3c50 : 0xffffff7fa1736b24 
0xffffff92370a3cc0 : 0xffffff7fa1858ea2 
0xffffff92370a3d10 : 0xffffff7fa186114d 
0xffffff92370a3d60 : 0xffffff7fa0bc2d0b 
0xffffff92370a3dc0 : 0xffffff8020842d78 
0xffffff92370a3e20 : 0xffffff7fa0bc2a2c 
0xffffff92370a3e30 : 0xffffff8020827d34 
0xffffff92370a3ea0 : 0xffffff8020827b3a 
0xffffff92370a3ec0 : 0xffffff80201811d5 
0xffffff92370a3f40 : 0xffffff8020180d01 
0xffffff92370a3fa0 : 0xffffff80200e613e 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOUSBHostFamily(1.2)[70761A70-11A9-3F83-B2F7-0FCEC749EE99]@0xffffff7fa0baf000->0xffffff7fa0ca6fff
            dependency: com.apple.driver.AppleBusPowerController(1.0)[B4001F35-E1A9-385A-9B45-1AF7ECD5E51E]@0xffffff7fa0b95000
            dependency: com.apple.driver.usb.AppleUSBCommon(1.0)[3DE1C066-5E08-3F07-BC5C-5273A7A38363]@0xffffff7fa0b9d000
            dependency: com.apple.driver.AppleUSBHostMergeProperties(1.2)[D890F1D7-ADE3-3D96-8A24-9A91E5D0F202]@0xffffff7fa0bab000
         com.apple.driver.usb.AppleUSBXHCI(1.2)[9CE96A1A-12FC-3D0B-9F48-14B8D8AB6A75]@0xffffff7fa1722000->0xffffff7fa1779fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[9BA2613C-A129-3EDD-8DC1-2CFF6B62B832]@0xffffff7fa0b62000
            dependency: com.apple.iokit.IOUSBHostFamily(1.2)[70761A70-11A9-3F83-B2F7-0FCEC749EE99]@0xffffff7fa0baf000
            dependency: com.apple.driver.usb.AppleUSBCommon(1.0)[3DE1C066-5E08-3F07-BC5C-5273A7A38363]@0xffffff7fa0b9d000
         com.apple.driver.usb.AppleUSBXHCIPCI(1.2)[85BA754E-AEB3-3F93-9252-684F340EE404]@0xffffff7fa1851000->0xffffff7fa1883fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[9BA2613C-A129-3EDD-8DC1-2CFF6B62B832]@0xffffff7fa0b62000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[E6E855AD-A5B2-3518-A997-1384B4360708]@0xffffff7fa0b25000
            dependency: com.apple.iokit.IOUSBHostFamily(1.2)[70761A70-11A9-3F83-B2F7-0FCEC749EE99]@0xffffff7fa0baf000
            dependency: com.apple.driver.usb.AppleUSBCommon(1.0)[3DE1C066-5E08-3F07-BC5C-5273A7A38363]@0xffffff7fa0b9d000
            dependency: com.apple.driver.usb.AppleUSBXHCI(1.2)[9CE96A1A-12FC-3D0B-9F48-14B8D8AB6A75]@0xffffff7fa1722000

BSD process name corresponding to current thread: kernel_task
Boot args: darkwake=0 gfxrst=1

Mac OS version:
19A602

Kernel version:
Darwin Kernel Version 19.0.0: Wed Sep 25 20:18:50 PDT 2019; root:xnu-6153.11.26~2/RELEASE_X86_64
Kernel UUID: 70EDD61F-86EE-3E1B-873F-98D909B78160
Kernel slide:     0x000000001fe00000
Kernel text base: 0xffffff8020000000
__HIB  text base: 0xffffff801ff00000
System model name: MacBookPro14,1 (Mac-B4831CEBD52A0C4C)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 19609254790744
```

`log show --last 1d | grep IntelIn`
 
```
2020-01-08 12:50:24.903176+0100 0x153      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: recv event=0x000e, length=4
2020-01-08 12:50:24.903911+0100 0x153      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: recv event=0x000e, length=4
2020-01-08 12:50:24.904601+0100 0x153      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: recv event=0x000e, length=4
2020-01-08 12:50:24.905199+0100 0x153      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: recv event=0x000e, length=4
2020-01-08 12:50:24.905356+0100 0x153      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: recv event=0x000e, length=4
[... 88 lines cut ...]
	params.boot_option = 0x00;
	params.boot_param = cpu_to_le32(0x00000000);

	skb = __hci_cmd_sync(hdev, 0xfc01, sizeof(params),
			     &params, HCI_INIT_TIMEOUT);
	if (IS_ERR(skb)) {
		bt_dev_err(hdev, "FW download error recovery failed (%ld)",
			   PTR_ERR(skb));
		return;
	}
	bt_dev_info(hdev, "Intel reset sent to retry FW download");
	kfree_skb(skb);

	/* Current Intel BT controllers(ThP/JfP) hold the USB reset
	 * lines for 2ms when it receives Intel Reset in bootloader mode.
	 * Whereas, the upcoming Intel BT controllers will hold USB reset
	 * for 150ms. To keep the delay generic, 150ms is chosen here.
	 */
	msleep(150);
}
```
这是目前最新的 Linux-next 里面新加的`https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/tree/drivers/bluetooth/btintel.c?h=next-20191121`

附件是完整的(`log show --last 1d | grep IntelIn`)log
[errorLOG.zip](https://github.com/zxystd/IntelBluetoothFirmware/files/4035166/errorLOG.zip)

zxystd, 2020-01-09:
哪个版本？是1.0.1么？

zxystd, 2020-01-09:
看日志并没有执行异常，我刚看了固件有更新，晚上更新一版看看

Menchen, 2020-01-09:
是 1.0.1

Menchen, 2020-01-09:
我再看了一遍log发现这里有异常: (这是今天的, 前几天的我也发现有类似的错误(-536870165))
```
2020-01-09 17:32:02.264121+0100 0x1c02     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware done
2020-01-09 17:32:02.264122+0100 0x1c02     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-01-09 17:32:02.264125+0100 0x1c02     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_RESET
2020-01-09 17:32:02.264129+0100 0x1c02     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: opCode=0xfc01, paramLen=8
2020-01-09 17:32:02.264131+0100 0x1c02     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: check=00
2020-01-09 17:32:02.264337+0100 0x1c02     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Intel reset succeed
2020-01-09 17:32:03.265411+0100 0x1c02     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-01-09 17:32:03.266109+0100 0x154      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: recv event=0x00ff, length=5
2020-01-09 17:32:03.266114+0100 0x154      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 设备固件上传完成
2020-01-09 17:32:03.266118+0100 0x154      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-01-09 17:32:03.266124+0100 0x1c02     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-01-09 17:32:03.266128+0100 0x1c02     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-01-09 17:32:03.266131+0100 0x1c02     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: opCode=0xfc52, paramLen=8
2020-01-09 17:32:03.266135+0100 0x1c02     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: check=87
2020-01-09 17:32:03.266407+0100 0x1c02     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: opCode=0xc03, paramLen=0
2020-01-09 17:32:03.266644+0100 0x1c02     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-01-09 17:32:03.275149+0100 0x154      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: recv event=0x000e, length=4
2020-01-09 17:32:03.275175+0100 0x1c02     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-01-09 17:32:03.275179+0100 0x1c02     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-01-09 17:32:03.275281+0100 0x1be9     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 咋了，这是咋了？(-536870165) operation was aborted)
```

Menchen, 2020-01-10:
今天没崩溃但是 USBHost 出现了问题....
[image]

鼠标和键盘都没用了...再插拔也没用...

zxystd, 2020-01-10:
查看一下usb驱动是否正常

nvollmar, 2020-02-04:
I'm experience the same issue with my Intel AX200. It crashes when entering sleep:

```
panic(cpu 0 caller 0xffffff80102a2d1c): Sleep transition timed out after 180 seconds while calling power state change callbacks. Suspected bundle: com.apple.iokit.IOUSBHostFamily. Thread 0x5d27.
Failure code:: 0x00000004 00000014

Backtracing specified thread
Backtrace (CPU 0), Frame : Return Address
0xffffffa3e24eb900 : 0xffffff800fc60178 mach_kernel : _machine_switch_context + 0xc8
0xffffffa3e23bb9b0 : 0xffffff800fb5d771 mach_kernel : _thread_unstop + 0x1771
0xffffffa3e23bba20 : 0xffffff800fb5bf6f mach_kernel : _thread_block_reason + 0xaf
0xffffffa3e23bba70 : 0xffffff800fb4cdd3 mach_kernel : _lck_mtx_sleep_deadline + 0x73
0xffffffa3e23bbab0 : 0xffffff801023fb9f mach_kernel : __ZN10IOWorkLoop9sleepGateEPvyj + 0xbf
0xffffffa3e23bbaf0 : 0xffffff8010240ca3 mach_kernel : __ZN13IOEventSource9sleepGateEPvyj + 0x53
0xffffffa3e23bbb30 : 0xffffff7f90b79278 com.apple.iokit.IOUSBHostFamily : __ZN22AppleUSBHostController20lowerOnePowerStateToEm + 0x114
0xffffffa3e23bbca0 : 0xffffff7f94c39af4 com.apple.driver.usb.AppleUSBXHCI : __ZN12AppleUSBXHCI20lowerOnePowerStateToEm + 0x2e2
0xffffffa3e23bbd10 : 0xffffff7f94c8be1a com.apple.driver.usb.AppleUSBXHCIPCI : __ZN15AppleUSBXHCIPCI20lowerOnePowerStateToEm + 0x1fc
0xffffffa3e23bbd60 : 0xffffff7f90b78073 com.apple.iokit.IOUSBHostFamily : __ZN22AppleUSBHostController18setPowerStateGatedEmP9IOService + 0x2db
0xffffffa3e23bbdc0 : 0xffffff8010242578 mach_kernel : __ZN13IOCommandGate9runActionEPFiP8OSObjectPvS2_S2_S2_ES2_S2_S2_S2_ + 0x138
0xffffffa3e23bbe20 : 0xffffff7f90b77d94 com.apple.iokit.IOUSBHostFamily : __ZN22AppleUSBHostController13setPowerStateEmP9IOService + 0x32
0xffffffa3e23bbe30 : 0xffffff8010227534 mach_kernel : __ZN9IOService19driverSetPowerStateEv + 0x184
0xffffffa3e23bbea0 : 0xffffff801022733a mach_kernel : __ZN9IOService15pmDriverCalloutEPS_ + 0x2a
0xffffffa3e23bbec0 : 0xffffff800fb7d7e5 mach_kernel : _thread_call_delayed_timer + 0xea5
0xffffffa3e23bbf40 : 0xffffff800fb7d311 mach_kernel : _thread_call_delayed_timer + 0x9d1
0xffffffa3e23bbfa0 : 0xffffff800fae213e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.apple.iokit.IOUSBHostFamily(1.2)[B7CF50DA-73BD-3233-B207-9036BCAD8E1A]@0xffffff7f90b64000->0xffffff7f90c5cfff
            dependency: com.apple.driver.AppleBusPowerController(1.0)[9DFFFA4B-7650-37DD-9892-09A64F1A0BC4]@0xffffff7f90b4a000
            dependency: com.apple.driver.usb.AppleUSBCommon(1.0)[D19CDA63-3A3D-3C02-AEA2-C061630D17A7]@0xffffff7f90b52000
            dependency: com.apple.driver.AppleUSBHostMergeProperties(1.2)[2A37B852-2C0E-3B06-A0D9-B153DD653807]@0xffffff7f90b60000
         com.apple.driver.usb.AppleUSBXHCI(1.2)[4650C412-33F6-322C-A80A-150FC3DA1B18]@0xffffff7f94c25000->0xffffff7f94c7cfff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[3D78401B-5D2D-33BC-9E41-DD2164EA874D]@0xffffff7f90919000
            dependency: com.apple.iokit.IOUSBHostFamily(1.2)[B7CF50DA-73BD-3233-B207-9036BCAD8E1A]@0xffffff7f90b64000
            dependency: com.apple.driver.usb.AppleUSBCommon(1.0)[D19CDA63-3A3D-3C02-AEA2-C061630D17A7]@0xffffff7f90b52000
         com.apple.driver.usb.AppleUSBXHCIPCI(1.2)[71C22181-4BD4-3E9B-AD35-E10850981E3A]@0xffffff7f94c84000->0xffffff7f94cb5fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[3D78401B-5D2D-33BC-9E41-DD2164EA874D]@0xffffff7f90919000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[ADD485B5-3EF8-37C4-B3C5-F86326E497A4]@0xffffff7f9052f000
            dependency: com.apple.iokit.IOUSBHostFamily(1.2)[B7CF50DA-73BD-3233-B207-9036BCAD8E1A]@0xffffff7f90b64000
            dependency: com.apple.driver.usb.AppleUSBCommon(1.0)[D19CDA63-3A3D-3C02-AEA2-C061630D17A7]@0xffffff7f90b52000
            dependency: com.apple.driver.usb.AppleUSBXHCI(1.2)[4650C412-33F6-322C-A80A-150FC3DA1B18]@0xffffff7f94c25000

BSD process name corresponding to current thread: kernel_task
Boot args: keepsyms=1 npci=0x2000 alcid=11 agdpmod=pikera 

Mac OS version:
19D76

Kernel version:
Darwin Kernel Version 19.3.0: Thu Jan  9 20:58:23 PST 2020; root:xnu-6153.81.5~1/RELEASE_X86_64
Kernel UUID: A8DDE75C-CD97-3C37-B35D-1070CC50D2CE
Kernel slide:     0x000000000f800000
Kernel text base: 0xffffff800fa00000
__HIB  text base: 0xffffff800f900000
```

Once I'm removing the USB Port the module is connected to I can properly enter sleep mode (wakeup is still another issue altogether).

nvollmar, 2020-02-19:
Setting the internal USB Port from 'internal' to 'USB3' seems to fix this issue.

Menchen, 2020-02-19:
I’m pretty sure that bluetooth use usb 2.0... But I will try.. Does setting it to USB3 work with airdrop and hand off?

nvollmar, 2020-02-19:
Sorry, was too eager. It properly shuts down the fans but then still reboots a couple of minutes later. There is something else I will try (renaming the _PWR method).

Menchen, 2020-02-19:
@nvollmar By the way, sometimes instead of crashing the USB system just broken, no external USB device was working and no way to shut down except force power down(by pressing power 3 seconds)...

Also, you might want to check kext with `kextlibs` to see dependency issue(Change `Info.plist` to match your version).

nvollmar, 2020-02-19:
I don’t have wifi (yet) so air drop is not working.
I disabled wake up by bluetooth devices and tried sleep again. It properly entered sleep and stayed down until I pushed the power button again.

So far so good. I still have the issue that Catalina reboots when I wake it up... but that is not connected to Bluetooth.

Menchen, 2020-02-23:
By the way, I have tested uploading bt firmware with VM and no crash or weird USB glitch yet...

zjkyz8, 2020-05-05:
[image]
[image]
[image]
根据截图，感觉是返回了abort，此时是否需要放弃后面wakeup的执行，只执行unlock呢？驱动新手，望大佬指点

Menchen, 2020-05-05:
Abort 是正常的，在其他地方有abort

schspa, 2020-07-05:
I have same problems, and I tried last HEAD revision, and the problem has gone.

williambj1, 2020-07-13:
新版本已经修复了，如果问题依旧存在可以重新打开这个issue，谢谢

closed 2020-07-13 by williambj1