OpenIntelWireless/IntelBluetoothFirmware #32: Dependency problems with multiple apple libraries macOs 10.14.1

moda20, 2020-01-19:
hey !
 I downloaded the release version 1.0.1 and put it in /Library/Extensions. upon reloading my kextcache i got this error in the extension preview of system information:

[image]

the kext is not loadable, is this kext only working on catalina 10.15 ?

zxystd, 2020-01-20:
The release 1.0.1 is base on 10.15 sdk, which can not run below catalina, you can try this one
https://github.com/zxystd/IntelBluetoothFirmware/issues/7#issuecomment-570451673

moda20, 2020-01-20:
thx for the comment but I tried the linked file and it didn't change anything from the errors other than the signing error.

I still have the same dependency error : 
[image]

here is how I loaded my kext : 
1. I removed the old kexts
2. I copied the new kext to /Library/Extensions
3. I changed the kext ownership 
4. I reloaded the kextcache

I also started getting crashes every time I reload my kextcache with the bluetooth firmware kext being in the /L/E directory.

kirainmoe, 2020-01-21:
The build attached in that post was based on 10.14.6, which may not work in 10.14.1.

To build  the kext manually:
- Download Xcode from https://developer.apple.com/download/more/ , it would be better if you have the version with SDK that matches your system.
- Clone the repo
- Modify `Info.plist` -> `OSBundleLibraries`, manually change each dependencies' bundle version to the version that your system is loading. 

Takes `com.apple.iokit.IOBluetoothHostControllerTransport` for an example, in `Info.plist`, it looks like this initally:
```xml
		<key>com.apple.iokit.IOBluetoothHostControllerTransport</key>
		<string>7.0f8</string>
```

Open `Terminal.app`, type `kextstat | grep com.apple.iokit.IOBluetoothHostControllerTransport` and enter, you will see something like this:

```
122    2 0xffffff7f835cd000 0xb000     0xb000     com.apple.iokit.IOBluetoothHostControllerTransport (7.0.2f4) D2070464-95EB-3EE4-A1CA-89868BC6387A <114 27 13 12 8 7 6 5 3 1>
```

The "7.0.2f4" is the version of this kext that the system owns, so we change `7.0f8` to `7.0.2f4`.  Modify other dependencies' bundle version like this, then build it with Xcode, you will get a `IntelBluetoothFirmware.kext`, try `sudo kextutil IntelBluetoothFirmware.kext` and see if the kext loads, then copy it to `CLOVER/kexts` or `OC/kexts`.

moda20, 2020-01-21:
@kirainmoe how would I know if the kext would work with the old libraries ? do i just try it or we are sure it will work

williambj1, 2020-07-13:
Should be fixed a long time ago

williambj1, 2020-07-13:
Feel free to reopen if this issue still exists

closed 2020-07-13 by williambj1