OpenIntelWireless/IntelBluetoothFirmware #34: intel nuc 8i7beh无法驱动，好可惜，有没有成功的朋友？

xingxingzaixian, 2020-01-21:


xingxingzaixian, 2020-01-21:
系统是10.15.2，执行kextstat | grep IntelBluetooth命令，无显示信息

xingxingzaixian, 2020-01-21:
执行命令：sudo kextutil IntelBluetoothFirmware.kext
打印如下：
Kext with invalid signature (-2147409652) allowed: <OSKext 0x7f94e5c086d0 [0x7fff86f31090]> { URL = "file:///Users/small_bud/Downloads/IntelBluetoothFirmware.kext/", ID = "com.zxystd.IntelBluetoothFirmware" }
Code Signing Failure: code signature is invalid
Disabling KextAudit: SIP is off
(kernel) Notice - new kext com.apple.driver.usb.AppleUSBHostPlatformProperties, v1.2 matches prelinked kext but can't determine if executables are the same (no UUIDs).

tsingui, 2020-01-22:
该型号能够正常驱动，但是蓝牙键盘无法正常连接！系统10.15.2

wondercoon, 2020-03-13:
NUC8i5正常，0x8087, 0x0aaa
没有用usbinjectall

进bios看下蓝牙设备有没有被禁用

wveilleux, 2020-05-07:
Hi xingxingzaixian, I have exactly the same NUC8 and I have the same "Invalid signature" with this Kext. Did you find out how to resolve it ? (Thanks)

williambj1, 2020-07-13:
Invalid Sig is not a issue and it's fine. Try to troubleshoot on your local machine.

closed 2020-07-13 by williambj1