OpenIntelWireless/IntelBluetoothFirmware #35: 自动睡眠后蓝牙无法使用

Harry-zklcdc, 2020-01-22:
自动睡眠后蓝牙无法使用

zxystd, 2020-01-22:
Please try to use release 1.0.1, it solved this problem.

Harry-zklcdc, 2020-01-22:
> Please try to use release 1.0.1, it solved this problem.

I have tried it. When I‘m closing my laptop's display, my laptop go to sleep, once  I open the display, it can work rightly. But when my laptop sleep automatically(couldn't close the display), I make my laptop awaken, the bluetooth doesn't work

Harry-zklcdc, 2020-01-22:
[image]

kvn1351, 2020-02-08:
I have this very same problem one Matebook X Pro. After sleep it doesn't work anymore. It also fails loading 1/2 boots.

mn19941219, 2020-03-21:
> I have this very same problem one Matebook X Pro. After sleep it doesn't work anymore. It also fails loading 1/2 boots - sometimes it works, sometimes it doesn't.

这应该是USB的问题吧

aluveitie, 2020-04-02:
I don't have any issues after sleep, though it fails often on cold boots and working only after a reboot.

williambj1, 2020-07-13:
已经在新版本中修复

closed 2020-07-13 by williambj1