OpenIntelWireless/IntelBluetoothFirmware #38: SetPowerState, Log spam

mkole, 2020-01-30:
Hi,

the system log is full of these messages if I use  IntelBluetoothFirmware.kext:

kernel: (kernel) IntelFirmware: setPowerState powerStateOrdinal=0
kernel: (kernel) IntelFirmware: setPowerState powerStateOrdinal=1

It looks like my Bluetooth works even without that kext.

---

macOS  - 10.15
Bluetooth  - Intel 3165

Skimige, 2020-02-09:
Can confirm. Using Dell Inspiron 7590 w/ AX200, macOS 10.15.3
BTW, this may cause abnormal power consumption, rasing from 12w to ~31w.
---
这边也能复现相同问题（日志重复），灵越 7590 更换 AX200，macOS 10.15.3
顺带一提，发现如果卸载蓝牙固件可以显著降低功耗（从大概 31w 降到 12w），也和这个重复乱跳的日志很有关系，所以可能这部分还有点 bug

williambj1, 2020-07-13:
Already fixed, thanks for the report

closed 2020-07-13 by williambj1