OpenIntelWireless/IntelBluetoothFirmware #40: 1600X 微星B350 Intel AX200成功驱动

wfsg00098, 2020-01-31:
1600X 微星B350 Intel AX200成功驱动
两个都打上之后启动缓慢用不了
只打Firmware就可以用了，无非是不能开关而已。
Opencore 0.5.5
谢谢大佬

nvollmar, 2020-01-31:
If I understood correctly I'm experience the same issue on AMD Ryzen 3900X, AsRock X570 with the same Intel AX200.

After a cold start with the Injector and Firmware the system is wacky: Apps freeze, I see a lot of errors in the system logs and Bluetooth is not detected at all. If I then reboot everything comes up as expected and the system is working normally including working Bluetooth.

@wfsg00098 thanks for the hint, I'll try without the injector.

wfsg00098, 2020-02-01:
Nope, it doesn't work. I still need a hot start. @nvollmar

wfsg00098, 2020-02-01:
再补充一下，我的系统版本是 10.15.3

zxystd, 2020-02-01:
My test device is also ax200 which is well work with this kext, can you please provide your log via command "log show --last 1d | grep IntelFir" and more info to me.

zxystd, 2020-02-01:
By the way, you should remove -v boot param because of too many logs log by this kext, logging waste  more time, if upload firmware rotinue last over 5 seconds, it will failed.

wfsg00098, 2020-02-01:
[hot_start.txt](https://github.com/zxystd/IntelBluetoothFirmware/files/4142172/hot_start.txt)
2月1日第一次启动是冷启动，只有firmware没有injector。后面几次操作就加入了injector
如果还需要什么启动测试，请回复我

zxystd, 2020-02-01:
I can not download the attachment, please send it to my email 1051244836@qq.com

wfsg00098, 2020-02-01:
sent

zxystd, 2020-02-01:
According to the log, everything works fine, is it version 1.0.1 you are  using?

wfsg00098, 2020-02-01:
yes!

nvollmar, 2020-02-01:
On a cold boot I found this: `2020-02-01 14:09:37.463018+0100 0x25c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-02-01 14:09:37.464686+0100 0x1b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: recv event=0x00ff, length=5
2020-02-01 14:09:37.464688+0100 0x1b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 设备固件上传完成
2020-02-01 14:09:37.464691+0100 0x1b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-02-01 14:09:37.464695+0100 0x25c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-02-01 14:09:37.464697+0100 0x25c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-02-01 14:09:37.464698+0100 0x25c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: opCode=0xfc52, paramLen=8
2020-02-01 14:09:37.464700+0100 0x25c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: check=87
2020-02-01 14:09:37.467682+0100 0x25c      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: opCode=0xc03, paramLen=0`

aluveitie, 2020-04-02:
This issue still persists with macOS 10.15.4 and version 1.0.3 of this kext. I don't even see any logs related to IntelFirmware in either cold and warm boots.

aluveitie, 2020-04-04:
It looks like disabling Deep Sleep in S5 in the BIOS improves the situation.

wfsg00098, 2020-04-05:
@aluveitie Thanks a lot. I will have a try

aluveitie, 2020-04-10:
I've noticed that clearing NVRAM usually results in bluetooth not loading on the next boot. As I checked I've found the `bluetoothActiveControllerInfo` entry in NVRAM and manually added it to the OpenCore section. Afterwards bluetooth did not fail to load anymore after an NVRAM clear. Next I'll try enabling deep sleep again to see if it has an effect.

alanxurox, 2020-04-16:
> I've noticed that clearing NVRAM usually results in bluetooth not loading on the next boot. As I checked I've found the `bluetoothActiveControllerInfo` entry in NVRAM and manually added it to the [...]

@aluveitie That's some exciting & insightful information! By chance, could you share what the NVRAM entry looks like and share just a bit more? Do cold boots now work consistently and is Deep Sleep the only factor causing it to fail in before?

Thanks a whole lot!

aluveitie, 2020-04-17:
@alanxurox Here my NVRAM entry (zeroed out):
[image]

Disabling Deep Sleep fixed cold boots so far. I still got the issue from time to time when updating existing or adding new kexts in OC.

alanxurox, 2020-04-17:
Got it! Thanks! I will look into it this weekend and report back if I find
anything helpful.

williambj1, 2020-07-13:
Should be fixed in v1.1.1, feel free to reopen of this issue still exists

closed 2020-07-13 by williambj1