OpenIntelWireless/IntelBluetoothFirmware #49: 小米笔记本Air13.3 i5-6200u失败，日志文件如下

mn19941219, 2020-02-12:
2020-02-12 19:15:09.279434+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: boot_param=0x40800
2020-02-12 19:15:09.279592+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: send firmware done
2020-02-12 19:15:09.279595+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-02-12 19:15:09.279598+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_RESET
2020-02-12 19:15:09.279734+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Intel reset succeed
2020-02-12 19:15:10.279789+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-02-12 19:15:10.280752+0500 0x147      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 设备固件上传完成
2020-02-12 19:15:10.280757+0500 0x147      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: can not proceed response
2020-02-12 19:15:10.280764+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-02-12 19:15:10.280769+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-02-12 19:15:10.281110+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-02-12 19:15:10.289768+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-02-12 19:15:10.289772+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: End download
2020-02-12 19:15:10.289872+0500 0x1ff      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 咋了，这是咋了？(-536870165) operation was aborted)
2020-02-12 19:15:10.289913+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Stop()
2020-02-12 19:15:10.296053+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver init()
2020-02-12 19:15:10.296062+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Probe()
2020-02-12 19:15:10.296075+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0A2B
2020-02-12 19:15:10.296124+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Start()
2020-02-12 19:15:10.297053+0500 0x1ff      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: setPowerState powerStateOrdinal=1
2020-02-12 19:15:11.797275+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: set configuration to 1
2020-02-12 19:15:11.797766+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-02-12 19:15:11.797770+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-02-12 19:15:11.797773+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-02-12 19:15:11.797776+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了只找到了！！！
2020-02-12 19:15:11.797794+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到中断端点了！！！
2020-02-12 19:15:11.798052+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到Bulk输出端点了！！！
2020-02-12 19:15:11.798298+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到Bulk输入端点了！！！
2020-02-12 19:15:11.798550+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: usb init succeed
2020-02-12 19:15:11.798554+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_VERSION
2020-02-12 19:15:11.799374+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-02-12 19:15:11.799821+0500 0x147      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Firmware revision 0.0 build 10 week 41 2018
2020-02-12 19:15:11.799827+0500 0x147      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: firmware had been download.
2020-02-12 19:15:11.799844+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-02-12 19:15:11.799848+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: End download
2020-02-12 19:15:11.799881+0500 0x1a3      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Stop()

mn19941219, 2020-02-12:
0x8087, 0x0a2b

zxystd, 2020-02-14:
具体表现呢?

mn19941219, 2020-02-16:
> 具体表现呢?

现在已经可以正常使用，就是睡眠唤醒之后失效

mn19941219, 2020-03-13:
已经完美，睡眠唤醒失效不是蓝牙驱动的问题

closed 2020-03-13 by mn19941219

jimmymitlavigne, 2020-03-14:
@mn19941219    你好我和你是同一款的小米air13.3 系统版本是 15.3 
log显示和你的差不多 
请问你是怎么解决这个问题的   我的WiFi现在用不了 蓝牙可用 USBWiFi也用不了

huanghao-sc, 2020-03-15:
你好我也是同款电脑，也遇到一样的问题请问是怎么解决的

huanghao-sc, 2020-03-15:
> @mn19941219 你好我和你是同一款的小米air13.3 系统版本是 15.3
> log显示和你的差不多
> 请问你是怎么解决这个问题的 我的WiFi现在用不了 蓝牙可用 USBWiFi也用不了

我也是一样的问题，请问你解决了吗

jimmymitlavigne, 2020-03-15:
@huanghao-sc    没有解决
你的Wi-Fi是怎么解决的  我的usb Wi-Fi无法识别  目前Mac系统无法上网

mn19941219, 2020-03-15:
> @huanghao-sc 没有解决
> 你的Wi-Fi是怎么解决的 我的usb Wi-Fi无法识别 目前Mac系统无法上网

哪款USB WIFI

jimmymitlavigne, 2020-03-15:
> > @huanghao-sc 没有解决
> > 你的Wi-Fi是怎么解决的 我的usb Wi-Fi无法识别 目前Mac系统无法上网
> 
> 哪款USB WIFI

小米的随身Wi-Fi 型号好像是RT2870