OpenIntelWireless/IntelBluetoothFirmware #51: ThinkPad X250 -Intel 7265AC Bluetooth can connect,but disconnect after 1 second,Pls Help,Thanks.

pwjun, 2020-02-13:
Log:
2020-02-13 17:03:41.976836+0800 0x6fc      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: setPowerState powerStateOrdinal=0
2020-02-13 17:03:41.977160+0800 0xb17f     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: setPowerState powerStateOrdinal=1
2020-02-13 17:03:42.526624+0800 0xb17f     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: setPowerState powerStateOrdinal=0
2020-02-13 17:03:42.526874+0800 0xb17f     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: setPowerState powerStateOrdinal=1
2020-02-13 17:03:43.043825+0800 0x6fc      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: setPowerState powerStateOrdinal=0
2020-02-13 17:03:43.044100+0800 0xb17f     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: setPowerState powerStateOrdinal=1
2020-02-13 17:03:43.600698+0800 0xb17f     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: setPowerState powerStateOrdinal=0
2020-02-13 17:03:43.601026+0800 0xb17f     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: setPowerState powerStateOrdinal=1
2020-02-13 17:03:44.101552+0800 0xb17f     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: setPowerState powerStateOrdinal=0
2020-02-13 17:03:44.101799+0800 0xb17f     Default     0x0                  0      0    kernel: IntelFirmware: setPowerState powerStateOrdinal=0
2020-02-13 17:03:44.101803+0800 0xb17f     Default     0x0                  0      0    kernel: (kernel) IntelFirmware: setPowerState powerStateOrdinal=1
2020-02-13 17:03:44.602188+0800 0xb17f

pwjun, 2020-02-13:
My system 10.15.3

influenist, 2020-02-22:
Same issue here.

williambj1, 2020-02-22:
@pwjun @influenist 

After chatting with @zxystd and did some testing, it seems to be clear that macOS will ALWAYS disconnect devices that seem to be useless.

In order to maintain the connection with another non-apple or non-HID device, like Android phones, you have to turn on Bluetooth  Internet tethering in the phone settings. With this enabled, you'll have a reliable connection and could send files successfully.

This kext, based on what I've understood, is literally a **Firmware Uploader**, it does the same job as other solutions like using VMs, directly rebooting from Windows, and so on. So, if you experience the same issue with other solutions, it has nothing to do with this kext. @zxystd said he will probably make some improvements based on the Lilu patch engine to enhance the user experience but that might happen only after he finishes his juicy Intel Wifi kext.

Hope this can help you guys.

Cheers for @zxystd!

pwjun, 2020-03-29:
@williambj1，Thank your help,on your help,I can connect my computer and smart phone、bluetooth musicbox，but bluetooth mouse can not work，hope help，thanks

williambj1, 2020-03-29:
Bluetooth 4.0 devices are problematic on Intel Cards right now, I can't connect to my keyboard as well. It's an issue mainly caused by incompatibility with Apple's drivers.

pwjun, 2020-03-29:
thank your reply
John

Xpl0itU, 2020-07-17:
May I have your efi? I have the same card but it’s not working for me

opzhang, 2020-07-20:
I put the Bluetooth driver in clover, and it doesn't load at all when I put it in SLE. Intel 3160 network card

elliotnash, 2020-08-06:
is this fixable or a limitation in the system that might not be able to be fixed?

the-eric-kwok, 2020-08-12:
My BT mouse works laggy in BT3.0 mode and not able to connect in BT4.0 mode. 
Subscribed this issue, hopefully there will be some solution of this situation.

msiLiu, 2020-09-08:
> 蓝牙 4.0 设备现在在英特尔卡上有问题， 我也无法连接到键盘。这是一个主要由于与苹果司机不相容而引起的问题。

Will Bluetooth 4.0 be supported in the future？

70599, 2020-09-24:
Nintendo Switch Joy-Con(Bluetooth 4.1) can't be found.

maxis7567, 2020-10-24:
i have same issue with xiaomi mi mouse 
mac 10.15.7
intel 1560

BenDevelopment, 2020-11-10:
Same with MX Anywhere 3 and Satechi M1.
OpenCore 0.6.3
MacOS 10.15.7
IntelBluetoothFirmware 1.1.2

BenDevelopment, 2020-11-16:
I'll continue this discussion https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/217#issuecomment-727535960 here (because this other issue is closed).
I'm a developper but I'm not familiar with this kind of developpment. However, having this issue too, I'd like to propose my help if anyone can lead this :-)!
I'm a C#/Angular developper but I'm confortable with learning new technologies.

zxystd, 2020-11-16:
@BenDevelopment seems that it is HCI command related problems, you can use the HCI packet logger or capture tools to see what happend on connecting.

kilng, 2020-11-17:
[image]
ble debbuger返回是这样，感觉是linux固件兼容问题。
https://stackoverflow.com/questions/32588325/ios-bluetooth-le-code-6-the-connection-has-timed-out-unexpectedly

mackonsti, 2020-11-17:
Hi everyone and thanks for your analysis and contributions. Could it be however that we're uploading the wrong or rather, not-well-compatible firmware on a per-device/card basis? I mean, what if we try other firmwares for my Intel Bluetooth 9460/9560 Module (BTLE v5.1) device [8087:0aaa] and this BTLE has PCI ID is https://pci-ids.ucw.cz/read/PC/8087/0aaa

Could this not be a possible reason for macOS to not connect to some mice, game controllers etc. at all? Just curious. Thanks!

VoltaVX, 2020-11-19:
I only have this issue when using High Sierra, for whatever reason, macOS Catalina and Big Sur don't have any problems with playing Bluetooth audio. My WF-100XM3 earbuds disconnect when I play audio on my computer, but it Catalina and Big Sur, they work perfectly.

VoltaVX, 2020-11-19:
I wonder if there's some way to port the Bluetooth stack from Catalina to High Sierra, if it's even possible

jgyjiayou, 2020-12-04:
希望大佬挤出时间，解决热粉的难题！！已经等待6个月了

SnowsonZ, 2020-12-05:
+11111111

Basstorm, 2020-12-07:
`Dec 07 09:03:18.714  HCI Command      0x0000                     LE Set Scan Parameters - Active - 30/300 (ms)  
Dec 07 09:03:18.715  HCI Command      0x0000                     LE Set Scan Enable - 0x01, Filter Duplicates: 1  
Dec 07 09:03:18.717  HCI Command      0x0000                     LE Create Connection - 00:00:00:00:00:00, Scan Window/Interval: 30ms/60ms, Min/Max Conn Interval: 7.5ms/15ms   
Dec 07 09:03:21.558  HCI Command      0x0000                     HCI Inquiry Cancel  
Dec 07 09:03:21.559  HCI Command      0x0000                     LE Set Scan Enable - 0x00, Filter Duplicates: 1  
Dec 07 09:03:21.835  HCI Command      0x0000                     Read Scan Enable  
Dec 07 09:03:21.842  HCI Command      0x0000                     LE Create Connection Cancel  
Dec 07 09:03:21.845  HCI Command      0x0000                     Vendor Specific Command  
Dec 07 09:03:21.847  Error            0x0000                     **** [IOBluetoothHostController][ValidateHCIRequestReceivedEvents] -- ERROR -- Received Command Status event for opCode 0xFCE9 (Broadcom VSC -- LE Meta VSC) -- this = 0x8a00 ****  
Dec 07 09:03:21.847  Error            0x0000                     [IOBluetoothHostController][ProcessEventDataWL] -- broadcastDataPtr is NOT NULL and broadcastDataSize > 0 -- ParseHCIEvent() failed -- Error code 0xE00002C2 (kIOReturnBadArgument) -- opCode = 0xFCE9 (Broadcom VSC -- LE Meta VSC),  requestPtr = 0xffffff8038259000, eventStatus = 0x0c (kBluetoothHCIErrorCommandDisallowed)   
Dec 07 09:03:21.847  Error            0x0000                     [IOBluetoothHostController][ProcessEventDataWL] -- Not enough room in request's result data buffer (3 bytes) to copy the actual data (2048 bytes) -- opCode = 0xFCE9 (Broadcom VSC -- LE Meta VSC)  
`
用packetlogger看 这几句一直循环
[image]

m2faridi, 2020-12-22:
MacOS can't connect to Logitech MX Master/Anywhere Mouse
pls help

mishurov, 2021-01-01:
Something wrong with the hardware settings. It has problems on Linux too in the GUI. If I use the command line tool, it always asks `Accept pairing (yes/no)` and only after that it works properly. May be there're some vendor specific HCI commands which enable the Intel device to automatically accept pairing requests.

```
bluetoothctl
[bluetooth]# scan on
Discovery started
[NEW] Device xx:72:07:xx:xx:xx BluetoothMouse360
[CHG] Device xx:72:07:xx:xx:xx Connected: yes
Request authorization
[agent] Accept pairing (yes/no): yes
```

mishurov, 2021-01-04:
@Basstorm, I watched the logs on my machine too, it looks like the problem with the Broadcom vendor specific commands which aren't understandable by the Intel controllers, may be only this one is a culprit `0xFCE9 (Broadcom LE Meta VSC)`. They should return kIOReturnSuccess looking at the logs on the actual Macs. May be there's a way to hack into this process and ignore those commands.

I also wonder if enabling Microsoft vendor extension would help.
https://github.com/torvalds/linux/blob/master/drivers/bluetooth/btusb.c#L2923

jgyjiayou, 2021-01-06:
大佬，啥时候解决兼容性问题啊。

Herrnobiz, 2021-02-08:
Same issue with bluetooth 4.X on 2 different machines both with Big sur:
First machine use an intel AX200, i can discover the bt4 devices but cannot finish the connection (xiaomi miiiw s500 mouse)
Second machine use a intel 8260 or 8265, discovery of devices works, connection to an apple wireless keyboard 2009 works, cannot connect to a bt4 mouse.

eyalyoli, 2021-03-13:
Same issue here... I can help to test anything...

ByTyoma, 2021-03-23:
Same issue for me.. 
Strix B450i with AX200
Paired my airpods pro immediately 
Can't connect any BT4.x device

@eyalyoli תעדכן אם הצלחת

Sheripov, 2021-03-23:
I try connect bluetooth headset "mi air 2" it has bluetooth 5.0 and it connected successfully but microphone all not work.

eyalyoli, 2021-03-23:
The current solution to avoid the problem is to upgrade the Bluetooth card to v5 or better go with Broadcom cards. 
The best you can try with current card is to install itlwm (with Heliport) instead of Airportitlwm as it has better compatibily with the Bluetooth.

ByTyoma, 2021-03-23:
> The current solution to avoid the problem is to upgrade the Bluetooth card to v5 or better go with Broadcom cards.
> The best you can try with current card is to install itlwm (with Heliport) instead of Airportitlwm as it has better compatibily with the Bluetooth.

my card is AX200 (wifi 6 + BT5.1)..

eyalyoli, 2021-03-23:
> > The current solution to avoid the problem is to upgrade the Bluetooth card to v5 or better go with Broadcom cards.
> > The best you can try with current card is to install itlwm (with Heliport) instead of Airportitlwm as it has better compatibily with the Bluetooth.
> 
> my card is AX200 (wifi 6 + BT5.1)..

Then try the 2nd solution I wrote...

ByTyoma, 2021-03-24:
> Then try the 2nd solution I wrote...

I tried.. 
i think that the problem is somewhere in new bios v4204 or opencore 0.6.7.. 🤷🏼‍♂️

wmailn, 2021-03-25:
同样遇到，可以用蓝牙音响、手绘板、耳机，就是没法用蓝牙键鼠，键盘逻辑 mx keys,鼠标 mx anywhere2

wszcug, 2021-03-26:
the same problem with rog strix b550i which the wifi&bluetooth chip is ax200, i found that the magickeybord、airpod pro work well but my mouse mx anywhere 2s do not work..so pity

sojab0on, 2021-03-27:
Same issue here on Intel 9560 160mhz combo card 
Logs from the command for the intel firmware 
michel@MacBook-Air-van-michel ~ % log show --last boot | grep IntelFirmware
2021-03-27 19:15:22.590414+0100 0x191      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI Timeout, retry
2021-03-27 19:15:22.623287+0100 0x191      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2021-03-27 19:15:22.650828+0100 0x191      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_RESET_BL
2021-03-27 19:15:22.893550+0100 0x191      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2021-03-27 19:15:22.943082+0100 0x191      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...
2021-03-27 19:15:22.971962+0100 0x6f       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
2021-03-27 19:15:22.998472+0100 0x6f       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()
2021-03-27 19:15:23.002282+0100 0x75a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2021-03-27 19:15:23.018563+0100 0x75a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2021-03-27 19:15:23.018891+0100 0x75a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0AAA
2021-03-27 19:15:23.019409+0100 0x75a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2021-03-27 19:15:24.520992+0100 0x75a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2021-03-27 19:15:24.521615+0100 0x75a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found interface!!!
2021-03-27 19:15:24.521642+0100 0x75a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Interrupt endpoint!
2021-03-27 19:15:24.521941+0100 0x75a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk out endpoint!
2021-03-27 19:15:24.522172+0100 0x75a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found Bulk in endpoint!
2021-03-27 19:15:24.522436+0100 0x75a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2021-03-27 19:15:24.522450+0100 0x75a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2021-03-27 19:15:24.522916+0100 0x75a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2021-03-27 19:15:24.523351+0100 0x175      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Firmware revision 0.0 build 109 week 20 2020
2021-03-27 19:15:24.523355+0100 0x175      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: suspect device firmware: ibt-19-16-0.sfi
2021-03-27 19:15:24.523358+0100 0x175      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: firmware had been download.
2021-03-27 19:15:24.523370+0100 0x75a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2021-03-27 19:15:24.523373+0100 0x75a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2021-03-27 19:15:24.523621+0100 0x75a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2021-03-27 19:15:24.524372+0100 0x75a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2021-03-27 19:15:24.524394+0100 0x75a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2021-03-27 19:15:24.524407+0100 0x75a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Clean up...

I have tested multiple Bluetooth 4.0 devices 
Not working 
MX Master 
Microsoft Designer mouse 
Also my MX5500 Bluetooth desktop won't work either 

Strage thing is that things like Bluetooth headphones or speakers work no problem 
tested with 
Jabra T110BT and Kove 101 headphones 
Also with JBL Charge 2 and Kove commuter 2 Bluetooth speakers

khmyznikov, 2021-04-01:
@zxystd @williambj1 @stevezhengshiqi @aisk Guys can we donate somewhere to help you with this problem?

williambj1, 2021-04-01:
@khmyznikov Sorry but we do not accept donations. You are welcome to submit pull requests.

khmyznikov, 2021-04-01:
> @khmyznikov Sorry but we do not accept donations. You are welcome to submit pull requests.
I have 0 experience in kext fixing, but I and other guys can support development. Feel free to DM in telegram or twitter to share credentials.

skorpions2000, 2021-04-27:
Мы останомся без блютуз?

khmyznikov, 2021-04-27:
> Мы останомся без блютуз?

Денег брать не хотят, чинить тоже не хотят... Не знаю даже.

ByTyoma, 2021-04-27:
> > Мы останомся без блютуз?
> 
> Денег брать не хотят, чинить тоже не хотят... Не знаю даже.

Я за**ался ждать.. взял себе пока BCM95360NG.. 
если починят, поставлю ax200 обратно

khmyznikov, 2021-04-27:
> > > Мы останомся без блютуз?
> > 
> > 
> > Денег брать не хотят, чинить тоже не хотят... Не знаю даже.
> 
> Я за**ался ждать.. взял себе пока BCM95360NG..
> если починят, поставлю ax200 обратно

Самый нормальный выход сейчас. Жаль не у всех можно поменять, на моем вшит wifi.

ByTyoma, 2021-04-27:
> > > > Мы останомся без блютуз?
> > > 
> > > 
> > > Денег брать не хотят, чинить тоже не хотят... Не знаю даже.
> > 
> > 
> > Я за**ался ждать.. взял себе пока BCM95360NG..
> > если починят, поставлю ax200 обратно
> 
> Самый нормальный выход сейчас. Жаль не у всех можно поменять, на моем вшит wifi.

Какая у тебя материнка?

khmyznikov, 2021-04-27:
> > > > > Мы останомся без блютуз?
> > > > 
> > > > 
> > > > Денег брать не хотят, чинить тоже не хотят... Не знаю даже.
> > > 
> > > 
> > > Я за**ался ждать.. взял себе пока BCM95360NG..
> > > если починят, поставлю ax200 обратно
> > 
> > 
> > Самый нормальный выход сейчас. Жаль не у всех можно поменять, на моем вшит wifi.
> 
> Какая у тебя материнка?

Ноут, XPS свежий.

skorpions2000, 2021-04-27:
а действительно что делать? есть ли какой-то способ это исправить? может какой-то юсб блютуз? какой из коробки поддерживается?

ByTyoma, 2021-04-27:
> Ноут, XPS свежий.

Точно вшита? там везде они апгрейдятся без проблем.. какая модель?

> а действительно что делать? есть ли какой-то способ это исправить? может какой-то юсб блютуз? какой из коробки поддерживается?

Asus BT400 поддерживается, и все которые на CSR8510

skorpions2000, 2021-04-27:
> > > > > > > Мы останомся без блютуз?
> > > > > > 
> > > > > > 
> > > > > > Денег брать не хотят, чинить тоже не хотят... Не знаю даже.
> > > > > 
> > > > > 
> > > > > Я за**ался ждать.. взял себе пока BCM95360NG..
> > > > > если починят, поставлю ax200 обратно
> > > > 
> > > > 
> > > > Самый нормальный выход сейчас. Жаль не у всех можно поменять, на моем вшит wifi.
> > > 
> > > 
> > > Какая у тебя материнка?
> > 
> > 
> > Ноут, XPS свежий.
> 
> Точно вшита? там везде они апгрейдятся без проблем.. какая модель?
> 
> > а действительно что делать? есть ли какой-то способ это исправить? может какой-то юсб блютуз? какой из коробки поддерживается?
> 
> Asus BT400 поддерживается, и все которые на CSR8510. (airdrop не будет работать)

BCM20702A0 этот должен работать? =)

ByTyoma, 2021-04-27:
> > > > > > > > Мы останомся без блютуз?
> > > > > > > 
> > > > > > > 
> > > > > > > Денег брать не хотят, чинить тоже не хотят... Не знаю даже.
> > > > > > 
> > > > > > 
> > > > > > Я за**ался ждать.. взял себе пока BCM95360NG..
> > > > > > если починят, поставлю ax200 обратно
> > > > > 
> > > > > 
> > > > > Самый нормальный выход сейчас. Жаль не у всех можно поменять, на моем вшит wifi.
> > > > 
> > > > 
> > > > Какая у тебя материнка?
> > > 
> > > 
> > > Ноут, XPS свежий.
> > 
> > 
> > Точно вшита? там везде они апгрейдятся без проблем.. какая модель?
> > > а действительно что делать? есть ли какой-то способ это исправить? может какой-то юсб блютуз? какой из коробки поддерживается?
> > 
> > 
> > Asus BT400 поддерживается, и все которые на CSR8510. (airdrop не будет работать)
> 
> BCM20702A0 этот должен работать? =)
> наверное это он, выглядит так же
> https://www.asus.com/us/networking-iot-servers/adapters/all-series/usbbt400/

Должен работать.. (может понадобится кекст [BrcmPatchRam](https://github.com/acidanthera/BrcmPatchRAM) )

skorpions2000, 2021-04-27:
> Должен работать.. (может понадобится кекст [BrcmPatchRam](https://github.com/acidanthera/BrcmPatchRAM) )

можно какой-то линк на инструкцию по установке, хоть с чего начинать? даже не знаю с чего начинать)

ByTyoma, 2021-04-27:
> можно какой-то линк на инструкцию по установке, хоть с чего начинать? даже не знаю с чего начинать)

 Посмотри [ЭТО](https://www.youtube.com/watch?v=O2WaZ9G-bJc)

mishurov, 2021-07-04:
Fixed in Monterey. Apple rewrote the BT stack. The only problem now is when I reboot from Linux with the latest firmware, it doesn't work in MacOS, probably the driver needs to downgrade a loaded firmware if a version is different.

lgariv, 2021-07-20:
> Fixed in Monterey. Apple rewrote the BT stack. The only problem now is when I reboot from Linux with the latest firmware, it doesn't work in MacOS, probably the driver needs to downgrade a loaded firmware if a version is different.

Which devices have you tried that with? Which Intel card?

mishurov, 2021-07-20:
@lgariv Microsoft Bluetooth Mobile Mouse 3600, Intel® Wi-Fi 6 AX200 (Gig+)

lgariv, 2021-07-20:
> @lgariv Microsoft Bluetooth Mobile Mouse 3600, Intel® Wi-Fi 6 AX200 (Gig+)

So a Bluetooth 5.0 card if I’m not mistaken?

Can anyone confirm that with a Bluetooth 4.0 card as well? Anyone with a Logitech device (as they tend to create a lot of problems of this sort)?

mishurov, 2021-07-20:
@lgariv yes, depending on a loaded firmware it supports BT 5.0, 5.1 and 5.2

SnowsonZ, 2021-10-17:
麻烦问一下，升级到Monterey后，4.x的鼠标就正常了吗？

wszcug, 2021-10-17:
不是 最后我还是换了主板自带的蓝牙模块

发自我的iPhone

[33 more replies]