OpenIntelWireless/IntelBluetoothFirmware #53: 10.15.3 matebook14 蓝牙耳机无法使用麦克风

zwd1216, 2020-02-14:
10.15.3，matebook14，蓝牙耳机无法使用麦克风。
输入端将输入调至蓝牙耳机时会导致电脑卡死，且无任何声音

handhackaaron, 2020-02-18:
同样问题

zxystd, 2020-02-20:
Some bluetooth device has compatibility problems on macOS, maybe I can try to find out why and make some patch to fix these problems when I have time. You can provide your bluetooth device model here. Thanks for your feedback.

ha7rock, 2020-02-21:
同样问题，但是我发现上一个版本（1.0.1）我曾经正常使用过 micphone 功能。蓝牙设备是 Apple airpods 。 感谢作者辛苦付出

zwd1216, 2020-02-21:
测试sony1000XM2 以及AirPods Pro 
热启动时偶可正常使用micphone功能

zwd1216, 2020-02-21:
发现热启动偶可正常使用麦克风

trevorbenyack, 2020-03-31:
Having the same problem with AirPods Pro (as well as a pair of Taotronics and Phaiser bt earbuds).... Can listen, but no mic/audio in.

kuibug, 2020-04-09:
9260ac在10.15.4中有同样的问题，附上一段日志

```
2020-04-09 18:54:46.161406+0800  localhost kernel[0]: (kernel) IntelFirmware: interrupt wait
2020-04-09 18:54:46.161727+0800  localhost kernel[0]: (kernel) IntelFirmware: 设备固件上传完成
2020-04-09 18:54:46.161732+0800  localhost kernel[0]: (kernel) IntelFirmware: can not proceed response
2020-04-09 18:54:46.161739+0800  localhost kernel[0]: (kernel) IntelFirmware: interrupt continue
2020-04-09 18:54:46.161747+0800  localhost kernel[0]: (kernel) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-04-09 18:54:46.162067+0800  localhost kernel[0]: (kernel) IntelFirmware: interrupt wait
2020-04-09 18:54:46.172695+0800  localhost kernel[0]: (kernel) IntelFirmware: interrupt continue
2020-04-09 18:54:46.172708+0800  localhost kernel[0]: (kernel) IntelFirmware: End download
2020-04-09 18:54:46.172779+0800  localhost kernel[0]: (kernel) IntelFirmware: 咋了，这是咋了？(-536870165) operation was aborted)
2020-04-09 18:54:46.172824+0800  localhost kernel[0]: (kernel) IntelFirmware: Driver Stop()
2020-04-09 18:54:46.172903+0800  localhost kernel[0]: (IOBluetoothHostControllerUSBTransport) <IOBluetoothHostControllerUSBTransport`IOBluetoothHostControllerUSBTransport::start(IOService*)> **** [IOBluetoothHostControllerUSBTransport][start] -- completed -- result = TRUE -- 0x9000 ****
2020-04-09 18:54:46.172908+0800  localhost kernel[0]: (BroadcomBluetoothHostControllerUSBTransport) <BroadcomBluetoothHostControllerUSBTransport`BroadcomBluetoothHostControllerUSBTransport::start(IOService*)> **** [BroadcomBluetoothHostControllerUSBTransport][start] -- calling retain() on mBluetoothUSBHostDevice -- Matched on device ****
2020-04-09 18:54:46.172930+0800  localhost kernel[0]: (BroadcomBluetoothHostControllerUSBTransport) <BroadcomBluetoothHostControllerUSBTransport`BroadcomBluetoothHostControllerUSBTransport::start(IOService*)> **** [BroadcomBluetoothHostControllerUSBTransport][start] -- Completed (matched on Device) -- 0x9000 ****
```

jonktsui, 2020-06-08:
Hey! Thanks for the development of this kext. I have similar issue with Audio working, but Mic not working.
Wifi Card: Intel 9560 (on Asus z390-i mini-itx)
Headphones: Bose QuietComfort 35ii

Logs: 
`2020-06-06 21:31:25.913285-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-06-06 21:31:25.922966-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-06-06 21:31:25.932595-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0AAA
2020-06-06 21:31:25.943509-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-06-06 21:31:27.454583-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2020-06-06 21:31:27.465881-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-06-06 21:31:27.477677-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-06-06 21:31:27.488429-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-06-06 21:31:27.499454-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了只找到了！！！
2020-06-06 21:31:27.509907-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到中断端点了！！！
2020-06-06 21:31:27.520469-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到Bulk输出端点了！！！
2020-06-06 21:31:27.531222-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到Bulk输入端点了！！！
2020-06-06 21:31:27.541892-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2020-06-06 21:31:27.551544-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-06-06 21:31:27.561466-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-06-06 21:31:27.570915-0400 0x13b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-06-06 21:31:27.580319-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-06-06 21:31:27.589624-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-06-06 21:31:27.598945-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-06-06 21:31:27.608105-0400 0x13b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-06-06 21:31:27.617376-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-06-06 21:31:27.626366-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-06-06 21:31:27.635565-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-06-06 21:31:27.644202-0400 0x13b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Firmware revision 0.1 build 20 week 53 2019
2020-06-06 21:31:27.645375-0400 0x13b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: firmware had been download.
2020-06-06 21:31:27.654377-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-06-06 21:31:27.663150-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-06-06 21:31:27.671353-0400 0x184      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
2020-06-06 21:31:41.525933-0400 0x4df      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-06-06 21:31:41.530217-0400 0x4df      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-06-06 21:31:41.530375-0400 0x4df      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0AAA
2020-06-06 21:31:41.530608-0400 0x4df      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-06-06 21:31:43.032334-0400 0x4df      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
[... cut ...]
2020-06-06 21:31:43.133399-0400 0x13b      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: firmware had been download.
2020-06-06 21:31:43.142094-0400 0x4df      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-06-06 21:31:43.150434-0400 0x4df      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-06-06 21:31:43.158492-0400 0x4df      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()`

williambj1, 2020-07-13:
Unfortunately this kext is only a firmware uploader, if it's related to the firmware please specify which version of the kext worked well. Other than that, this kext doesn't handle audio related stuff. Sorry for the inconvenience.

closed 2020-07-13 by williambj1