OpenIntelWireless/IntelBluetoothFirmware #54: System Information thinks BT manuf is now Ericsson

sajjad365, 2020-02-14:
I used both Intel Kexts, didn't work so I reverted to original config.plist and removed the kexts but the System Information > Bluetooth is now showing manufacturer as Ericsson instead of Intel previously:

Well, for reasons unknown, System Information > Bluetooth now thinks it's an Ericsson (instead of Intel previously). I have undone everything but it's stuck like this...
Address: 00-00-00-00-00-00
Bluetooth Low Energy Supported: No
Handoff Supported: No
Instant Hot Spot Supported: No
**Manufacturer: Ericsson**
Transport: USB
Firmware Version: 0.0 (0.0)
Bluetooth Power: On
Discoverable: Off
Connectable: No
Auto Seek Pointing: Off
Remote wake: On
Vendor ID: 0x8087
Product ID: 0x0AAA
HCI Version: 1.0b (0x0)
HCI Revision: 0x00
LMP Version: 1.0b (0x0)
LMP Subversion: 0x00
Auto Seek Keyboard: Off

I didn't capture this information prior to installing the kexts.

Before before installation: BT was on, could not be turned off and scanning but nothing discovered. After installing kexts, same behavior. After removing kext, same behavior.

Running Mojave using Opencore.

Menchen, 2020-02-14:
A full shutdown down, and wait 5m before you boot again should fix it.

sajjad365, 2020-02-14:
After reversing all the changes, I restarted several times, complete shutdown and also unplugged power cable. No changes.

zxystd, 2020-02-14:
dont worry about the manufacture what it display. it seems that the kexts are not work, may be there have wrong dependencies error, you can try to load it manually, following steps.
open terminal and change directory to the kexts path.
sudo chown -R root:wheel IntelBluetoothFirmware.kext
sudo kextutil -v IntelBluetoothFirmware.kext
kextlibs -undef-symbols IntelBluetoothFirmware.kext

and then show me the log. thanks

sajjad365, 2020-02-14:
> dont worry about the manufacture what it display. it seems that the kexts are not work, may be there have wrong dependencies, you can try to load it manually, following steps.

Do I need to make any changes to config.plist? Sorry for ignorant question, I am new to Hackintosh and OC too.

sajjad365, 2020-02-14:
I dont think it liked that:

```
Bashirs-Hackintosh:Desktop bashir$ log show --last boot | grep IntelFirmware
Bashirs-Hackintosh:Desktop bashir$ sudo chown -R root:wheel IntelBluetoothFirmware.kext
Password:
Bashirs-Hackintosh:Desktop bashir$ sudo kextutil -v IntelBluetoothFirmware.kext
Defaulting to kernel file '/System/Library/Kernels/kernel'
Untrusted kexts are not allowed
Kext with invalid signature (-2147409652) denied: /Library/StagedExtensions/Users/bashir/Desktop/C4B9023B-12B3-40E4-8E2B-239F82A6819C.kext
Bundle (/Users/bashir/Desktop/IntelBluetoothFirmware.kext) failed to validate, deleting: /Library/StagedExtensions/Users/bashir/Desktop/C4B9023B-12B3-40E4-8E2B-239F82A6819C.kext
Unable to stage kext (/Users/bashir/Desktop/IntelBluetoothFirmware.kext) to secure location.
Bashirs-Hackintosh:Desktop bashir$ kextlibs -undef-symbols IntelBluetoothFirmware.kext
For all architectures:
    com.apple.iokit.IOUSBHostFamily = 1.2
    com.apple.kpi.iokit = 18.7
    com.apple.kpi.libkern = 18.7
    com.apple.kpi.mach = 18.7

For x86_64:
    2 symbols not found in any library kext:
	__ZN9IOService8DispatchE5IORPC
	__ZN15OSMetaClassBase8DispatchE5IORPC

Bashirs-Hackintosh:Desktop bashir$ log show --last boot | grep IntelFirmware
Bashirs-Hackintosh:Desktop bashir$
```

zxystd, 2020-02-14:
you should close SIP to support unauthorize kext.

nvollmar, 2020-02-14:
I had Ericsson showing up as manufacturer before installing this kext. I guess MacOS partially recognized the Intel Bluetooth by itself.

sajjad365, 2020-02-14:
> you should close SIP to support unauthorize kext.

Disabled and still same result and output as earlier.

sajjad365, 2020-02-14:
Nvm, it's enabled again automatically...WTF? I disabled from Recovery...why is it going back UGH!!!!

```
Bashirs-Hackintosh:Desktop bashir$ csrutil status
System Integrity Protection status: enabled.
```

sajjad365, 2020-02-14:
> you should close SIP to support unauthorize kext.

OK, disabled from config.plist in EFI and here's the result:

```
Bashirs-Hackintosh:Desktop bashir$ sudo chown -R root:wheel IntelBluetoothFirmware.kext
Password:
Bashirs-Hackintosh:Desktop bashir$ sudo kextutil -v IntelBluetoothFirmware.kext
Defaulting to kernel file '/System/Library/Kernels/kernel'
Kext with invalid signatured (-2147409652) allowed: <OSKext 0x7fb67ae23790 [0x7fff8ce0c8e0]> { URL = "file:///Users/bashir/Desktop/IntelBluetoothFirmware.kext/", ID = "com.zxystd.IntelBluetoothFirmware" }
/Users/bashir/Desktop/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOBluetoothFamily.
/Users/bashir/Desktop/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOUSBHostFamily.
/Users/bashir/Desktop/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOBluetoothFamily.
/Users/bashir/Desktop/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOUSBHostFamily.
/Users/bashir/Desktop/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOBluetoothFamily.
/Users/bashir/Desktop/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOUSBHostFamily.
Code Signing Failure: code signature is invalid
Dependency Resolution Failures: 
    Only incompatible kexts found for these libraries: 
        com.apple.iokit.IOBluetoothFamily
        com.apple.iokit.IOUSBHostFamily

Diagnostics for /Users/bashir/Desktop/IntelBluetoothFirmware.kext:
Dependency Resolution Failures: 
    Only incompatible kexts found for these libraries: 
        com.apple.iokit.IOBluetoothFamily
        com.apple.iokit.IOUSBHostFamily

Bashirs-Hackintosh:Desktop bashir$ kextlibs -undef-symbols IntelBluetoothFirmware.kext
For all architectures:
    com.apple.iokit.IOUSBHostFamily = 1.2
    com.apple.kpi.iokit = 18.7
    com.apple.kpi.libkern = 18.7
    com.apple.kpi.mach = 18.7

For x86_64:
    2 symbols not found in any library kext:
	__ZN9IOService8DispatchE5IORPC
	__ZN15OSMetaClassBase8DispatchE5IORPC
```

Menchen, 2020-02-14:
You can try to disable it from OpenCore... in nvram-> csr-active-config

```
<key>NVRAM</key>
	<dict>
		<key>Add</key>
		<dict>
			<key>7C436110-AB2A-4BBB-A880-FE41995C9F82</key>
			<dict>
				<key>boot-args</key>
				<string>darkwake=0 gfxrst=1</string>
				<key>bootercfg</key>
				<string>log=0 debug=0 level=0</string>
				<key>csr-active-config</key>
				<data>ZwAAAA==</data>
				<key>nvda_drv</key>
				<data></data>
				<key>prev-lang:kbd</key>
				<data></data>
			</dict>
		</dict>
		<key>Block</key>
		<dict/>
		<key>LegacyEnable</key>
		<false/>
		<key>LegacySchema</key>
		<dict/>
	</dict>
```

sajjad365, 2020-02-14:
> You can try to disable it from OpenCore... in nvram-> csr-active-config

Yes, that's how I did it. And ran the commands you provided earlier. The result is posted above. Please advise.

zxystd, 2020-02-15:
Now the error is the mistaken dependencies version, according to your command result
For all architectures:
    com.apple.iokit.IOUSBHostFamily = 1.2
    com.apple.kpi.iokit = 18.7
    com.apple.kpi.libkern = 18.7
    com.apple.kpi.mach = 18.7

you can edit the info.plist, find these dependencies and replace with your version and remove other not used, just like this:
[image]

sajjad365, 2020-02-15:
> you can edit the info.plist, find these dependencies and replace with your version and remove other not used, just like this:

Sorry, could you clarify where this file is located? And where can I obtain the required architecture files?

zxystd, 2020-02-15:
Right click on the driver, select "show detail", and you will see a directory, find the info.plist and modify it. just use the version what i just provided.

zxystd, 2020-02-15:
if you still dont know how to do, I will give the correct driver to you on the morning.

sajjad365, 2020-02-15:
> if you still dont know how to do, I will give the correct driver to you on the morning.

I got it, but some new error now:

```
Bashirs-Hackintosh:Desktop bashir$ sudo chown -R root:wheel IntelBluetoothFirmware.kext
Password:
Bashirs-Hackintosh:Desktop bashir$ sudo kextutil -v IntelBluetoothFirmware.kext
Defaulting to kernel file '/System/Library/Kernels/kernel'
Kext with invalid signatured (-2147409652) allowed: <OSKext 0x7fdc05e00250 [0x7fffa18df8e0]> { URL = "file:///Users/bashir/Desktop/IntelBluetoothFirmware.kext/", ID = "com.zxystd.IntelBluetoothFirmware" }
Code Signing Failure: code signature is invalid
/Users/bashir/Desktop/IntelBluetoothFirmware.kext appears to be loadable (not including linkage for on-disk libraries).
Loading /Users/bashir/Desktop/IntelBluetoothFirmware.kext.
Disabling KextAudit: SIP is off
KextAudit initialized: audit=F
(kernel) kxld[com.zxystd.IntelBluetoothFirmware]: The super class vtable '__ZTV9IOService' for vtable '__ZTV22IntelBluetoothFirmware' is out of date. Make sure your kext has been built against the correct headers.
(kernel) Can't load kext com.zxystd.IntelBluetoothFirmware - link failed.
(kernel) Failed to load executable for kext com.zxystd.IntelBluetoothFirmware.
(kernel) Kext com.zxystd.IntelBluetoothFirmware failed to load (0xdc008016).
(kernel) Failed to load kext com.zxystd.IntelBluetoothFirmware (error 0xdc008016).
Failed to load /Users/bashir/Desktop/IntelBluetoothFirmware.kext - (libkern/kext) link error.
Failed to load /Users/bashir/Desktop/IntelBluetoothFirmware.kext - (libkern/kext) link error.
Check library declarations for your kext with kextlibs(8).
Bashirs-Hackintosh:Desktop bashir$ kextlibs -undef-symbols IntelBluetoothFirmware.kext
For all architectures:
    com.apple.iokit.IOUSBHostFamily = 1.2
    com.apple.kpi.iokit = 18.7
    com.apple.kpi.libkern = 18.7
    com.apple.kpi.mach = 18.7

For x86_64:
    2 symbols not found in any library kext:
	__ZN9IOService8DispatchE5IORPC
	__ZN15OSMetaClassBase8DispatchE5IORPC
```

sajjad365, 2020-02-20:
@zxystd any feedback? thanks.

zxystd, 2020-02-20:
[归档.zip](https://github.com/zxystd/IntelBluetoothFirmware/files/4230935/default.zip)
try this one.

sajjad365, 2020-02-29:
Nope, still no go in Mojave:

```
Defaulting to kernel file '/System/Library/Kernels/kernel'
Kext with invalid signatured (-2147409652) allowed: <OSKext 0x7f8e33f0d440 [0x7fffa035b8e0]> { URL = "file:///Users/bashir/Downloads/default/IntelBluetoothFirmware.kext/", ID = "com.zxystd.IntelBluetoothFirmware" }
Code Signing Failure: code signature is invalid
/Users/bashir/Downloads/default/IntelBluetoothFirmware.kext appears to be loadable (not including linkage for on-disk libraries).
Loading /Users/bashir/Downloads/default/IntelBluetoothFirmware.kext.
Disabling KextAudit: SIP is off
KextAudit initialized: audit=F
(kernel) kxld[com.zxystd.IntelBluetoothFirmware]: The super class vtable '__ZTV9IOService' for vtable '__ZTV22IntelBluetoothFirmware' is out of date. Make sure your kext has been built against the correct headers.
(kernel) Can't load kext com.zxystd.IntelBluetoothFirmware - link failed.
(kernel) Failed to load executable for kext com.zxystd.IntelBluetoothFirmware.
(kernel) Kext com.zxystd.IntelBluetoothFirmware failed to load (0xdc008016).
(kernel) Failed to load kext com.zxystd.IntelBluetoothFirmware (error 0xdc008016).
Failed to load /Users/bashir/Downloads/default/IntelBluetoothFirmware.kext - (libkern/kext) link error.
Failed to load /Users/bashir/Downloads/default/IntelBluetoothFirmware.kext - (libkern/kext) link error.
Check library declarations for your kext with kextlibs(8).
Bashirs-Hackintosh:default bashir$ kextlibs -undef-symbols IntelBluetoothFirmware.kext
For all architectures:
    com.apple.iokit.IOUSBHostFamily = 1.2
    com.apple.kpi.iokit = 18.7
    com.apple.kpi.libkern = 18.7
    com.apple.kpi.mach = 18.7

For x86_64:
    2 symbols not found in any library kext:
	__ZN9IOService8DispatchE5IORPC
	__ZN15OSMetaClassBase8DispatchE5IORPC
```

yetiddbb, 2020-03-01:
> [归档.zip](https://github.com/zxystd/IntelBluetoothFirmware/files/4230935/default.zip)
> try this one.
sh : kextstat| grep IntelBlue
   92    0 0xffffff7f832f9000 0xd21000   0xd21000   com.zxystd.IntelBluetoothFirmware (1.0.2) AC350FBA-0A28-301D-A5DB-2C2BA350A6C0 <27 8 6 5 3 1>

But System Information BT is null.
sh :log show --last boot | grep IntelFi
no Information

sajjad365, 2020-03-06:
I give up, will just go with USB BT adapter.

williambj1, 2020-03-06:
Your kernel cache seems to be broken.

sajjad365, 2020-03-06:
> Your kernel cache seems to be broken.

I don't know what that means without doing some research. And if that's the case, I wonder what else is being affected by it. Any guidance would be appreciated.

williambj1, 2020-03-06:
You should probably do research yourself instead of asking me to do it for you. Anyway.

Try to rebuild kernel cache by running `sudo kextcache -i/` in terminal.
If this still doesn't work, consider reinstalling macOS because other kexts may be in trouble as well.

drdaz, 2020-04-18:
I ran into this too. I deleted the Bluetooth plist from Library and rebooted to fix.

closed 2020-07-13 by williambj1

wishie, 2020-08-15:
I have deleted all bluetooth related plists I could find, even removed the IntelBluetooth and IntelFirmware kexts (to go back to using my previous BT card) and it still does not work.. even with NO BLUETOOTH card in the machine, macOS insists I have an Ericsson BT adapter with a MAC of 00-00-00-00-00-00

Menchen, 2020-08-15:
@wishie Check your usb map to disable intel bt..