OpenIntelWireless/IntelBluetoothFirmware #55: Newer firmware of BT may cause "IOKitWaitQuiet() timed out"

o00nmq, 2020-02-19:
Dell Inspiron 7590 with AX200

While I am using Windows, the Windows Update has upgraded BT's firmware to version 10.x automatically. It cause macOS with intelbluetoothfirmware.kext booting up very slow. And showed "IOKitWaitQuiet() timed out" in log and `sudo kextcache -i /`. The BT disappeared as well.
By downgrade BT's driver in Windows, it will downgrade firmware as well (version 9.x). Then BT get functional in macOS.
Though intelbluetoothfirmware.kext has been load in `sudo kextstat`, but i could not find any log with "intelfirmware".

zxystd, 2020-02-20:
can you provide the log? it can help me to see which firmware and test why it cause this problem.

o00nmq, 2020-02-22:
I can't found any log by searching "intelfirmware" in log

williambj1, 2020-07-13:
If this issue still exists in the latest version v1.1.1, feel free to reopen.

closed 2020-07-13 by williambj1