OpenIntelWireless/IntelBluetoothFirmware #56: Could not load Catalina 10.15.3

dracmic, 2020-02-22:
Hello, 

I cannot load and the below is what I get. Could you help me out?

 kernel: Notice - new kext com.zxystd.IntelBluetoothInjector, v2.5 matches prelinked kext but can't determine if executables are the same (no UUIDs).
2020-02-22 15:38:59.776863+0200 0x25c      Error       0x0                  115    3    bluetoothd: (IOBluetooth) [com.apple.bluetooth:InteropManager] Apple80211GetIfListCopy returns error -3903
2020-02-22 15:39:19.777529+0200 0x25c      Error       0x0                  115    3    bluetoothd: (IOBluetooth) [com.apple.bluetooth:InteropManager] Apple80211GetIfListCopy returns error -3903
2020-02-22 15:41:04.206706+0200 0x237      Error       0x0                  62     0    kextd: (IOKit) [com.apple.kext:kextlog] Kext with invalid signature (-2147409652) allowed: <OSKext 0x6000020fee80 [0x7fff8eddf690]> { URL = "file:///Library/Extensions/IntelBluetoothFirmware.kext/", ID = "com.zxystd.IntelBluetoothFirmware" }
2020-02-22 15:41:04.224414+0200 0x237      Error       0x0                  62     0    kextd: (IOKit) [com.apple.kext:kextlog] /Library/Extensions/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOBluetoothFamily.
2020-02-22 15:41:04.237536+0200 0x237      Error       0x0                  62     0    kextd: (IOKit) [com.apple.kext:kextlog] /Library/Extensions/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOUSBHostFamily.
2020-02-22 15:41:04.248968+0200 0x237      Error       0x0                  62     0    kextd: (IOKit) [com.apple.kext:kextlog] com.zxystd.IntelBluetoothFirmware's dependencies failed security checks; failing.

zxystd, 2020-02-22:
Disable SIP. Also, type command "kextlibs -undef-symbols IntelBluetoothFirmware.kext" to see your dependencies version.

trevjonez, 2020-02-28:
I am seeing the same thing in my logs with SIP using 0x3E7.

trevjonez, 2020-02-28:
these worked for me: https://github.com/zxystd/IntelBluetoothFirmware/issues/54#issuecomment-589025073

zxystd, 2020-04-07:
try the latest one, fix dependencies issue.

trevjonez, 2020-04-07:
I've been on https://github.com/zxystd/IntelBluetoothFirmware/releases/tag/1.0.3 for a few days, with 10.15.4 and it has been more stable than the previous one i was using. Still hangs up or randomly disconnects once or twice a day. but better than the last.

williambj1, 2020-07-13:
Glad that the original issue was solved.

closed 2020-07-13 by williambj1