OpenIntelWireless/IntelBluetoothFirmware #58: Bluetooth headphones stuttering / skipping | Catalina 10.15

acoutts, 2020-02-23:
Driver works great! Loads up and I can connect my devices. It works with my BT 5.0 device (Intel 9260): https://www.amazon.com/gp/product/B07DMDZ888/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

But I cannot fix skipping/stuttering issues:
- RSSI does not report on this driver, but using a genuine Apple Broadcom BCM943602CDP I found my RSSI to be around -45. Very acceptable signal strength.
- Bluetooth headphones will stutter every few seconds and it seems very random.
- I tried these settings but they have no noticeable effect:
```
defaults write com.apple.BluetoothAudioAgent "Apple Bitpool Max (editable)" 80
defaults write com.apple.BluetoothAudioAgent "Apple Bitpool Min (editable)" 80 
defaults write com.apple.BluetoothAudioAgent "Apple Initial Bitpool (editable)" 80
defaults write com.apple.BluetoothAudioAgent "Apple Initial Bitpool Min (editable)" 80
defaults write com.apple.BluetoothAudioAgent "Negotiated Bitpool" 80
defaults write com.apple.BluetoothAudioAgent "Negotiated Bitpool Max" 80
defaults write com.apple.BluetoothAudioAgent "Negotiated Bitpool Min" 80
```
- I have mapped my USB ports and am certain the port the BT device is using is mapped- I have mapped it as type 255 (internal header).

The same headphones do not skip/stutter in my Macbook Pro 16" (2019) which also has BT 5.0. Is this a macOS driver issue or a configuration issue I can resolve?

acoutts, 2020-02-23:
For what it's worth, I have tried the following 3 adapters now and they all have the stuttering issue:
- Intel 9260 - Bluetooth 5.0
- Genuine Apple Broadcom BCM943602CDP from existing iMac - Bluetooth 4.2
- Fenvi T1919 (BCM94360CD) - Bluetooth 4.0

The stuttering is slightly better in the 4.2 and 5.0 cards, and the range is improved, but it's still not prefect like my Macbook. There are no issues in my Macbook so it must be something I can fix.

tikscr, 2020-02-25:
I can confirm this issue happening very rarely and random ( i've had like 2 or 3 times within first day of testing) on combination AC-9560 (0xaaa, bt 5.0) & Airpods 2nd gen.

nvollmar, 2020-02-25:
Do you have inference nearby? Bluetooth and Wifi use the same frequency spectrum.

I suspect this is an antenna/radio quality issue. My mouse was also stuttering when transfering files over Wifi on my MacBook nearby, I had to switch to the 5GHz network to stop it. I guess the MacBooks antennas and radio can handle it better.

acoutts, 2020-02-25:
This is good feedback. I've tried a number of antennas so far. This is the antenna and card I'm using now:
https://www.amazon.com/dp/B07DMDZ888/ref=cm_sw_r_tw_apa_i_rkqvEbHBJRTRH

BT 5.0 seemed to help a lot. In 4.0 and 4.2 the airpods would completely die when chopping out and I'd have to reconnect them to get audio working again. With the BT 5.0 card there is chopping occasionally but at least they don't completely die anymore.

williambj1, 2020-07-13:
This kext is only a firmware uploader and is not responsible for audio related bugs. In fact, those parts are handled by native macos kexts. Sorry for the inconvenience but we really can't do anything to solve this.

closed 2020-07-13 by williambj1

itbigsu, 2021-12-27:
I also encountered this problem, but every time I enter the system, re pairing my aerods can directly solve skipping.I still think it's kext's problem

williambj1, 2021-12-27:
@itbigsu If you believe it is a firmware issue, please report it to Intel instead.