OpenIntelWireless/IntelBluetoothFirmware #60: Broadcom Chipset ok, others not

tromPeter, 2020-02-25:
First, thank you for this fine release!!!!!

Here's my issue:
With this kext bluetooth works like a charm on my machine, e.g. apple keyboard, magicmouse and even bluetooth speakers. 
When trying to connect third party mice, microsoft or logitech with a 
Nordic Semiconductor (0x7, 0x5A)
they cannot connect or they're gone after restart.

It's a MSI board with bulid in intel WIFI/BT, 
0x8087, 0x0aaa

[image]

is there anything, i can do to fix this?

Menchen, 2020-02-25:
I think even the real Mac has the problem connecting third party Bluetooth HID device... 

A simple google search like “MX Master MacOS Bluetooth” reveal many problem in real Mac.

tromPeter, 2020-02-25:
But these two bt mice (microsoft and logitech) work on my macbook pro without a problem.

closed 2020-02-28 by tromPeter