OpenIntelWireless/IntelBluetoothFirmware #62: Intel9260AC可以驱动,可以被发现,无法查找到其他设备

silasliu, 2020-02-26:
### 描述
1. log显示蓝牙成功被驱动  
2. 无法搜索到其他设备,包括airpods,蓝牙键盘  
3. 可以被其他设备发现并建立连接,如iphone
### 环境
IntelBluetoothFirmware版本:1.0.2
 网卡型号: Intel9260AC
系统版本: 10.15.3
  供应商ID：	0x8087
  产品ID：	0x0025
### 系统报告-蓝牙信息如下
```
 Apple蓝牙软件版本：	7.0.3f5
  硬件、功能和设置：
  名称：	~~~~~~~
  地址：	5C-87-9C-0A-5F-7C
  支持低功耗蓝牙：	是
  支持“接力”：	否
  支持智能热点：	否
  生产企业：	Intel
  传输：	USB
  固件版本：	9.256 (9.256)
  蓝牙电源：	打开
  可被发现：	打开
  可连接：	是
  自动寻找点：	打开
  远程唤醒：	打开
  供应商ID：	0x8087
  产品ID：	0x0025
  蓝牙核心规范：	5.0 (0x9)
  HCI修正版：	0x100
  LMP版本：	5.0 (0x9)
  LMP子版本：	0x100
  设备类型（主要）：	Computer
  设备类型（完整）：	Mac Desktop
  复合类设备：	0x380104
  设备类（主要）：	0x01
  设备类（次要）：	0x01
  服务类：	0x1C0
  自动寻找键盘：	打开
```
### 日志如下
```
Last login: Thu Feb 27 02:48:00 on ttys000
➜  ~ log show --last boot | grep IntelFirmware
2020-02-27 02:45:44.242421+0800 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver init()
2020-02-27 02:45:44.242430+0800 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Probe()
2020-02-27 02:45:44.242438+0800 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0025
2020-02-27 02:45:44.242484+0800 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Start()
2020-02-27 02:45:44.242582+0800 0x16e      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: setPowerState powerStateOrdinal=1
2020-02-27 02:45:45.755466+0800 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: set configuration to 1
2020-02-27 02:45:45.773149+0800 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-02-27 02:45:45.773153+0800 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-02-27 02:45:45.773155+0800 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-02-27 02:45:45.773158+0800 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了只找到了！！！
2020-02-27 02:45:45.773188+0800 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到中断端点了！！！
2020-02-27 02:45:45.773224+0800 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到Bulk输出端点了！！！
2020-02-27 02:45:45.773249+0800 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到Bulk输入端点了！！！
2020-02-27 02:45:45.773284+0800 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: usb init succeed
2020-02-27 02:45:45.773287+0800 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_VERSION
2020-02-27 02:45:45.778420+0800 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-02-27 02:45:45.780224+0800 0x164      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Firmware revision 0.1 build 233 week 36 2019
2020-02-27 02:45:45.780231+0800 0x164      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: firmware had been download.
2020-02-27 02:45:45.780252+0800 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-02-27 02:45:45.780254+0800 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: End download
2020-02-27 02:45:45.780289+0800 0x17f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Stop()
2020-02-27 03:12:42.315430+0800 0x16e      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: setPowerState powerStateOrdinal=0
2020-02-27 03:12:42.320347+0800 0x16e      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: setPowerState powerStateOrdinal=1
2020-02-27 03:12:42.873906+0800 0x16e      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: setPowerState powerStateOrdinal=0
2020-02-27 03:12:42.878835+0800 0x16e      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: setPowerState powerStateOrdinal=1
```

zxystd, 2020-02-27:
9260有成功的案例，你可以试试其他蓝牙设备

acoutts, 2020-02-28:
I have this 9260 card and it is working for me: https://www.amazon.com/Fenvi-Wireless-AC-2030Mbps-802-11ac-Miracast/dp/B07DMDZ888

closed 2020-03-04 by silasliu