OpenIntelWireless/IntelBluetoothFirmware #66: AC 9560 working, but report "can not proceed response" "operation was aborted"

taina0407, 2020-02-28:
**Description**
I saw some failed logs, just want to report it incase its useful

**Environment**

- Kext Version: 1.0.2
- WiFi Card Model: Intel(R) Wireless-AC 9560
- Product ID: 0x8087 0x0AAA
- macOS Version: 10.15.3 (19D76)

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```
 Apple Bluetooth Software Version:	7.0.3f5
  Hardware, Features, and Settings:
  Name:	*******s iMac
  Address:	0C-DD-24-2F-84-50
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	9.256 (9.256)
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	No
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0AAA
  Bluetooth Core Spec:	5.0 (0x9)
  HCI Revision:	0x100
  LMP Version:	5.0 (0x9)
  LMP Subversion:	0x100
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Desktop
  Composite Class Of Device:	0x380104
  Device Class (Major):	0x01
  Device Class (Minor):	0x01
  Service Class:	0x1C0
  Auto Seek Keyboard:	On
  Services:
  Bluetooth File Transfer:
  Folder other devices can browse:	~/Public
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth File Exchange:
  Folder for accepted items:	~/Downloads
  When other items are accepted:	Save to location
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth Internet Sharing:
  State:	Disabled
  Incoming Serial Ports:
  Bluetooth-Incoming-Port:
  RFCOMM Channel:	3
  Requires Authentication:	No
```

**Logs**

```log
2020-02-29 01:15:36.147494+0700 0x1de      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-02-29 01:15:36.152959+0700 0x15d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 设备固件上传完成
2020-02-29 01:15:36.158698+0700 0x15d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-02-29 01:15:36.164424+0700 0x1de      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-02-29 01:15:36.169853+0700 0x1de      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-02-29 01:15:36.175786+0700 0x1de      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-02-29 01:15:36.186145+0700 0x1de      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-02-29 01:15:36.191517+0700 0x1de      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-02-29 01:15:36.196757+0700 0x15d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 咋了，这是咋了？(-536870165) operation was aborted)
2020-02-29 01:15:36.203318+0700 0x1de      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
```

**Kernel Panic Logs**
No Panic (yet)

```

```

**Screenshots**
[image]
[image]

Timothybas, 2020-03-10:
How did you installed can you please guide me to do.(Because im also using the specs you have but not working at my end.)

Thanks in Advice.

tvhung83, 2020-04-17:
@Timothybas you need to fix USB ports first, and then just need to put `IntelBluetoothFirmware.kext` to Bootloader.

tvhung83, 2020-04-24:
@taina0407 just curious, I'm having issue with audio input (microphone). There is no audio in, and when I check Sound pref (System Preferences), every time I open Input tab, it hangs a little bit and then the device disconnects. I grab some logs but not sure if it's helpful, I might open another issue though:

<details>
  <summary>logs</summary>

```error	15:52:55.437592+0700	coreaudiod	Failed to receive incoming SCO data
default	15:52:55.437736+0700	coreaudiod	stopIO activeIO:1 relatedDevice activeIO:0 (input)
default	15:52:55.437840+0700	coreaudiod	stopIO:0 activeIO:0 (input)
default	15:52:55.437897+0700	coreaudiod	startIO returns:e00002d6 (activeIO:0)
default	15:52:55.439682+0700	bluetoothaudiod	XPC: BTStopIO for SBH24
default	15:52:55.439825+0700	bluetoothaudiod	Stop IO - set input activeBeforeSleep = NO
default	15:52:55.440010+0700	bluetoothaudiod	Enqueue - BTStopIO for SBH24
default	15:52:55.440207+0700	bluetoothaudiod	Tried to dequeue StopIO but 1 seconds hasn't passed (-0.000231), continue
default	15:52:56.440953+0700	bluetoothaudiod	Current command: BTStopIO
default	15:52:56.441015+0700	bluetoothaudiod	Starting XPC timer for 15.000000 seconds
default	15:52:56.441077+0700	bluetoothaudiod	ProcessXPC: stopIO input
default	15:52:56.441123+0700	bluetoothaudiod	[updateAudioState] sampleRate:8000.000000 isSystemSleeping:0 currentProtocol:HFP state:Active inputActive:0 outputActive:0
default	15:52:56.441268+0700	bluetoothaudiod	SBH24 has protocol A2DP in state Connected (pending:None)
default	15:52:56.441317+0700	bluetoothaudiod	Input preferred:HFP state:Active pending:None changing?:0  Output preferred:HFP state:Active pending:None changing?:0
default	15:52:56.441911+0700	bluetoothaudiod	HFP pending None->Connected currentState:Active stateChanging:0 active:0
default	15:52:56.441993+0700	bluetoothaudiod	Stop audio
default	15:52:56.442121+0700	bluetoothaudiod	handsFree:connectingSCO: 0
default	15:52:56.442198+0700	bluetoothaudiod	Disconnect SCO connected:1 (disconnectAfterDisconnectingSCO:0)
default	15:52:56.442626+0700	bluetoothaudiod	Update activity signaled:0
default	15:52:56.443219+0700	bluetoothaudiod	SBH24 has protocol A2DP in state Connected (pending:None)
default	15:52:56.443274+0700	bluetoothaudiod	sendAudioStatus isAudioActive:0
default	15:52:57.235302+0700	coreaudiod	BluetoothHALPlugIn_StartIO: triggered by (null) (0) - INPUT - SBH24
default	15:52:57.235387+0700	coreaudiod	startIO activeIO:0 (input) sampleRate:8000.000000 format:1 mode:0
default	15:52:57.235754+0700	bluetoothaudiod	XPC: BTStartIO for SBH24
default	15:52:57.235821+0700	bluetoothaudiod	Enqueue - BTStartIO for SBH24
default	15:52:57.235841+0700	bluetoothaudiod	ProcessNextXPC already handling BTStopIO
default	15:52:57.367120+0700	coreaudiod	1 second delay - has input been reactivated? 0
default	15:52:57.367199+0700	coreaudiod	Request device configuration change: Sample rate to 48000
default	15:52:57.367506+0700	coreaudiod	setSampleRate 8000.000000->48000.000000
default	15:52:57.367614+0700	coreaudiod	Number of channels:2 frames/buffer:10240 deviceLatency:0.010000 codecLatency:0.002000 safetyOffset:768
default	15:52:57.379379+0700	bluetoothaudiod	At 15:52:57.379 - Set Absolute Volume: 100 - SENDING - (DIRECT)
default	15:53:00.930078+0700	coreaudiod	1 second delay - has input been reactivated? 0
default	15:53:06.492072+0700	bluetoothd	   EVENT:  ModeChange
default	15:53:06.492290+0700	bluetoothd	Apple HID device count: 0
default	15:53:06.629534+0700	bluetoothd	updateAppleHIDConnectionIntervals - connectedAppleDevices 0
default	15:53:06.629566+0700	bluetoothd	updateAppleHIDConnectionIntervals - twoSniffAttemptDeviceCount 0
default	15:53:06.629585+0700	bluetoothd	updateAppleHIDConnectionIntervals - connectedSCODeviceCount 1
error	15:53:07.295972+0700	bluetoothd	Apple80211GetIfListCopy returns error -3903
default	15:53:08.385092+0700	kernel	Sandbox: 1 duplicate report for bluetoothd deny(1) mach-lookup com.apple.server.bluetooth
error	15:53:11.441970+0700	bluetoothaudiod	XPC Timeout current command:BTStopIO
default	15:53:11.442224+0700	bluetoothaudiod	Starting shutdown for SBH24. Current command:BTStopIO
default	15:53:11.442297+0700	bluetoothaudiod	BTStopIO complete. New command (null)
default	15:53:11.442357+0700	bluetoothaudiod	Current command: BTStartIO
default	15:53:11.442395+0700	bluetoothaudiod	Starting XPC timer for 15.000000 seconds
default	15:53:11.442449+0700	bluetoothaudiod	ProcessXPC: startIO input sampleRate:8000.000000 (prev input:1 output:0)
default	15:53:11.442493+0700	bluetoothaudiod	[updateAudioState] sampleRate:8000.000000 isSystemSleeping:0 currentProtocol:HFP state:Active inputActive:1 outputActive:0
default	15:53:11.442619+0700	bluetoothaudiod	SBH24 has protocol A2DP in state Connected (pending:None)
default	15:53:11.442668+0700	bluetoothaudiod	Input preferred:HFP state:Active pending:Connected changing?:0  Output preferred:HFP state:Active pending:Connected changing?:0
default	15:53:11.443288+0700	bluetoothaudiod	Set current protocol HFP->HFP
default	15:53:11.443345+0700	bluetoothaudiod	HFP pending Connected->Active currentState:Active stateChanging:0 active:1
default	15:53:11.443392+0700	bluetoothaudiod	HFP:Active currentCommand:BTStartIO
default	15:53:11.443455+0700	bluetoothaudiod	SendXPCResponse:0 replyMessage:1 xpcConnection:1 to method:BTStartIO
default	15:53:11.443767+0700	coreaudiod	Request device configuration change: SCO
[... cut ...]
default	15:53:14.792346+0700	mediaremoted	[AVRoutingServer] Setting (null) to application picked routes for client <MRDMediaRemoteClient 0x7ff9b4116270, bundleIdentifier = bluetoothaudiod, pid = 955>
default	15:53:19.469557+0700	coreaudiod	1 second delay - has input been reactivated? 0
default	15:53:19.469640+0700	coreaudiod	Request device configuration change: Sample rate to 48000
error	15:53:22.907430+0700	kernel	Sandbox: bluetoothd(169) deny(1) mach-lookup com.apple.server.bluetooth
error	15:53:27.298202+0700	bluetoothd	Apple80211GetIfListCopy returns error -3903
error	15:53:47.302189+0700	bluetoothd	Apple80211GetIfListCopy returns error -3903
```

</details>

williambj1, 2020-07-17:
Please reopen of this issue is still reproduceable with the latest version.

closed 2020-07-17 by williambj1