OpenIntelWireless/IntelBluetoothFirmware #68: Upon waking from standby, bluetooth doesn't wake up until the bluetooth power is cycled off/on

acoutts, 2020-03-05:
**Description**
After the computer sits for a long time (multiple days) and I come back to it, the bluetooth mouse / keyboard do not re-connect until I turn the bluetooth module off and back on. I have disabled sleep so it never goes into sleep mode but I suppose it would be in standby mode during this time.

Everything comes back to life immediately once I turn bluetooth off and back on.

The issue only happens after the system sits for a very extended amount of time. If I leave it just a few hours or overnight, the bluetooth wakes up fine.

**Environment**

- Kext Version: 1.0.2
- WiFi Card Model: AC9260
- Product ID: 0x0025
- macOS Version: 10.15.3

**Info in System Report - Bluetooth**

```
 Apple Bluetooth Software Version:	7.0.3f5
  Hardware, Features, and Settings:
  Name:	xff0c-h
  Address:	A4-C3-F0-85-55-88
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	9.256 (9.256)
  Bluetooth Power:	Off
  Discoverable:	On
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0025
  Bluetooth Core Spec:	5.0 (0x9)
  HCI Revision:	0x100
  LMP Version:	5.0 (0x9)
  LMP Subversion:	0x100
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Desktop
  Composite Class Of Device:	0x380104
  Device Class (Major):	0x01
  Device Class (Minor):	0x01
  Service Class:	0x1C0
  Auto Seek Keyboard:	On

```

**Logs**
```log
No logs found
```

**Kernel Panic Logs**

```
No kernel panics
```

**Screenshots**
N/A

influenist, 2020-03-19:
Same issue here.

uparix, 2020-08-20:
Same for me

zxystd, 2020-10-30:
Have you try the latest release and if it is fixed?

closed 2021-07-04 by zxystd