OpenIntelWireless/IntelBluetoothFirmware #70: 0x8087, 0x0aaa Z390 master成功使用，非常感谢

sxiaojian88, 2020-03-10:
支持airdrop
支持蓝牙耳机，支持开启aptx
handoff还没有试成功

感谢，解决了我电脑蓝牙断断续续困扰了我很久的问题

sxiaojian88, 2020-03-10:
但是麦克风 不能开启

Timothybas, 2020-03-10:
I am using the same z390 0x8087, 0x0AAA but not working.

Throws me this error

```

iMac:~ TestMac$ log show --last boot | grep IntelB
2020-03-10 14:32:43.159422+0530 0x172      Default     0x0                  0      0    kernel: Not entitled to link kext 'com.zxystd.IntelBluetoothFirmware'
2020-03-10 14:32:43.162051+0530 0x172      Default     0x0                  0      0    kernel: Failed to load executable for kext com.zxystd.IntelBluetoothFirmware.
2020-03-10 14:32:43.162054+0530 0x172      Default     0x0                  0      0    kernel: Kext com.zxystd.IntelBluetoothFirmware failed to load (0xdc008004).
2020-03-10 14:32:43.162057+0530 0x172      Default     0x0                  0      0    kernel: Failed to load kext com.zxystd.IntelBluetoothFirmware (error 0xdc008004).
2020-03-10 14:32:43.162928+0530 0x172      Default     0x0                  0      0    kernel: Couldn't alloc class "IntelBluetoothFirmware"
```

sxiaojian88, 2020-03-11:
第二天蓝牙驱动丢了，有什么办法吗？
Bluetooth:Not Avaliable

Timothybas, 2020-03-12:
@sxiaojian88 finally i get this done. now its deducting BT but its not listing all my BT devices and i had another issue now it shows BT not available yesterday it was worked fine.

sxiaojian88, 2020-03-12:
the motherboard bios config "wifi" was disabled can cause Bluetooth Not Avaliable  @Timothybas

Timothybas, 2020-03-12:
@sxiaojian88 i just checked it was enabled in BIOS

Timothybas, 2020-03-12:
```
USB 3.1 Bus:

  Host Controller Driver:	AppleIntelCNLUSBXHCI
  PCI Device ID:	0xa36d 
  PCI Revision ID:	0x0010 
  PCI Vendor ID:	0x8086 

Bluetooth HCI:

  Product ID:	0x0aaa
  Vendor ID:	0x8087  (Intel Corporation)
  Version:	0.02
  Speed:	Up to 12 Mb/s
  Location ID:	0x14e00000 / 3
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0

```

Timothybas, 2020-03-12:
But in System information -> BT ->No information found.

xs9627, 2020-03-17:
同样的问题，Z390 I AORUS PRO WIFI主板。usb端口内建，使用麦克风以后蓝牙设备短暂丢失2~3秒，在IORegist里可以看到USB端口下的Bluetooth USB Host Controller重新加载了。后面会尝试抓一下日志。
比较奇怪的是用BCM943602CS蓝牙也遇到同样的问题，我怀疑和USB端口定制有关，各种方法尝试都没效果。
远景帖子链接，
http://bbs.pcbeta.com/viewthread-1842178-1-1.html

Timothybas, 2020-03-21:
@sxiaojian88 can you share your System report->USB screenshot?

sxiaojian88, 2020-03-23:
@Timothybas 
USB 3.1 Bus:

  Host Controller Driver:	AppleIntelCNLUSBXHCI
  PCI Device ID:	0xa36d 
  PCI Revision ID:	0x0010 
  PCI Vendor ID:	0x8086 

USB5744:

  Product ID:	0x5744
  Vendor ID:	0x0424  (SMSC)
  Version:	2.86
  Speed:	Up to 5 Gb/s
  Manufacturer:	Microchip Tech
  Location ID:	0x14300000 / 5
  Current Available (mA):	900
  Current Required (mA):	0
  Extra Operating Current (mA):	0

Bluetooth HCI:

  Product ID:	0x0aaa
  Vendor ID:	0x8087  (Intel Corporation)
  Version:	0.02
  Speed:	Up to 12 Mb/s
  Location ID:	0x14e00000 / 6
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0

USB2.0 Hub:

  Product ID:	0x0608
  Vendor ID:	0x05e3  (Genesys Logic, Inc.)
  Version:	85.36
  Speed:	Up to 480 Mb/s
  Location ID:	0x14d00000 / 3
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0

Timothybas, 2020-03-23:
@sxiaojian88 Thanks Finally BLE working now seems its not listing any devices near to me.

Timothybas, 2020-03-23:
[image]

Timothybas, 2020-03-23:
[image]

xs9627, 2020-03-23:
@Timothybas , you install the antenna, right?

sxiaojian88, 2020-03-23:
@Timothybas I have the same situation when antenna is not installed.

Timothybas, 2020-03-24:
@zxystd @Timothybas No i have not installed yet. let me try to install the antenna and try. Thanks

Timothybas, 2020-03-24:
@xs9627 @sxiaojian88 As you saw on the screenshot Ble on/off also not working (disabled) as I know IntelBluetoothInjector.kext needs to be installed to enable on/off switch but in my case when I installed that kext file Bluetooth becomes not available.

Thanks in Advance .

williambj1, 2020-03-24:
A codeless kext has nothing to do with basic functionality. It makes no sense.

Timothybas, 2020-03-24:
Yes, it just having the **plist** file and its used to inject the Intel Device by its vendor & device id in Decimal format like this
```

<key>9560</key>
		<dict>
			<key>CFBundleIdentifier</key>
			<string>com.apple.iokit.BroadcomBluetoothHostControllerUSBTransport</string>
			<key>IOClass</key>
			<string>BroadcomBluetoothHostControllerUSBTransport</string>
			<key>IOProbeScore</key>
			<integer>3000</integer>
			<key>IOProviderClass</key>
			<string>IOUSBHostDevice</string>
			<key>idProduct</key>
			<integer>2730</integer>
			<key>idVendor</key>
			<integer>32903</integer>
		</dict>
```

But the fact is when I install this kext in L/E or C/K/O folder and rebuild catch Bluetooth stops working and says **Bluetooth not available** .

williambj1, 2020-03-24:
Probably some issue related to your personal setup/config, and I'm afraid you have to deal with it yourself (try reinstalling macOS, try OpenCore etc.), this codeless kext works fine for me and others.

Timothybas, 2020-03-24:
@williambj1 I just reinstalled completely on Gigabyte Z390 Arous Pro WiFi 

Getting this issue 

```

iMac:Extensions timothy$ sudo kextutil -v IntelBluetoothInjector.kext/
Defaulting to kernel file '/System/Library/Kernels/kernel'
Kext with invalid signature (-67030) allowed: <OSKext 0x7f96d5c167c0 [0x7fff94e4b690]> { URL = "file:///Library/Extensions/IntelBluetoothInjector.kext/", ID = "com.zxystd.IntelBluetoothInjector" }
Code Signing Failure: code signature is invalid
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        3165ac
        3168
        ax200
        726x
        0026
        8265
        926x
        9560

Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        3165ac
        3168
        ax200
        726x
        0026
        8265
        926x
        9560

/Library/Extensions/IntelBluetoothInjector.kext appears to be loadable (not including linkage for on-disk libraries).
Loading /Library/Extensions/IntelBluetoothInjector.kext.
Disabling KextAudit: SIP is off
KextAudit initialized: audit=F
/Library/Extensions/IntelBluetoothInjector.kext successfully loaded (or already loaded).
Invalid signature -67030 for kext <OSKext 0x7f96d5c167c0 [0x7fff94e4b690]> { URL = "file:///Library/Extensions/IntelBluetoothInjector.kext/", ID = "com.zxystd.IntelBluetoothInjector" }
iMac:Extensions timothy$ 

```

Timothybas, 2020-03-24:
After loading this dummy kext Bluetooth on/off is enabled and it works until I shutdown. when i unplug and replug to power its stoped working and displaying something like this[image]

acoutts, 2020-04-01:
> After loading this dummy kext Bluetooth on/off is enabled and it works until I shutdown. when i unplug and replug to power its stoped working and displaying something like this[image]

I don't have a solution, but this usually means the firmware is not uploaded to the card. So the card is powered in and plugged in via USB but there's no firmware uploaded in the RAM of the card.

The reason it worked before is because it was injected once (maybe you booted another OS with a compatible driver like Windows or Linux), and it will persist until the computer is turned off for some time. After that, the firmware needs to be re-uploaded to the RAM in the bluetooth card because RAM is volatile memory (clears when power is removed from it). That is what this kext primarily does.

zxystd, 2020-04-01:
Try and test the latest release file? thanks.

Timothybas, 2020-04-02:
Issue Fixed @zxystd Thanks

closed 2020-05-13 by zxystd