OpenIntelWireless/IntelBluetoothFirmware #71: Bluetooth Not Available (Yesterday it was worked fine)

Timothybas, 2020-03-12:
**Description**
I have installed kext yesterday it was fine awesome and worked as expected but thr was some issue like it's not listing my BLE headphones and few Android devices. but today its suddenly stoped working and showing **Bluetooth Not available** but inside System information -> USB its listing my BLE device (Screenshot attached)

**Environment**

- Kext Version: 1.0.2
- WiFi Card Model: Intel AC 9560
- Product ID:	0x0aaa
- Vendor ID:	0x8087  (Intel Corporation)
- macOS Version: 10.15.3 (19D76)

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```
No information found.
```

**Info in System Report - USB**

```
USB 3.1 Bus:

  Host Controller Driver:	AppleIntelCNLUSBXHCI
  PCI Device ID:	0xa36d 
  PCI Revision ID:	0x0010 
  PCI Vendor ID:	0x8086 

Bluetooth HCI:

  Product ID:	0x0aaa
  Vendor ID:	0x8087  (Intel Corporation)
  Version:	0.02
  Speed:	Up to 12 Mb/s
  Location ID:	0x14e00000 / 3
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0
```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

```log

```

**Kernel Panic Logs**
> In case there is a kernel panic, add `keepsyms=1` in boot-args then paste the panic info below between ```

```

```

**Screenshots**
> If applicable, add screenshots to help explain your problem.

Timothybas, 2020-03-12:
[image]

aluveitie, 2020-03-12:
I sometimes get this too, but a reboot usually fixes it

Timothybas, 2020-03-12:
I tried many times but not fixed at my end.

Timothybas, 2020-03-17:
@zxystd can you help me on this.

zxystd, 2020-03-17:
Any logs exists?

Timothybas, 2020-03-17:
No logs when i enter this in terminal 
`log show --last 1d | grep IntelFir`

Timothybas, 2020-03-17:
But in system report-> USB i can see this 
```

USB 3.1 Bus:

  Host Controller Driver:	AppleIntelCNLUSBXHCI
  PCI Device ID:	0xa36d 
  PCI Revision ID:	0x0010 
  PCI Vendor ID:	0x8086 

Bluetooth HCI:

  Product ID:	0x0aaa
  Vendor ID:	0x8087  (Intel Corporation)
  Version:	0.02
  Speed:	Up to 12 Mb/s
  Location ID:	0x14e00000 / 3
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0
```

zxystd, 2020-03-17:
Seems the driver not match this device, maybe you can try to map it as usb2.0 device?

Timothybas, 2020-03-17:
How to do this?
P.S: But when I install this for first time its worked as expected.. but after a few reboots and shutdown, its stopped working.

Timothybas, 2020-03-17:
Another strange issue is when Bluetooth works (First time) its not listing all available devices near to me but its visible to all other devices.

Timothybas, 2020-03-17:
Now I deleted kext from L/E and cleared cache and reinstalled to same location L/E again now its started working but it's not listing any BLE  devices near to me.

Here is the log.
[log.txt](https://github.com/zxystd/IntelBluetoothFirmware/files/4341339/log.txt)

Timothybas, 2020-03-21:
@zxystd When I delete **IntelBluetoothInjector.kext** and I**ntelBluetoothFirmware.kext** from L/E and C/K/O Bluetooth again comes in System preference.

Timothybas, 2020-03-23:
Finally, I fixed the issue by deleting **IntelBluetoothInjector.kext**

closed 2020-03-23 by Timothybas