OpenIntelWireless/IntelBluetoothFirmware #72: logs

SaeedHaidar, 2020-03-14:
- WiFi Card 型号: intel ax1200 160mhz
- macOS 版本: 10.15.2

**系统报告 - 蓝牙**
> 把所有内容都粘贴到下方的两个 ``` 之间

```

```

**日志**
> 终端运行 `log show --last boot | grep IntelFir`，将得到的内容粘贴至 ```log 和 ``` 之间
Last login: Sat Mar 14 08:14:04 on console
abo-haidar@Abo-Haidars-MacBook-Pro ~ % log show --last boot | grep IntelFirmware 
2020-03-14 08:13:52.353025+0100 0x189      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: boot_param=0x24800
2020-03-14 08:13:52.353185+0100 0x189      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: send firmware done
2020-03-14 08:13:52.353189+0100 0x189      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-03-14 08:13:52.353191+0100 0x189      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_RESET
2020-03-14 08:13:52.353366+0100 0x189      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Intel reset succeed
2020-03-14 08:13:53.354411+0100 0x189      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: hci read pipe fail. stop
2020-03-14 08:13:53.354414+0100 0x189      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: End download
2020-03-14 08:13:53.354442+0100 0x189      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Stop()
2020-03-14 08:13:53.354811+0100 0x74       Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver free()
abo-haidar@Abo-Haidars-MacBook-Pro ~ %

zxystd, 2020-03-19:
ax1200?

SaeedHaidar, 2020-04-05:
intel wifi 6 ax200 160mhz Sorry for delay I total forgot

zxystd, 2020-04-07:
my test device is also ax200 and it works fine for me, the log says it cannot open hci pipe, does the card work in Windows? maybe you can check your usb configuration.

SaeedHaidar, 2020-04-07:
Yes it work in Windows I have mapped my USB ports anyway I am going to buy a new WiFi card 
Thanks anyway

williambj1, 2020-07-13:
Closed because the conversation has ended. Feel free to open if necessary.

closed 2020-07-13 by williambj1