OpenIntelWireless/IntelBluetoothFirmware #73: Work fine on Catalina, however it doesn't work on Mojave

quan3c10, 2020-03-17:
**Description**
This kext work fine on my Catalina hackintosh, however, when I reinstall with Mojave using the same EFI, the bluetooth doesn't appear. Tried get log with provided command and get no result. I also tried with the older release got the same result.

**Environment**

- Kext Version: 1.0.2
- WiFi Card Model: AC 9560 - 9560NGW
- Product ID: PD99560NG
- macOS Version: 10.14.6

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```

```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

```log

```

**Kernel Panic Logs**
> In case there is a kernel panic, add `keepsyms=1` in boot-args then paste the panic info below between ```

```

```

**Screenshots**
> If applicable, add screenshots to help explain your problem.

zxystd, 2020-03-17:
Modify the dependencies with your current OS Library version, this link for your reference. https://github.com/zxystd/IntelBluetoothFirmware/issues/54#issuecomment-586577178

quan3c10, 2020-03-17:
I rebuild the kext with new dependencies which found by command "kextlibs -undef-symbols IntelBluetoothFirmware.kext", got the same result. Tried clear kext cache doesn't help.

quan3c10, 2020-03-17:
I got this on Console
[image]

zxystd, 2020-03-17:
still not loaded.
try to manual load by following command

sudo chown -R root:wheel IntelBluetoothFirmware.kext
sudo kextutil -v IntelBluetoothFirmware.kext

and show me the error log.

zxystd, 2020-03-17:
maybe you can download my source code and build with 10.14 sdk.

zxystd, 2020-03-17:
[image]
and dependencies version are like that you have provided, they are your system local library version.

quan3c10, 2020-03-17:
I tried above suggestion and got the same issue, the error log as previous post. The issue is strange, your kext work well on my Catalina, however, it failed on Mojave in the same machine.

zxystd, 2020-03-17:
https://github.com/zxystd/IntelBluetoothFirmware/issues/7#issuecomment-570504331
try this one if your os version is 10.14.6, I will compile another one if I have more spare time.

quan3c10, 2020-03-17:
Then it works now. Thank you very much!
I am going to close this issue with the solution above.

closed 2020-03-17 by quan3c10

renemendoza, 2020-06-22:
> #7 (comment)

Would this work with 10.14.2?

zxystd, 2020-06-22:
yes it can. version 1.0.3