OpenIntelWireless/IntelBluetoothFirmware #74: Dependency Resolution Failures Only incompatible kexts found for these libraries

suheet1, 2020-03-18:
I don't know if I am doing it right or wrong I am new to Hackintosh so sorry for wasting time I tried pasting kext in efi/oc/kext and also added in config.plist but still cannot find any Bluetooth option like others are getting then tried manually add kext getting this error 
I did check on windows under device id my bluetooth devid or say hardware id was 0x8087, 0x0029
so thought of giving a try thank you  

kextutil -nt IntelBluetoothFirmware.kext
Kext with invalid signature (-67050) allowed: <OSKext 0x7fb82ce27be0 [0x7fff93129690]> { URL = "file:///Users/suheet/Downloads/IntelBluetooth/IntelBluetoothFirmware.kext/", ID = "com.zxystd.IntelBluetoothFirmware" }
/Users/suheet/Downloads/IntelBluetooth/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOBluetoothFamily.
/Users/suheet/Downloads/IntelBluetooth/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOUSBHostFamily.
/Users/suheet/Downloads/IntelBluetooth/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOBluetoothFamily.
/Users/suheet/Downloads/IntelBluetooth/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOUSBHostFamily.
/Users/suheet/Downloads/IntelBluetooth/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOBluetoothFamily.
/Users/suheet/Downloads/IntelBluetooth/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOUSBHostFamily.
Code Signing Failure: code signature is invalid
Dependency Resolution Failures: 
    Only incompatible kexts found for these libraries: 
        com.apple.iokit.IOBluetoothFamily
        com.apple.iokit.IOUSBHostFamily

Diagnostics for /Users/suheet/Downloads/IntelBluetooth/IntelBluetoothFirmware.kext:
Dependency Resolution Failures: 
    Only incompatible kexts found for these libraries: 
        com.apple.iokit.IOBluetoothFamily
        com.apple.iokit.IOUSBHostFamily

zxystd, 2020-03-19:
I will fix it these days, just need to change the dependencies in config.plist, also you can do it.

suheet1, 2020-03-19:
> I will fix it these days, just need to change the dependencies in config.plist, also you can do it.

How to do that lil help thank you

suheet1, 2020-03-20:
I tried changing info.plist but after build success in Xcode cannot find target folder

Menchen, 2020-03-20:
Right click- reveals in Finder.

suheet1, 2020-03-20:
> Right click- reveals in Finder.

[image]

suheet1, 2020-03-20:
shows nothing

Menchen, 2020-03-20:
Close all finder windows and right click on the next file, and show in Finder.

suheet1, 2020-03-20:
Sorry for the trouble and noob question I am trying 

its saving it 
/Users/suheet/Library/Developer/Xcode/DerivedData/IntelBluetoothFirmware-bypaokyajquqvpemwwvwefdrbbmq/Build/Products/Debug/

but when I change my info.plist it doesn't create Products folder

Menchen, 2020-03-20:
You change Info.plist that’s inside of final build.
And If you can’t find the Kext file, chances that you didn’t compile it.

suheet1, 2020-03-20:
Done building and pasted kext in oc/kext now what to do ?

Menchen, 2020-03-20:
Edit open core config file to include kext

suheet1, 2020-03-20:
My config file already has entry for kext I did a restart also

Menchen, 2020-03-20:
Does BT show up in System Preferences?

suheet1, 2020-03-20:
nope :(

suheet1, 2020-03-20:
[image]

when I add executable path to IntelBluetoothInjector.kext my system won't boot so I have added nothing there

Menchen, 2020-03-20:
If there's no BT in your System Preferences, the problem is USB port(You disabled the port with some SSDT).

suheet1, 2020-03-20:
Was this Kext for external bluetooth hardware cause mine is built in embedded on Motherboard

Menchen, 2020-03-20:
It doesn't matter, most Intel Bluetooth are embedded in WiFi cards. But BT still uses USB protocol, so event it's embedded it's still USB.

suheet1, 2020-03-21:
Kext with invalid signature (-67050) allowed: <OSKext 0x7fd85842f060 [0x7fff93347690]> { URL = "file:///Users/suheet/Documents/s/IntelBluetoothFirmware.kext/", ID = "com.zxystd.IntelBluetoothFirmware" }
Code Signing Failure: code signature is invalid
Disabling KextAudit: SIP is off
(**kernel) Notice - new kext com.apple.driver.usb.AppleUSBHostPlatformProperties, v1.2 matches prelinked kext but can't determine if executables are the same (no UUIDs).**
Invalid signature -67050 for kext <OSKext 0x7fd85842f060 [0x7fff93347690]> { URL = "file:///Users/suheet/Documents/s/IntelBluetoothFirmware.kext/", ID = "com.zxystd.IntelBluetoothFirmware" }

suheet1, 2020-03-31:
2020-03-31 22:39:53.298215+0530 0x65       Default     0x0                  0      0    kernel: Notice - new kext com.zxystd.IntelBluetoothInjector, v1.0.3 matches prelinked kext but can't determine if executables are the same (no UUIDs).

suheet1, 2020-04-02:
You can close this connected all 3 devices iPad iPhone and AirPods everything seems to work fine after using new updated kext thank you so much appreciate the help and sorry for the trouble

Menchen, 2020-04-02:
@suheet1 Does audio input(mics) work with AirPods?. Also, you can close the issue too.

suheet1, 2020-04-02:
didn't try that just can hear audio output from AirPods

closed 2020-04-02 by suheet1