OpenIntelWireless/IntelBluetoothFirmware #75: Removing personalities from kernel

RocheTechnology, 2020-03-19:
**Description**
> See below log. The Bluetooth preferences pane allows me to toggle on/off (it also did before using this kext), but it won't pair any devices. Installed to Clover kexts location

**Environment**

- Kext Version: 1.0.2
- WiFi Card Model: Intel 0x8087,0x0AA7 (built in bluetooth/wifi on X399 Taichi motherboard)
- Product ID: 0x0AA7
- macOS Version: 10.13.6

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```
Apple Bluetooth Software Version:	6.0.7f10
  Hardware, Features, and Settings:
  Name:	HQ-ML-N2O4HV2H
  Address:	C8-09-A8-A7-78-FC
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	8.4352 (8.4352)
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0AA7
  HCI Version:	4.2 (0x8)
  HCI Revision:	0x1100
  LMP Version:	4.2 (0x8)
  LMP Subversion:	0x1100
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Desktop
  Composite Class Of Device:	0x380104
  Device Class (Major):	0x01
  Device Class (Minor):	0x01
  Service Class:	0x1C0
  Auto Seek Keyboard:	On
  Devices (Paired, Configured, etc.):

```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

```log

<hostname>:~ user$ log show --last boot | grep Intel
2020-03-18 19:44:42.908911-0700 0x2fa      Default     0x0                  0      0    kernel: (IOPlatformPluginFamily) ACPI_SMC_PlatformPlugin::start - waitForService(resourceMatching(AppleIntelCPUPowerManagement) timed out
2020-03-18 19:44:42.908914-0700 0x2fa      Default     0x0                  0      0    kernel: (IOPlatformPluginFamily) ACPI_SMC_PlatformPlugin::start - waitForService(resourceMatching(AppleIntelCPUPowerManagement) timed out
2020-03-18 19:44:42.911541-0700 0x2fa      Default     0x0                  44     0    kextd: Kext id com.zxystd.IntelBluetoothFirmware not found; removing personalities from kernel.

```

zxystd, 2020-04-06:
can you try the latest relese?

williambj1, 2020-07-13:
Closed since this issue is no longer active for quite a long time, feel free to reopen.

closed 2020-07-13 by williambj1