OpenIntelWireless/IntelBluetoothFirmware #77: 0x8087, 0x0aaa LG Gram 17 17z990 problem

rdmitry0911, 2020-03-26:
Hi there, I'm trying to use your IntelBluetoothFirmware.kext and have a problem using it. macos 10.14.5 shows the bt device, however no other devices can be found around. On the other hand after I run openwrt virtual machine in vmware player with the bt device being passthrough there, the device become fully initialized and after openwrt shutdown and bt device become visible in macos again, I can connect anything to it. It is fully operational.

Menchen, 2020-03-26:
Log?

rdmitry0911, 2020-03-26:
log show --last boot | grep IntelFirmware gives nothing

Menchen, 2020-03-26:
Then the Kext is not loaded or no bt is detected

rdmitry0911, 2020-03-26:
what might be the reason?

rdmitry0911, 2020-03-26:
> Then the Kext is not loaded or no bt is detected
bdmesg shows:

```
5:262  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\OEM\17Z990\kexts\Other
5:262  0:000  ->Extra kext: EFI\CLOVER\OEM\17Z990\kexts\Other\IntelBluetoothFirmware.kext (v.1.0.2)
5:670  0:408  ->Extra kext: EFI\CLOVER\OEM\17Z990\kexts\Other\WhateverGreen.kext (v.1.3.0)
5:705  0:034  ->Extra kext: EFI\CLOVER\OEM\17Z990\kexts\Other\VoodooPS2Controller.kext (v.1.9.1)
5:719  0:013      |-- PlugIn kext: EFI\CLOVER\OEM\17Z990\kexts\Other\VoodooPS2Controller.kext\Contents\PlugIns\VoodooPS2Keyboard.kext (v.1.9.1)
5:732  0:013  ->Extra kext: EFI\CLOVER\OEM\17Z990\kexts\Other\VoodooI2CHID.kext (v.1)
5:757  0:025  ->Extra kext: EFI\CLOVER\OEM\17Z990\kexts\Other\VoodooI2C.kext (v.2.2)
5:783  0:025      |-- PlugIn kext: EFI\CLOVER\OEM\17Z990\kexts\Other\VoodooI2C.kext\Contents\PlugIns\VoodooI2CServices.kext (v.1)
5:792  0:009      |-- PlugIn kext: EFI\CLOVER\OEM\17Z990\kexts\Other\VoodooI2C.kext\Contents\PlugIns\VoodooGPIO.kext (v.1.1)
5:806  0:013  ->Extra kext: EFI\CLOVER\OEM\17Z990\kexts\Other\VirtualSMC.kext (v.1.0.3)
5:821  0:015  ->Extra kext: EFI\CLOVER\OEM\17Z990\kexts\Other\USB-LG-Gram-17z990.kext (v.0.7.1)
5:836  0:014  ->Extra kext: EFI\CLOVER\OEM\17Z990\kexts\Other\SATA-unsupported.kext (v.0.9.2)
5:841  0:005  ->Extra kext: EFI\CLOVER\OEM\17Z990\kexts\Other\Lilu.kext (v.1.3.6)
5:851  0:009  ->Extra kext: EFI\CLOVER\OEM\17Z990\kexts\Other\HibernationFixup.kext (v.1.2.9)
5:864  0:012  ->Extra kext: EFI\CLOVER\OEM\17Z990\kexts\Other\AppleALC.kext (v.1.3.9)
6:424  0:560  ->Extra kext: EFI\CLOVER\OEM\17Z990\kexts\Other\ACPIBatteryManager.kext (v.1.90.1)
6:442  0:018  Preparing kexts injection for arch=x86_64 from \EFI\CLOVER\kexts\Off
6:442  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\OEM\17Z990\kexts\10
6:442  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\OEM\17Z990\kexts\10_normal
6:442  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\OEM\17Z990\kexts\10.14
6:442  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\OEM\17Z990\kexts\10.14_normal
6:442  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\OEM\17Z990\kexts\10.14.5
6:442  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\OEM\17Z990\kexts\10.14.5_normal

```
macos detects bt device quite well. 
ioreg -c IOBluetoothHCIController shows 
```
      +-o IOBluetoothHCIController  <class IOBluetoothHCIController, id 0x10000041a, registered, matched, active, busy 0 (5 ms), retain 28>
      | | {
      | |   "ClassOfDevice" = 3670284
      | |   "SkipAutoFWUpdate" = No
      | |   "BootFromROM" = No
      | |   "BluetoothDeviceAddress" = "D0-C6-37-8B-C1-ED"
      | |   "IOMatchCategory" = "IOBluetoothHCIController"
      | |   "SupportBTRS" = No
      | |   "ActiveControllerInfo" = {"ActiveControllerProductID"=2730,"ActiveControllerActiveConnections"=0,"ActiveControllerLocationID"=341835776,"ActiveControllerVendorID"=32903,"ActiveController$
      | |   "RemoteWakeEnabled" = Yes
      | |   "BluetoothLocalName" = "macos-LG17 (2)"
      | |   "Built-In" = Yes
      | |   "d8-30-62-40-02-77" = {"VendorIDSource"=2,"LinkSupervisionTimeout"=8000,"VendorID"=1452,"CountryCode"=0,"HIDVirtualDevice"=No,"Product"="Dmitri Baliakhov?s Mouse","VersionNumber"=132,"Qo$
      | |   "TransportType" = "USB"
      | |   "HCIRevision" = 256
      | |   "DefaultIdleTime" = 8356
      | |   "IOProbeScore" = 0
      | |   "IOClass" = "IOBluetoothHCIController"
      | |   "SupportDPLE" = Yes
      | |   "SupportWoBT" = No
      | |   "IOGeneralInterest" = "IOCommand is not serializable"
      | |   "IOReportLegend" = ({"IOReportChannels"=((4779480634386507122,17180000258,"Bluetooth Internal Power State")),"IOReportGroupName"="Bluetooth","IOReportChannelInfo"={"IOReportChannelUnit"=$
      | |   "SupportBTPD" = No
      | |   "HCIVersion" = 9
      | |   "ConfigState" = 0
      | |   "SupportBTLP" = No
      | |   "CFBundleIdentifier" = "com.apple.iokit.IOBluetoothFamily"
      | |   "AllowedIncomingRFCOMMChannels" = ({"ChannelIsSecure"=0,"ChannelNumber"=4},{"ChannelIsSecure"=0,"ChannelNumber"=1},{"ChannelIsSecure"=0,"ChannelNumber"=3},{"ChannelIsSecure"=0,"ChannelNu$
      | |   "BluetoothTransportConnected" = Yes
      | |   "idVendor" = 32903
      | |   "HCIControllerPowerIsOn" = Yes
      | |   "HCIResetHappenedDuringWake" = No
      | |   "SupportPowerOff" = No
      | |   "SupportConcurrentCreateConnection" = No
      | |   "20-73-00-35-b3-58" = {"VendorIDSource"=2,"LinkSupervisionTimeout"=8000,"VendorID"=1256,"CountryCode"=33,"HIDVirtualDevice"=No,"Product"="Bluetooth 3.0 Keyboard","VersionNumber"=283,"QoS$
      | |   "SupportNewIdlePolicy" = No
      | |   "IOResourceMatch" = "IOKit"
      | |   "FeatureFlags" = 6
      | |   "HCISupportedFeatures" = <877bffdbfe0ffebf>
      | |   "IOReportLegendPublic" = Yes
      | |   "ManufacturerName" = 2
      | |   "LMPSubversion" = 256
      | |   "BluetoothVendor" = "Unknown"
      | |   "SupportBTPU" = No
      | |   "PowerOptions" = 0
      | |   "BluetoothDeviceAddressData" = <d0c6378bc1ed>
      | |   "ActiveBluetoothControllerVendor" = "Apple USB"
      | |   "FirmwareVersion" = 0
      | |   "SupportDeepIdle" = No
      | |   "IOProviderClass" = "IOResources"
      | |   "SupportLighthouse" = No
      | |   "LocalDeviceScanState" = 2
      | |   "SupportBTRB" = No
      | |   "LMPVersion" = 9
      | |   "LMPLoggingEnabled" = No
      | |   "idProduct" = 2730
      | |   "FirmwareVersionString" = "v0 c256"
      | |   "AllowedIncomingL2CAPChannels" = ({"PSM"=3},{"PSM"=1},{"PSM"=31},{"PSM"=17},{"PSM"=19},{"PSM"=23},{"PSM"=25},{"PSM"=15})
      | |   "LocationID" = 341835776
      | | }

```

rdmitry0911, 2020-03-26:
> Then the Kext is not loaded or no bt is detected

Kext is not loaded. You right 

bash-3.2# kextutil -v IntelBluetoothFirmware.kext
Defaulting to kernel file '/System/Library/Kernels/kernel'
Kext with invalid signatured (-2147409652) allowed: <OSKext 0x7fb25462d880 [0x7fff8d5468e0]> { URL = "file:///Users/dima/Desktop/IntelBluetoothFirmware.kext/", ID = "com.zxystd.IntelBluetoothFirmware" }
/Users/dima/Desktop/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOBluetoothFamily.
/Users/dima/Desktop/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOUSBHostFamily.
/Users/dima/Desktop/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOBluetoothFamily.
/Users/dima/Desktop/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOUSBHostFamily.
/Users/dima/Desktop/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOBluetoothFamily.
/Users/dima/Desktop/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOUSBHostFamily.
Code Signing Failure: code signature is invalid
Dependency Resolution Failures: 
    Only incompatible kexts found for these libraries: 
        com.apple.iokit.IOBluetoothFamily
        com.apple.iokit.IOUSBHostFamily

Diagnostics for /Users/dima/Desktop/IntelBluetoothFirmware.kext:
Dependency Resolution Failures: 
    Only incompatible kexts found for these libraries: 
        com.apple.iokit.IOBluetoothFamily
        com.apple.iokit.IOUSBHostFamily

Menchen, 2020-03-26:
I suppose you know how to edit Info.plist, you only need to change those two value. Use Kextlibs to get them

rdmitry0911, 2020-03-26:
> I suppose you know how to edit Info.plist, you only need to change those two value. Use Kextlibs to get them

I know how to edit, however, I don't know what to edit:
bash-3.2# Kextlibs -undef-symbols IntelBluetoothFirmware.kext
```
For all architectures:
    com.apple.iokit.IOUSBHostFamily = 1.2
    com.apple.kpi.iokit = 18.6
    com.apple.kpi.libkern = 18.6
    com.apple.kpi.mach = 18.6

For x86_64:
    2 symbols not found in any library kext:
	__ZN9IOService8DispatchE5IORPC
	__ZN15OSMetaClassBase8DispatchE5IORPC

```

rdmitry0911, 2020-03-26:
After I changed 

com.apple.iokit.IOUSBHostFamily -> 1.2
com.apple.iokit.IOBluetoothFamily -> 1.2

I got this:

bash-3.2# kextutil -v IntelBluetoothFirmware.kext
```
Defaulting to kernel file '/System/Library/Kernels/kernel'
Kext with invalid signatured (-2147409652) allowed: <OSKext 0x7fafa8c02c40 [0x7fff91b728e0]> { URL = "file:///Users/dima/Desktop/IntelBluetoothFirmware.kext/", ID = "com.zxystd.IntelBluetoothFirmware" }
Code Signing Failure: code signature is invalid
/Users/dima/Desktop/IntelBluetoothFirmware.kext appears to be loadable (not including linkage for on-disk libraries).
Loading /Users/dima/Desktop/IntelBluetoothFirmware.kext.
Disabling KextAudit: SIP is off
KextAudit initialized: audit=F
(kernel) kxld[com.zxystd.IntelBluetoothFirmware]: The super class vtable '__ZTV9IOService' for vtable '__ZTV22IntelBluetoothFirmware' is out of date. Make sure your kext has been built against the correct headers.
(kernel) Can't load kext com.zxystd.IntelBluetoothFirmware - link failed.
(kernel) Failed to load executable for kext com.zxystd.IntelBluetoothFirmware.
(kernel) Kext com.zxystd.IntelBluetoothFirmware failed to load (0xdc008016).
(kernel) Failed to load kext com.zxystd.IntelBluetoothFirmware (error 0xdc008016).
Failed to load /Users/dima/Desktop/IntelBluetoothFirmware.kext - (libkern/kext) link error.
Failed to load /Users/dima/Desktop/IntelBluetoothFirmware.kext - (libkern/kext) link error.
Check library declarations for your kext with kextlibs(8).

```

Menchen, 2020-03-26:
Use `kextstat | grep -i IOBluetoothFamily` to get `IOBluetoothFamily` version in parentheses. Also, it's recommended to build from source as there's a weir link problem.

rdmitry0911, 2020-03-26:
> Use `kextstat | grep -i IOBluetoothFamily` to get `IOBluetoothFamily` version in parentheses. Also, it's recommended to build from source as there's a weir link problem.

```
bash-3.2# kextstat | grep -i IOBluetoothFamily
   95    4 0xffffff7f83289000 0xf9000    0xf9000    com.apple.iokit.IOBluetoothFamily (6.0.12f1) 9A07BB23-7147-3B6B-A5C0-784BDC83D48D <25 18 12 8 7 6 5 3 1>
bash-3.2# 

```

I built from source. The same result. Probably some project settings have to changed to make the kext to be loadable by multiple different kernel versions

rdmitry0911, 2020-03-27:
I made it work. I installed old sdk and build against 10.12 sdk Changed info.plist and now it is good to go in my 10.14

[IntelBluetoothFirmware.kext.zip](https://github.com/zxystd/IntelBluetoothFirmware/files/4394337/IntelBluetoothFirmware.kext.zip)

zxystd, 2020-04-06:
fix with 1.0.3 release.

closed 2020-04-06 by zxystd