OpenIntelWireless/IntelBluetoothFirmware #78: Kext in not secure

subodhkapoor, 2020-03-28:
kextstat | grep -i "IOBluetoothFamily"
  120    0 0xffffff7f8299d000 0xfd000    0xfd000    com.apple.iokit.IOBluetoothFamily (7.0.4f6) 51107598-1753-36B6-B724-EAB1A97BE563 <119 28 18 12 8 7 6 5 3 1>
sshtech@SSHTechs-iMac-Pro ~ % kextlibs
No kext specified.usage: kextlibs [options] kext

use kextlibs -help for a list of options
sshtech@SSHTechs-iMac-Pro ~ % kextlibs /Users/sshtech/Desktop/IntelBluetoothFirmware.kext 
For all architectures:
    com.apple.iokit.IOUSBHostFamily = 1.2
    com.apple.kpi.iokit = 19.4
    com.apple.kpi.libkern = 19.4
    com.apple.kpi.mach = 19.4

For x86_64:
    1 symbol not found in any library kext.

sshtech@SSHTechs-iMac-Pro ~ % sudo kextutil -v /Users/sshtech/Desktop/IntelBluetoothFirmware.kext 
Defaulting to kernel file '/System/Library/Kernels/kernel'
Untrusted kexts are not allowed
Kext with invalid signature (-67062) denied: /Library/StagedExtensions/Users/sshtech/Desktop/97742BD3-8740-4FC5-9E36-01886DBA185C.kext
Bundle (/Users/sshtech/Desktop/IntelBluetoothFirmware.kext) failed to validate, deleting: /Library/StagedExtensions/Users/sshtech/Desktop/97742BD3-8740-4FC5-9E36-01886DBA185C.kext
Unable to stage kext (/Users/sshtech/Desktop/IntelBluetoothFirmware.kext) to secure location.

suheet1, 2020-03-29:
check csrutil status if enabled disable it

subodhkapoor, 2020-04-04:
another issue is going on to support the same problem

closed 2020-04-04 by subodhkapoor