OpenIntelWireless/IntelBluetoothFirmware #80: conflict with chris1111 / Wireless-USB-Adapter

joemulder9, 2020-03-29:
**Description**
>kernel panic when used with chris1111's Tplink wifi usb adapter. cannot boot into system.
**Environment**

- Kext Version: 1.0.2
- WiFi Card Model: TL-WN823Nv3
- Product ID: 
- macOS Version: 10.15.4

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```

```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

```log

```

**Kernel Panic Logs**
> In case there is a kernel panic, add `keepsyms=1` in boot-args then paste the panic info below between ```

```

```

**Screenshots**
> If applicable, add screenshots to help explain your problem.

tvieira20, 2020-03-29:
If you remove the Wi-Fi Adapter can you use the Intel's bluetooth ?

joemulder9, 2020-03-30:
yes, without wifi adapter, bluetooth works. however, bluetooth 4.0 devices doesn't work.

joemulder9, 2020-04-01:
yes without wifi adapter. bluetooth works .but not with bluetooth 4 devices.

zxystd, 2020-04-06:
use the driver download from rtl website, works well for me.

closed 2020-04-06 by zxystd