OpenIntelWireless/IntelBluetoothFirmware #81: Intel 8265NGW

tvieira20, 2020-03-29:
**Description**
After loading kext using Clover (Kext is CLOVER > kexts > Other) system can now detect a Bluetooth Card (it couldn't before and without the kext). When I open the Bluetooth Menu I can't toggle it off/on. Can't also discover devices nearby.

I am using a Ralink USB Wi-Fi Adapter. I saw some related issues with this Kext and Wi-Fi USB Adapters. Not sure if it's causing the issue

**Environment**

- Kext Version: Latest (1.0.2)
- WiFi Card Model: Intel Dual Band Wireless-AC 8265
- Product ID: 0x0A2B
- macOS Version: 10.15.4 [(19E266)]

**Info in System Report - Bluetooth**
  Apple Bluetooth Software Version:	7.0.4f6
  Hardware, Features and Settings:
  Address:	[HIDDEN]
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	8.256 (8.256)
  Bluetooth Power:	On
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0A2B
  Bluetooth Core Spec:	4.2 (0x8)
  HCI Revision:	0x100
  LMP Version:	4.2 (0x8)
  LMP Subversion:	0x100
  Auto Seek Keyboard:	On
  Services:
  Bluetooth File Transfer:
  Folder other devices can browse:	~/Public
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth File Exchange:
  Folder for accepted items:	~/Downloads
  When other items are accepted:	Save to location
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth Internet Sharing:
  State:	Disabled
  Incoming Serial Ports:
  Bluetooth-Incoming-Port:
  RFCOMM Channel:	3
  Requires Authentication:	No

**Logs**
MacBook-Pro:~ tvieira$ log show --last boot | grep IntelFirmware
2020-03-29 20:49:26.225463+0100 0x95f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver init()
2020-03-29 20:49:26.225472+0100 0x95f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Probe()
2020-03-29 20:49:26.225481+0100 0x95f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0A2B
2020-03-29 20:49:26.225535+0100 0x95f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Start()
2020-03-29 20:49:26.225631+0100 0x17d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: setPowerState powerStateOrdinal=1
2020-03-29 20:49:27.727728+0100 0x95f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: set configuration to 1
2020-03-29 20:49:27.728267+0100 0x95f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-03-29 20:49:27.728272+0100 0x95f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-03-29 20:49:27.728275+0100 0x95f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-03-29 20:49:27.728277+0100 0x95f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了只找到了！！！
2020-03-29 20:49:27.728294+0100 0x95f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到中断端点了！！！
2020-03-29 20:49:27.728535+0100 0x95f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到Bulk输出端点了！！！
2020-03-29 20:49:27.728767+0100 0x95f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到Bulk输入端点了！！！
2020-03-29 20:49:27.729020+0100 0x95f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: usb init succeed
2020-03-29 20:49:27.729026+0100 0x95f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_VERSION
2020-03-29 20:49:27.737267+0100 0x95f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-03-29 20:49:27.738251+0100 0x172      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Firmware revision 0.1 build 50 week 12 2019
2020-03-29 20:49:27.738255+0100 0x172      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: firmware had been download.
2020-03-29 20:49:27.738263+0100 0x95f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-03-29 20:49:27.738266+0100 0x95f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: End download
2020-03-29 20:49:27.738333+0100 0x95f      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Stop()

**Screenshots**
https://ibb.co/Y2bHw0S
https://ibb.co/YjKJ4r2

zxystd, 2020-05-13:
Did you use the `intelbluetoothinjector` kext?

Mandi89x, 2020-05-14:
I'm in the same boat with the 8265NGw and on Windows works perfectly

2020-05-14 02:01:26.076412+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2020-05-14 02:01:26.082428+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-05-14 02:01:26.082431+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-05-14 02:01:26.082433+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-05-14 02:01:26.082436+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了只找到了！！！
2020-05-14 02:01:26.082456+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到中断端点了！！！
2020-05-14 02:01:26.084608+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到Bulk输出端点了！！！
2020-05-14 02:01:26.086953+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到Bulk输入端点了！！！
2020-05-14 02:01:26.089763+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2020-05-14 02:01:26.089766+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-05-14 02:01:26.092380+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-05-14 02:01:26.093191+0200 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Bootloader revision 0.0 build 26 week 38 2015
2020-05-14 02:01:26.093195+0200 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: supect device firmware: ibt-12-18-0.sfi
2020-05-14 02:01:26.093211+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-05-14 02:01:26.093215+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_BOOT_PARAMS
2020-05-14 02:01:26.097185+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-05-14 02:01:26.098114+0200 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Device revision is 16
2020-05-14 02:01:26.098118+0200 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Secure boot is enabled
2020-05-14 02:01:26.098120+0200 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: OTP lock is enabled
2020-05-14 02:01:26.098122+0200 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: API lock is enabled
2020-05-14 02:01:26.098124+0200 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Debug lock is disabled
2020-05-14 02:01:26.098132+0200 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Minimum firmware build 1 week 10 2014
2020-05-14 02:01:26.098343+0200 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found device firmware: ibt-12-16.sfi
2020-05-14 02:01:26.098353+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-05-14 02:01:26.098357+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware header
2020-05-14 02:01:26.099104+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware pkey
2020-05-14 02:01:26.100258+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware signature
2020-05-14 02:01:26.101667+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware data
2020-05-14 02:01:28.631721+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: boot_param=0x40800
2020-05-14 02:01:28.631977+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware done
2020-05-14 02:01:28.631979+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-05-14 02:01:28.631980+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_RESET
2020-05-14 02:01:28.634415+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Intel reset succeed
2020-05-14 02:01:29.634491+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
[... cut ...]  IntelFirmware: 设备固件上传完成
2020-05-14 02:01:29.635549+0200 0x271      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-05-14 02:01:29.635552+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-05-14 02:01:29.635554+0200 0x2b4      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK

Mandi89x, 2020-05-14:
Mine tells "Bluetooth not available"

tvieira20, 2020-06-06:
> Did you use the `intelbluetoothinjector` kext?

I did... I downloaded the kext from [here](https://github.com/zxystd/IntelBluetoothFirmware/releases)

tvieira20, 2020-06-06:
> Mine tells "Bluetooth not available"

I ended up replacing the Intel 8265NGW Wi-Fi/BT Card for a BCM4352Z in order to fix everything...

closed 2020-07-17 by williambj1