OpenIntelWireless/IntelBluetoothFirmware #82: 蓝牙鼠标连接异常

codmap, 2020-03-30:
**描述**
> 感谢大神，手机、音箱可用，苹果原装鼠标（鼠标型号A1657）可用，罗技鼠标（M590）不可用，我借用同事非苹果型号的鼠标连接也不可用，系统偏好设置>鼠标>设置蓝牙鼠标，搜索到我的罗技鼠标，点击连接后，“已连接”和“未连接”频率来回切换，过一会儿就又变成“连接”按钮

**环境**

- Kext 版本: 最新版本
- WiFi Card 型号: 
- 产品 ID: 0xc52b
- macOS 版本: 10.15.3 (19D76)

**系统报告 - 蓝牙**
> 把所有内容都粘贴到下方的两个 ``` 之间

```
  Apple蓝牙软件版本：	7.0.3f5
  硬件、功能和设置：
  名称：	m的MacBook Pro
  地址：	90-61-AE-1F-A7-01
  支持低功耗蓝牙：	是
  支持“接力”：	否
  支持智能热点：	否
  生产企业：	Intel
  传输：	USB
  固件版本：	8.256 (8.256)
  蓝牙电源：	打开
  可被发现：	关闭
  可连接：	是
  自动寻找点：	关闭
  远程唤醒：	打开
  供应商ID：	0x8087
  产品ID：	0x0A2B
  蓝牙核心规范：	4.2 (0x8)
  HCI修正版：	0x100
  LMP版本：	4.2 (0x8)
  LMP子版本：	0x100
  设备类型（主要）：	Computer
  设备类型（完整）：	Mac Portable
  复合类设备：	0x38010C
  设备类（主要）：	0x01
  设备类（次要）：	0x03
  服务类：	0x1C0
  自动寻找键盘：	关闭
  设备（已配对、已配置等）：
  “陌生人”的鼠标：
  地址：	18-65-90-C1-9C-AA
  主要类型：	Peripheral
  次要类型：	Mouse
  服务：	Magic Mouse 2
  已配对：	是
  已配置：	是
  已连接：	是
  生产企业：	Broadcom (0x5, 0x240C)
  电池电量：	93%
  蓝牙核心规范：	3.0
  固件版本：	0x0102
  供应商ID：	0x004C
  产品ID：	0x0269
  设备类：	0x05 0x20 0x2580
  AFH：	打开
  AFH映射：	FEFF1F00000FEC
  RSSI：	0
  角色：	主设备
  连接模式：	活跃模式
  时间间隔：	0 ms
  主机可连接：	是
  支持EDR：	是
  支持eSCO：	否
  支持SSP：	是
  服务：
  蓝牙文件传输：
  其他设备可以浏览的文件夹：	~/Public
  当接收项目时：	接受所有文件而不发出警告
  状态：	已停用
  蓝牙文件交换：
  存放已接受项目的文件夹：	~/Downloads
  当接受其他项目时：	存储到位置
  当接收项目时：	接受所有文件而不发出警告
  状态：	已停用
  蓝牙互联网共享：
  状态：	已停用
  传入串口：
  Bluetooth-Incoming-Port：
  RFCOMM通道：	3
  需要认证：	否

```

**日志**
> 终端运行 `log show --last boot | grep IntelFir`，将得到的内容粘贴至 ```log 和 ``` 之间

```log
（命令输入之后是空的）
```

**内核崩溃日志**
> 如果出现了内核崩溃, 将 `keepsyms=1` 加入到引导参数中，然后粘贴崩溃日志到下方的两个 ``` 之间

```

```

**截图**
> 如果有必要，请在下方附加截图

codmap, 2020-03-30:
$ sudo kextutil -v IntelBluetoothFirmware.kext
Password:
Defaulting to kernel file '/System/Library/Kernels/kernel'
Kext rejected due to improper filesystem permissions: <OSKext 0x7fbd70e02480 [0x7fff8fdec690]> { URL = "file:///Volumes/Untitled/EFI/CLOVER/kexts/Other/IntelBluetoothFirmware.kext/", ID = "com.zxystd.IntelBluetoothFirmware" }
/Volumes/Untitled/EFI/CLOVER/kexts/Other/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOBluetoothFamily.
/Volumes/Untitled/EFI/CLOVER/kexts/Other/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOUSBHostFamily.
/Volumes/Untitled/EFI/CLOVER/kexts/Other/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOBluetoothFamily.
/Volumes/Untitled/EFI/CLOVER/kexts/Other/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOUSBHostFamily.
/Volumes/Untitled/EFI/CLOVER/kexts/Other/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOBluetoothFamily.
/Volumes/Untitled/EFI/CLOVER/kexts/Other/IntelBluetoothFirmware.kext - no compatible dependency found for com.apple.iokit.IOUSBHostFamily.
Code Signing Failure: code signature is invalid
Authentication Failures:
    File owner/permissions are incorrect (must be root:wheel, nonwritable by group/other):
        /Volumes/Untitled/EFI/CLOVER/kexts/Other/IntelBluetoothFirmware.kext
        Contents
        _CodeSignature
        CodeResources
        ._CodeResources
        .__CodeSignature
        MacOS
        IntelBluetoothFirmware
        ._IntelBluetoothFirmware
        ._MacOS
        Info.plist
        ._Info.plist
        ._Contents

Dependency Resolution Failures:
    Only incompatible kexts found for these libraries:
        com.apple.iokit.IOBluetoothFamily
        com.apple.iokit.IOUSBHostFamily

Diagnostics for /Volumes/Untitled/EFI/CLOVER/kexts/Other/IntelBluetoothFirmware.kext:
Authentication Failures:
    File owner/permissions are incorrect (must be root:wheel, nonwritable by group/other):
        /Volumes/Untitled/EFI/CLOVER/kexts/Other/IntelBluetoothFirmware.kext
        Contents
        _CodeSignature
        CodeResources
        ._CodeResources
        .__CodeSignature
        MacOS
        IntelBluetoothFirmware
        ._IntelBluetoothFirmware
        ._MacOS
        Info.plist
        ._Info.plist
        ._Contents

Dependency Resolution Failures:
    Only incompatible kexts found for these libraries:
        com.apple.iokit.IOBluetoothFamily
        com.apple.iokit.IOUSBHostFamily

zxystd, 2020-04-06:
4.0 device seems have problem with intel bt connection due to apple drivers, firmware kext does not work for this section.

closed 2020-04-06 by zxystd