OpenIntelWireless/IntelBluetoothFirmware #83: How to build?

johnnync13, 2020-03-30:
I have this error.
(base) johnny@Johnnys-MacBook IntelBluetoothFirmware % ./fw_gen.sh 
./fw_gen.sh: line 14: /IntelBluetoothFirmware/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 15: /IntelBluetoothFirmware/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 21: /IntelBluetoothFirmware/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 22: /IntelBluetoothFirmware/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 23: /IntelBluetoothFirmware/fw/*.*: No such file or directory
./fw_gen.sh: line 24: /IntelBluetoothFirmware/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 25: /IntelBluetoothFirmware/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 26: /IntelBluetoothFirmware/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 29: /IntelBluetoothFirmware/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 31: /IntelBluetoothFirmware/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 37: /IntelBluetoothFirmware/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 40: /IntelBluetoothFirmware/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 41: /IntelBluetoothFirmware/FwBinary.cpp: No such file or directory
(base) johnny@Johnnys-MacBook IntelBluetoothFirmware %

Menchen, 2020-03-30:
Build the project in xcode.

johnnync13, 2020-03-30:
@Menchen yeah! but not generate kext...

Menchen, 2020-03-30:
What error in xcode?

Menchen, 2020-03-30:
@johnnync13  In xcode project navigator, under `Product` there's a kext

johnnync13, 2020-03-30:
> @johnnync13  In xcode project navigator, under `Product` there's a kext

But if I show on finder, not appear

Menchen, 2020-03-30:
How did you build it?

johnnync13, 2020-03-30:
@Menchen how to do this?

Menchen, 2020-03-30:
[image]
The play button

johnnync13, 2020-03-30:
yes! that's how I do it @Menchen But how can I extract kexts?

Menchen, 2020-03-30:
Right click the kext and show in finder..... Are you sure that it build without any changing sign setting? Because I had to change to mine...

Just in case that you still can't build, there's mine with latest repo.
There's no difference in term of feature... The only thing I noticed was size change and less log. 
[IntelBluetoothFirmware.kext.zip](https://github.com/zxystd/IntelBluetoothFirmware/files/4405898/IntelBluetoothFirmware.kext.zip)

johnnync13, 2020-03-30:
@Menchen thank you. This was the problem

closed 2020-03-30 by johnnync13