OpenIntelWireless/IntelBluetoothFirmware #84: no show

subodhkapoor, 2020-04-02:
**Description**
> A clear and concise description of what the bug is
whatecver i try i dont get to see the bluetooth adapater 
**Environment**
Macos Catalina 10.15.4 bluetooth 5.0 with intel AX200 (supported as per the listed IDs)

- Kext Version: latest one released march 2020 end
- WiFi Card Model: AX200
- Product ID: USB\VID_8087&PID_0029
- macOS Version: 10.15.4

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

In system information - Bluetooth ->  No information 

```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```
No lolgs shown 
```log

```

**Kernel Panic Logs**
> In case there is a kernel panic, add `keepsyms=1` in boot-args then paste the panic info below between ```

```

```

**Screenshots**
> If applicable, add screenshots to help explain your problem.

Menchen, 2020-04-02:
Can you upload `system_profiler SPBluetoothDataType` ?

aluveitie, 2020-04-02:
I've the same setup and issue, usually it works after rebooting.
In the past once it persisted for some time (through complete shutdowns) but lately I have it on nearly every fresh bootup.

subodhkapoor, 2020-04-02:
> Can you upload `system_profiler SPBluetoothDataType` ?

i dont see bluetooth info in the system_profiler

Menchen, 2020-04-02:
Then it’s you EFI problem, check USBPort

subodhkapoor, 2020-04-02:
> Then it’s you EFI problem, check USBPort
i actually have taken the dump of DSDT and am using that configureation .... works fine in both windows and linux

Menchen, 2020-04-02:
@aluveitie #23 #35

Menchen, 2020-04-02:
@subodhkapoor You should see BT in `system_profiler` even without this kext... so there's a problem with your EFI...

subodhkapoor, 2020-04-02:
> @subodhkapoor You should see BT in `system_profiler` even without this kext... so there's a problem with your EFI...

ok.. what do i need to look for in the EFI? something specific?

Menchen, 2020-04-02:
Remove USBPort and try usbinjectall to debug...

subodhkapoor, 2020-04-04:
Yes thank you ... i removed all USB ports and created a new USB Map.. now it shows as bluetooth device and also discovers my iphone but doesnty connect 

log 
log show --last boot | grep IntelFirmware
2020-04-04 18:36:08.474299+0530 0x23a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-04-04 18:36:08.478609+0530 0x23a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-04-04 18:36:08.481726+0530 0x23a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0029
2020-04-04 18:36:08.485106+0530 0x23a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()

subodhkapoor, 2020-04-04:
couple of updates - i replaced the kext and rebooted it .. it connected with my iphone for a second and after that it disconnedted. I tried switching bluetooth off and on.... but couldnt do it... it froze my mac.  is there something wrong with the firmware?

Menchen, 2020-04-04:
Certain device/protocol won’t work because macOS’s bt driver is not written for intel...(this Kext only upload firmware)

subodhkapoor, 2020-04-05:
guess nothing can be done... closing the issue

closed 2020-04-05 by subodhkapoor