OpenIntelWireless/IntelBluetoothFirmware #85: now bluetooth is completely gone

mpatton68, 2020-04-04:
Last login: Sat Apr  4 11:22:20 on ttys000
michaelpatton@Michaels-iMac-Pro ~ % log show --last boot | grep IntelFirmware
2020-04-04 11:21:05.050632-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2020-04-04 11:21:05.056743-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-04-04 11:21:05.056748-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-04-04 11:21:05.056750-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-04-04 11:21:05.056752-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了只找到了！！！
2020-04-04 11:21:05.056805-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到中断端点了！！！
2020-04-04 11:21:05.059073-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到Bulk输出端点了！！！
2020-04-04 11:21:05.059149-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到Bulk输入端点了！！！
2020-04-04 11:21:05.059221-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2020-04-04 11:21:05.059224-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-04-04 11:21:05.061609-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-04-04 11:21:05.062587-0700 0x171      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Bootloader revision 0.3 build 0 week 24 2017
2020-04-04 11:21:05.062591-0700 0x171      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: supect device firmware: ibt-20-1-3.sfi
2020-04-04 11:21:05.062596-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-04-04 11:21:05.062598-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_BOOT_PARAMS
2020-04-04 11:21:05.065605-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-04-04 11:21:05.067596-0700 0x171      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Device revision is 1
2020-04-04 11:21:05.067598-0700 0x171      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Secure boot is enabled
2020-04-04 11:21:05.067600-0700 0x171      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: OTP lock is enabled
2020-04-04 11:21:05.067602-0700 0x171      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: API lock is enabled
2020-04-04 11:21:05.067603-0700 0x171      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Debug lock is disabled
2020-04-04 11:21:05.067605-0700 0x171      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Minimum firmware build 1 week 10 2014
2020-04-04 11:21:05.067805-0700 0x171      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found device firmware: ibt-20-1-3.sfi
2020-04-04 11:21:05.067811-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-04-04 11:21:05.067814-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware header
2020-04-04 11:21:05.068098-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware pkey
2020-04-04 11:21:05.068618-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware signature
2020-04-04 11:21:05.069289-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware data
2020-04-04 11:21:06.352020-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: boot_param=0x24800
2020-04-04 11:21:06.352177-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware done
2020-04-04 11:21:06.352179-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-04-04 11:21:06.352181-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_RESET
2020-04-04 11:21:06.354545-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Intel reset succeed
2020-04-04 11:21:07.355583-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-04-04 11:21:07.357537-0700 0x171      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 设备固件上传完成
2020-04-04 11:21:07.357541-0700 0x171      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-04-04 11:21:07.357545-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-04-04 11:21:07.357548-0700 0x1e2      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
michaelpatton@Michaels-iMac-Pro ~ %

Menchen, 2020-04-04:
Could you try `sudo kextunload -b com.zxystd.IntelBluetoothFirmware` and `sudo kextload pathtokext`?

zxystd, 2020-04-06:
looks like kext works fine. maybe there is something wrong with your hackintosh?

Menchen, 2020-04-06:
@zxystd I have this exact same issue sometimes with exactly the same log, it stops at HCI_OP_INTEL_EVENT_MASK(should end at `Driver Stop()`) and re-updating the firmware annually(I  used VM) fix it. Seem only happening on battery for me.

XCreative, 2020-04-18:
@zxystd, i have the same problem.

```
> log show --last boot | grep IntelFirmware

2020-04-18 14:05:43.069253+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-04-18 14:05:43.069258+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-04-18 14:05:43.069262+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0025
2020-04-18 14:05:43.069267+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-04-18 14:05:44.032495+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2020-04-18 14:05:44.038499+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-04-18 14:05:44.038501+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-04-18 14:05:44.038502+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-04-18 14:05:44.038503+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了只找到了！！！
2020-04-18 14:05:44.038513+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到中断端点了！！！
2020-04-18 14:05:44.040581+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到Bulk输出端点了！！！
2020-04-18 14:05:44.040620+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到Bulk输入端点了！！！
2020-04-18 14:05:44.040651+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2020-04-18 14:05:44.040652+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-04-18 14:05:44.043491+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-04-18 14:05:44.044491+0100 0x173      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Bootloader revision 0.1 build 42 week 52 2015
2020-04-18 14:05:44.044494+0100 0x173      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: supect device firmware: ibt-18-16-1.sfi
2020-04-18 14:05:44.044497+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-04-18 14:05:44.044499+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_BOOT_PARAMS
2020-04-18 14:05:44.047487+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-04-18 14:05:44.049502+0100 0x173      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Device revision is 2
2020-04-18 14:05:44.049504+0100 0x173      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Secure boot is enabled
2020-04-18 14:05:44.049505+0100 0x173      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: OTP lock is enabled
2020-04-18 14:05:44.049507+0100 0x173      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: API lock is enabled
2020-04-18 14:05:44.049508+0100 0x173      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Debug lock is disabled
2020-04-18 14:05:44.049510+0100 0x173      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Minimum firmware build 1 week 10 2014
2020-04-18 14:05:44.049646+0100 0x173      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Found device firmware: ibt-18-16-1.sfi
2020-04-18 14:05:44.049650+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-04-18 14:05:44.049651+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware header
2020-04-18 14:05:44.049919+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware pkey
2020-04-18 14:05:44.050421+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware signature
2020-04-18 14:05:44.051169+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware data
2020-04-18 14:05:45.223587+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: boot_param=0x40800
2020-04-18 14:05:45.223741+0100 0x1db      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: send firmware done
[... cut ...]
> sudo kextutil -t IntelBluetoothFirmware.kext

Untrusted kexts are not allowed
Kext with invalid signature (-67062) denied: /Library/StagedExtensions/Volumes/EFI/EFI/OC/Kexts/6DB3725C-6BFA-46A8-824A-0CE28F5989DB.kext
Bundle (/Volumes/EFI/EFI/OC/Kexts/IntelBluetoothFirmware.kext) failed to validate, deleting: /Library/StagedExtensions/Volumes/EFI/EFI/OC/Kexts/6DB3725C-6BFA-46A8-824A-0CE28F5989DB.kext
Unable to stage kext (/Volumes/EFI/EFI/OC/Kexts/IntelBluetoothFirmware.kext) to secure location.
```

opzhang, 2020-07-20:
I put the Bluetooth driver in clover, and it doesn't load at all when I put it in SLE. Intel 3160 network card

sclai, 2020-12-22:
I think I know what is the issue here.
When you add the Kexts and modify the config.plist. Using proper tree and OpenCore, please do "OC Snapshot" on /EFI/EFI/OC then safe your config.plist. I have verified this working 100% on my system. Because I was having same issue.

closed 2021-06-22 by zxystd