OpenIntelWireless/IntelBluetoothFirmware #87: Bluetooth intel ac7260 still not loaded

guaychou, 2020-04-08:
**Description**
> A clear and concise description of what the bug is

**Environment**

- Kext Version: 1.0.3
- WiFi Card Model: Intel AC7260
- Product ID: 0x07DC
- macOS Version: 10.14.6  

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```

```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```
2020-04-08 13:03:52.341157+0700 0x151      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver init()
2020-04-08 13:03:52.341174+0700 0x151      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Probe()
2020-04-08 13:03:52.341182+0700 0x151      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x07DC
2020-04-08 13:03:52.341191+0700 0x151      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Start()
```log

```

**Kernel Panic Logs**
> In case there is a kernel panic, add `keepsyms=1` in boot-args then paste the panic info below between ```

```

```

**Screenshots**
> If applicable, add screenshots to help explain your problem.
[image]

[image]

[image]

guaychou, 2020-04-08:
please help me @zxystd

guaychou, 2020-04-20:
@zxystd

zxystd, 2020-04-20:
I think it is because you have block the BT USB port(maybe exist USBPorts.kext) or disable BT in bios. check these, thanks.

igorkulman, 2020-04-29:
I am also using Intel 7260 and had the same problem initially. Turns out I had  a few kexts installed that were in conflict with this one:

- AirportBrcmFixup.kext
- BT4LEContinuityFixup.kex
- BrcmFirmwareRepo.kext 
- BrcmPatchRAM2.kext

After removing them and loading this kext Bluetooth appeared in Preferences and I was able to pair and use a Bluetooth keyboard.

williambj1, 2020-07-13:
Closed because this issue is no longer active for quite some time. Feel free to reopen!

closed 2020-07-13 by williambj1