OpenIntelWireless/IntelBluetoothFirmware #88: Cannot turn off Bluetooth from prefpane.

drigio, 2020-04-09:
**Description**
> The option to turn off bluetooth is greyed out. Is there any alternative for this?

**Environment**

- Kext Version:  1.0.3
- WiFi Card Model: AC3168 NGW
- Bluetooth Device ID: 0aa7  
- macOS Version: 10.15.4

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between 
```
// Blank
```
**Screenshots**

[image]

Is this the normal behavior that is currently expected of Intel Bluetooth Adapters? I read somewhere on tonymacx86 that 
> only Broadcom Bluetooth devices currently have the ability to turn on or off.

Is this the case?

Menchen, 2020-04-09:
Did you install IntelBluetoothInjector.kext?

drigio, 2020-04-10:
Yes I had placed `IntelBluetoothInjector.kext` in `EFI/Clover/kexts/Other`. 
On a side note, I already had the bluetooth option available in prefpane before installing `IntelBluetoothFirmware`.

drigio, 2020-04-18:
@zxystd @Menchen Bump!

igorkulman, 2020-04-29:
Same problem on AC 7260, Bluetooth works, just cannot be turned off.

williambj1, 2020-07-13:
Please try the latest version, if this issue still exists, feel free to reopen, thanks!

closed 2020-07-13 by williambj1

drigio, 2020-07-19:
@williambj1 Tried with the latest release, still facing the same issue. Last boot logs are still blank. 

Please reopen the issue if  people are facing the same issue as I am still.

williambj1, 2020-07-19:
Can you post your ioreg?

drigio, 2020-07-19:
@williambj1 Do i export a something specific like by searching `bluetooth` or i export everything. Also the complete file in > 25 MB, how do i send that?

williambj1, 2020-07-19:
Just drop it in the chatbox, 25MB is fine.

drigio, 2020-07-19:
I searched for the `IntelFirmware` and `IntelBluetoothFirware` and the search was empty, I think the kexts are not loaded. 

Here's the ioreg copy. (It wasn't accepting the ioreg format so had to zip it. Also helped to compress the size a lot)
[Ioreg.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/4943709/Ioreg.zip)

Btw, thanks for the quick replies :)

williambj1, 2020-07-19:
Look like IntelBluetoothFirmware is not loaded, here's a side by side comparison with mine.

[image]

drigio, 2020-07-19:
Yes, I noticed the same, the kext isn't loaded. I have correctly put the kexts in `EFI/Clover/Kexts/Other`. What do you suggest I do in this case? The kext isn't getting loaded at boot I guess. Is this because I am using `VirtualSMC`? 

Here's the contents of `Clover/kexts/Other/`

[image]

drigio, 2020-07-19:
Hey I managed to get it working! As I am using `VirtualSMC` instead of `fakeSMC`, I had all my kexts in my `L/E` or `S/L/E`. I checked my clover configuration and as `VirtualSMC` users are advised to never inject kexts at boot, I had kept my `Inject Kext` variable to `DETECT` in `config.plist`. I changed that to `Yes` and now I am able to inject the kexts from `Clover/kexts/Other` folder at boot. 

Can confirm that it's working fine and I am able to turn on and off the bluetooth. If anyone has the same error check if your `Inject Kext` is set to `Yes` in config.plist.  

Thanks for everyone's support :) especially @williambj1