OpenIntelWireless/IntelBluetoothFirmware #89: Bluetooth was not working when this error show

harika09, 2020-04-10:
Every time this error show on bootscreen bluetooth will stop working. No logs at all when executing the

`log show --last boot | grep IntelFirmware `

`apfs_keybag_init:1583: failed to initialize volume keybag, err = 2
apfs_keybag_init:1583: failed to initialize volume keybag, err = 2
apfs_keybag_init:1583: failed to initialize volume keybag, err = 2
AGDCC: Unauthorized client 'PerfPowerService' (IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/IGPU@2/AppleIntelFramebufferController/IntelFBClientControl)
IOConsoleUsers: gIOScreenLockState 3, hs 0, bs 0, now 0, sm 0x0
AppleLMUController::smcReadKey Error: received error 0x84 when reading key 'MSLD' `

[image]

harika09, 2020-04-11:
Every time that error shows the only solution was re-install the kext to the EFI/Clover/kext

williambj1, 2020-07-13:
The verbose logs seem irrelevant, if this issue still exists with the latest version, feel free to reopen, thanks.

closed 2020-07-13 by williambj1