OpenIntelWireless/IntelBluetoothFirmware #91: 无法加载驱动

Archie87, 2020-04-13:
**描述**
> 简要描述 Kext 存在的问题
蓝牙无法驱动
**环境**
clover 
- Kext 版本: 1.0.3
- WiFi Card 型号: nuc8
- 产品 ID: 
- macOS 版本: os10.15.1**系统报告 - 蓝牙**
> 把所有内容都粘贴到下方的两个 ``` 之间

```
USB 3.1总线：

  主控制器驱动器：	AppleIntelCNLUSBXHCI
  PCI设备ID：	0x9ded 
  PCI修订版ID：	0x0030 
  PCI供应商ID：	0x8086 

蓝牙HCI：

  产品ID：	0x0aaa
  厂商ID：	0x8087  (Intel Corporation)
  版本：	0.02
  速度：	最高可达12 Mb/秒
  位置ID：	0x14700000 / 1
  可用电流(mA)：	500
  所需电流(mA)：	100
  额外的操作电流(mA)：	0
```

**日志**
> 终端运行 `log show --last boot | grep IntelFir`，将得到的内容粘贴至 ```log 和 ``` 之间

```log
2020-04-13 10:10:48.206282+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver init()
2020-04-13 10:10:48.206290+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Probe()
2020-04-13 10:10:48.206298+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0AAA
2020-04-13 10:10:48.206305+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Start()
2020-04-13 10:10:49.708320+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: set configuration to 1
2020-04-13 10:10:49.708921+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-04-13 10:10:49.708924+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-04-13 10:10:49.708927+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-04-13 10:10:49.708929+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了只找到了！！！
2020-04-13 10:10:49.708946+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到中断端点了！！！
2020-04-13 10:10:49.709154+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到Bulk输出端点了！！！
2020-04-13 10:10:49.709361+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到Bulk输入端点了！！！
2020-04-13 10:10:49.709584+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: usb init succeed
2020-04-13 10:10:49.709589+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_VERSION
2020-04-13 10:10:49.715155+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-04-13 10:10:49.715576+0800 0x144      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Bootloader revision 0.1 build 42 week 52 2015
2020-04-13 10:10:49.715579+0800 0x144      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: supect device firmware: ibt-17-16-1.sfi
2020-04-13 10:10:49.715583+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-04-13 10:10:49.715584+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_BOOT_PARAMS
2020-04-13 10:10:49.715782+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-04-13 10:10:49.716628+0800 0x144      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Device revision is 2
2020-04-13 10:10:49.716634+0800 0x144      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Secure boot is enabled
2020-04-13 10:10:49.716638+0800 0x144      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: OTP lock is enabled
2020-04-13 10:10:49.716661+0800 0x144      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: API lock is enabled
2020-04-13 10:10:49.716662+0800 0x144      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Debug lock is disabled
2020-04-13 10:10:49.716664+0800 0x144      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Minimum firmware build 1 week 10 2014
2020-04-13 10:10:49.716896+0800 0x144      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Found device firmware: ibt-17-16-1.sfi
2020-04-13 10:10:49.716906+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-04-13 10:10:49.716908+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: send firmware header
2020-04-13 10:10:49.717349+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: send firmware pkey
2020-04-13 10:10:49.718198+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: send firmware signature
2020-04-13 10:10:49.719216+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: send firmware data
2020-04-13 10:10:50.974021+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: boot_param=0x40800
2020-04-13 10:10:50.974180+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: send firmware done
2020-04-13 10:10:50.974181+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
[... 17 lines cut ...]
2020-04-13 10:10:53.489263+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-04-13 10:10:53.489268+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-04-13 10:10:53.489284+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 忙碌中，别急
2020-04-13 10:10:53.489286+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了只找到了！！！
2020-04-13 10:10:53.489306+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到中断端点了！！！
2020-04-13 10:10:53.489500+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到Bulk输出端点了！！！
2020-04-13 10:10:53.489693+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: 找到了找到了，找到Bulk输入端点了！！！
2020-04-13 10:10:53.489882+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: usb init succeed
2020-04-13 10:10:53.489885+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: HCI_OP_INTEL_VERSION
2020-04-13 10:10:53.494761+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt wait
2020-04-13 10:10:53.495576+0800 0x144      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Firmware revision 0.1 build 20 week 53 2019
2020-04-13 10:10:53.495578+0800 0x144      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: firmware had been download.
2020-04-13 10:10:53.495582+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: interrupt continue
2020-04-13 10:10:53.495585+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: End download
2020-04-13 10:10:53.495623+0800 0x50d      Default     0x0                  0      0    kernel: (kernel) IntelFirmware: Driver Stop()``

**内核崩溃日志**
> 如果出现了内核崩溃, 将 `keepsyms=1` 加入到引导参数中，然后粘贴崩溃日志到下方的两个 ``` 之间

```

```

**截图**
> 如果有必要，请在下方附加截图

zxystd, 2020-04-13:
尝试一下1.0.2版本，1.0.3版本更新成了全新的固件。log显示固件已经成功上传了，但是没有工作的话，我估计是固件导致的问题

Archie87, 2020-04-13:
> 尝试一下1.0.2版本，1.0.3版本更新成了全新的固件。log显示固件已经成功上传了，但是没有工作的话，我估计是固件导致的问题

1.0.2也是这样的。。。发现更新了新版所以来试试。。。
请问这个和我改过/System/Library/Extensions/IOBluetoothFamily.kext/Contents/PlugIns/BroadcomBluetoothHostControllerUSBTransport.kext/Contents/Info.plist
有关系吗？能否提供一个原来的BroadcomBluetoothHostControllerUSBTransport.kext我替换一下试试

Airomeo, 2020-04-13:
相同设备相同问题。config配置了，kext也放了，系统蓝牙plist文件也删了，打开蓝牙显示 空，鼠标变忙碌，窗口无响应。实际上手机等设备可以搜索到电脑蓝牙，但是电脑窗口由于无响应，无法搜索和显示其他设备。感觉是硬件驱动了，但是不知道什么原因导致无法使用。

chenyuyou, 2020-06-08:
我的机器是NUC8i5BEH，原装的intel网卡和蓝牙，mac版本是15.5.5，使用的是opencore0.5.8，目前蓝牙没有该问题，虽然蓝牙不支持handoff

williambj1, 2020-07-13:
如果新版本 v1.1.1 还存在这个问题欢迎随时再次打开这个 issue，谢谢

closed 2020-07-13 by williambj1