OpenIntelWireless/IntelBluetoothFirmware #95: BT module goes offline (kext downloads older firmware)

drdaz, 2020-04-18:
**Description**
> A clear and concise description of what the bug is

After some time, the Bluetooth module goes offline (in both Windows and macOS). It appears that the kext downloads different versions of the firmware. It appears that when it downloads the older firmware, that the BT module goes offline.

**Environment**

- Kext Version: 1.0.3
- WiFi Card Model: 
- Product ID: 0x0AAA
- macOS Version: 10.15.4

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```
Apple Bluetooth Software Version:	7.0.4f6
  Hardware, Features and Settings:
  Name:	Mac Pro
  Address:	
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	10.256 (10.256)
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0AAA
  Bluetooth Core Spec:	Unknown (0xA)
  HCI Revision:	0x100
  LMP Version:	Unknown (0xA)
  LMP Subversion:	0x100
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Desktop
  Composite Class Of Device:	0x380104
  Device Class (Major):	0x01
  Device Class (Minor):	0x01
  Service Class:	0x1C0
  Auto Seek Keyboard:	On
  Devices (Paired, Configured, etc.):
  Darren Black’s Keyboard:
  Address:	
  Major Type:	Peripheral
  Minor Type:	Keyboard
  Services:	Magic Keyboard with Numeric Keypad
  Paired:	Yes
  Configured:	Yes
  Connected:	Yes
  Manufacturer:	Broadcom (0x5, 0x240C)
  Battery Level:	4%
  Bluetooth Core Spec:	3,0
  Firmware Version:	0x0100
  Vendor ID:	0x004C
  Product ID:	0x026C
  Class of Device:	0x05 0x10 0x0540
  AFH:	On
  AFH Map:	FEFF1F00FF3000
  RSSI:	0
  Role:	Master
  Connection Mode:	Active Mode
  Interval:	0 ms
  Host Connectable:	Yes
  EDR Supported:	Yes
  eSCO Supported:	No
  SSP Supported:	Yes
  Darren Black’s Mouse:
  Address:	
  Major Type:	Peripheral
  Minor Type:	Mouse
  Services:	Magic Mouse 2
  Paired:	Yes
  Configured:	Yes
  Connected:	Yes
  Manufacturer:	Broadcom (0x5, 0x240C)
  Battery Level:	84%
  Bluetooth Core Spec:	3,0
  Firmware Version:	0x0102
  Vendor ID:	0x004C
  Product ID:	0x0269
  Class of Device:	0x05 0x20 0x0580
  AFH:	On
  AFH Map:	FEFF1F00FF3000
  RSSI:	0
  Role:	Master
  Connection Mode:	Active Mode
  Interval:	0 ms
  Host Connectable:	Yes
  EDR Supported:	Yes
  eSCO Supported:	No
  SSP Supported:	Yes
  Darren Black’s Trackpad:
  Address:	
  Major Type:	Peripheral
  Minor Type:	Trackpad
  Services:	Magic Trackpad 2
  Paired:	Yes
  Configured:	Yes
  Connected:	Yes
  Manufacturer:	Broadcom (0x5, 0x240C)
  Battery Level:	40%
  Bluetooth Core Spec:	3,0
  Firmware Version:	0x0102
  Vendor ID:	0x004C
  Product ID:	0x0265
  Class of Device:	0x05 0x25 0x0594
  AFH:	On
  AFH Map:	FEFF1F00FF3000
  RSSI:	0
  Role:	Master
  Connection Mode:	Active Mode
  Interval:	0 ms
  Host Connectable:	Yes
  EDR Supported:	Yes
  eSCO Supported:	No
  SSP Supported:	Yes
  Darren’s BeatsX:
  Address:	
  Major Type:	Miscellaneous
  Minor Type:	Unknown
  Services:	
  Paired:	Yes
  Configured:	Yes
  Connected:	No
  Bluetooth Core Spec:	1.0b
  Vendor ID:	0x004C
  Product ID:	0x2005
  Class of Device:	0x00 0x00 0x0000
  Services:
  Bluetooth File Transfer:
  Folder other devices can browse:	~/Public
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth File Exchange:
  Folder for accepted items:	~/Downloads
  When other items are accepted:	Save to location
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth Internet Sharing:
  State:	Disabled
  Incoming Serial Ports:
  Bluetooth-Incoming-Port:
  RFCOMM Channel:	3
  Requires Authentication:	No

```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

Old firmware:
```log
2020-04-17 23:43:18.360569+0200 0x7c65     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2020-04-17 23:43:18.360570+0200 0x7c65     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-04-17 23:43:18.360760+0200 0x7c65     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-04-17 23:43:18.361211+0200 0x1ad      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Firmware revision 0.1 build 8 week 47 2019
2020-04-17 23:43:18.361214+0200 0x1ad      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: firmware had been download.
2020-04-17 23:43:18.361218+0200 0x7c65     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-04-17 23:43:18.361221+0200 0x7c65     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
```

New firmware:
```log
2020-04-17 11:53:43.723790+0200 0x11960    Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2020-04-17 11:53:43.723791+0200 0x11960    Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-04-17 11:53:43.724106+0200 0x11960    Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-04-17 11:53:43.724633+0200 0x156      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Firmware revision 0.1 build 20 week 53 2019
2020-04-17 11:53:43.724636+0200 0x156      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: firmware had been download.
2020-04-17 11:53:43.724639+0200 0x11960    Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-04-17 11:53:43.724642+0200 0x11960    Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download

```

drdaz, 2020-04-18:
@zxystd Where is the firmware downloaded from? A website? I can't see any URLs in the code, so I'm guessing not, but...

williambj1, 2020-04-18:
@drdaz https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/intel

drdaz, 2020-04-18:
@williambj1 Thanks. Does the kext actively download from there? Where in the code does this happen?

There's also a firmware binary that gets included in the kext I see. Just trying to figure out how all this works :)

williambj1, 2020-04-18:
No, the kext does not download the firmware files from the Internet, and that's impossible.

Maybe @zxystd could explain further how it works exactly. Sorry for my lack of knowledge.

drdaz, 2020-04-18:
No need to apologise! Thank you for responding :)

zxystd, 2020-04-18:
> @zxystd Where is the firmware downloaded from? A website? I can't see any URLs in the code, so I'm guessing not, but...

This driver downloads(by hardware HCI protocol) the firmware pre compiled into kext to the hardware and restarts it to make it work. You can find these firmware in the `firmware` directory. They are all provided by Linux Intel. You can download(by common HTTP/HTTPS protocol) these firmware files from Intel official website, replace them to the directory, and then compile.

drdaz, 2020-04-18:
@zxystd Thanks. Do you have any idea why the kext might be selecting different firmwares at different times?

zxystd, 2020-04-18:
Maybe you can use 1.0.2 version and try again? To help confirm if is the firmware cause this problem, just because I have replace all of the firmware files in version 1.0.3 .

drdaz, 2020-04-18:
I typically experience this after sleep. I'll return to my machine and be unable to wake it using BT, and find once it's awake that BT is unavailable.

> Maybe you can use 1.0.2 version and try again? To help confirm if is the firmware cause this problem, just because I have replace all of the firmware files in version 1.0.3 .

Sure I'll give that a try... but as I mentioned this isn't consistent. Using the same version (1.0.3) of the kext yields different firmware versions being loaded according to the logs.

zxystd, 2020-04-18:
What do you mean by `both Windows and macOS` ? switch Windows to MacOS?

drdaz, 2020-04-18:
> What do you mean by `both Windows and macOS` ? switch Windows to MacOS?

I mean that the device appears to be switched off from BIOS. The hardware doesn't show up in either OS.

I've experienced this on 2 separate motherboards (same make and model - ASRock z390 Phantom Gaming itx/ac).

zxystd, 2020-04-18:
The Firmware version which the kext first query response is, if the hardware doesn't have firmware, it will return the ram version, but if it has firmware, it will return the firmware version it have. So I guess it is because of the different fimware version between your Windows and MacOS

drdaz, 2020-04-18:
So I guess it's possible that the issue I'm seeing isn't related to the firmware version being loaded...

I'm running 1.0.2 now btw. I actually can't find the log entries; when I run `log show --last 1h | grep IntelFirmware`, I get no result. Even though I rebooted minutes ago.

Using OpenCore. 

Perhaps I need to force a refresh of the kext cache?

zxystd, 2020-04-18:
No need to do this, I also get not logs sometimes... And the BT works(correct bt version from system information panel)?

drdaz, 2020-04-18:
Yeah it works. Different firmware version showing now (9.256 instead of 10.256 in my first post).

For what it's worth, it appeared to work without your kexts too (was correctly detected by macOS), but would usually cause a kernel panic when waking using the BT devices. Those crashes are no more, but using your kexts it takes a long time before the displays wake. Not sure if this is expected behaviour...

drdaz, 2020-04-18:
Oh and BT version is 5.0.

drdaz, 2020-04-18:
When the issue occurs, what I see in the BT menu is like this: https://github.com/zxystd/IntelBluetoothFirmware/issues/70#issuecomment-603022834

Just without any PacketLogger entry.

drdaz, 2020-04-19:
I'm seeing the same behaviour in the previous release 1.0.2. So it's not the firmware version.

closed 2020-08-06 by drdaz