OpenIntelWireless/IntelBluetoothFirmware #96: Issue with Catalina and x570 board.

josharmour, 2020-04-20:
I am not seeing any output for 'log show --last boot | grep IntelFirmware'

I did add both kexts into my OpenCore efi partition and updated my config.plist.

I had bluetooth working without any modifications on the same system with 10.13.

I ran the command suggested from a previous issue 'kextlibs -undef-symbols IntelBluetoothFirmware.kext'

I got the following back:

josharmour@Joshs-Mac-Pro IntelBluetooth % kextlibs -undef-symbols IntelBluetoothFirmware.kext
For all architectures:
    com.apple.iokit.IOUSBHostFamily = 1.2
    com.apple.kpi.iokit = 19.4
    com.apple.kpi.libkern = 19.4
    com.apple.kpi.mach = 19.4

For x86_64:
    1 symbol not found in any library kext:
	__ZN15OSMetaClassBase25_RESERVEDOSMetaClassBase3Ev

josharmour@Joshs-Mac-Pro IntelBluetooth % 

The motherboard I am using is the Asus Crosshair viii impact (mini-dtx).

Here is the pic of the bluetooth hardware in windows

[image]

josharmour, 2020-04-28:
I'm making progress.

Last login: Tue Apr 28 12:05:39 on console
josharmour@Joshs-Mac-Pro ~ % log show --last boot | grep IntelFirmware
2020-04-28 12:05:20.198964-0700 0x216      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-04-28 12:05:20.202749-0700 0x216      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-04-28 12:05:20.205780-0700 0x216      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0029
2020-04-28 12:05:20.208870-0700 0x216      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-04-28 12:05:48.358730-0700 0x74       Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver free()
2020-04-28 12:05:48.576162-0700 0x186f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-04-28 12:05:48.578610-0700 0x186f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-04-28 12:05:48.581984-0700 0x186f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0029
2020-04-28 12:05:48.589584-0700 0x186f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-04-28 12:05:50.093337-0700 0x186f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2020-04-28 12:05:50.099973-0700 0x186f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-04-28 12:05:50.102849-0700 0x186f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-04-28 12:05:50.105875-0700 0x186f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-04-28 12:05:50.108900-0700 0x186f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-04-28 12:05:50.111927-0700 0x186f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了只找到了！！！
2020-04-28 12:05:50.116007-0700 0x186f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到中断端点了！！！
2020-04-28 12:05:50.126148-0700 0x186f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到Bulk输出端点了！！！
2020-04-28 12:05:50.132266-0700 0x186f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到Bulk输入端点了！！！
2020-04-28 12:05:50.138486-0700 0x186f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2020-04-28 12:05:50.141298-0700 0x186f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-04-28 12:05:50.146965-0700 0x186f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-04-28 12:05:50.149500-0700 0x1d5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Firmware revision 0.0 build 100 week 47 2019
2020-04-28 12:05:50.154596-0700 0x1d5      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: firmware had been download.
2020-04-28 12:05:50.158403-0700 0x186f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-04-28 12:05:50.161429-0700 0x186f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-04-28 12:05:50.163954-0700 0x186f     Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
josharmour@Joshs-Mac-Pro ~ % 

I changed my smbios to use iMacPro1,1 and now I see something.  

I do have issues with turning bluetooth off and on (cause a reboot).  I also can't seem to see or connect to my airpods.

However I was able to connect to the hackintosh using my iphone, and then able to transfer a call from my iphone to the hackintosh from the iphone.

jkibasui, 2020-06-28:
How did you solve this?

me@MacBook-Pro IntelBluetooth 2 % kextlibs -undef-symbols IntelBluetoothFirmware.kext           
For all architectures:
    com.apple.iokit.IOUSBHostFamily = 1.2
    com.apple.kpi.iokit = 19.5
    com.apple.kpi.libkern = 19.5
    com.apple.kpi.mach = 19.5

For x86_64:
    1 symbol not found in any library kext:
	__ZN15OSMetaClassBase25_RESERVEDOSMetaClassBase3Ev

williambj1, 2020-07-17:
Should have been fixed a long time ago

closed 2020-07-17 by williambj1

flwmholtz, 2021-01-27:
This symbol has apparently gone away - it used to be resolved in System.kext/Plugins/LibKern.kext but it would appear Catalina's LibKern only goes from MetaClassBase4. 

Unforutunately, this is also the same symbol that the AppleRAIDCard kext is looking for in my MacPro5,1.. :p