OpenIntelWireless/IntelBluetoothFirmware #97: Kext cause kernel panic after wakeup

zjkyz8, 2020-04-20:
**Description**
> A clear and concise description of what the bug is
It seems that the kext cause kernel panic after wakeup from sleep.

**Environment**

- Kext Version: 
Kext from daliansky/XiaoMi-Pro-Hackintosh 1.3.8
- WiFi Card Model: 

- Product ID:  
Xiaomi Pro
- macOS Version: 
10.15.3
**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```

```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

```log
no log shows
```

**Kernel Panic Logs**
> In case there is a kernel panic, add `keepsyms=1` in boot-args then paste the panic info below between ```

```
panic(cpu 0 caller 0xffffff8000ca2d1c): Sleep transition timed out after 180 seconds while calling power state change callbacks. Suspected bundle: com.apple.iokit.IOUSBHostFamily. Thread 0x1130.
Failure code:: 0x00000004 00000014

Backtracing specified thread
Backtrace (CPU 0), Frame : Return Address
0xffffff81f132b900 : 0xffffff8000660178 
0xffffff81f1b7bc40 : 0xffffff800055d771 
0xffffff81f1b7bcb0 : 0xffffff800055bf6f 
0xffffff81f1b7bd00 : 0xffffff800065d2b9 
0xffffff81f1b7bd40 : 0xffffff800065cbdb 
0xffffff81f1b7bd70 : 0xffffff8000c3f578 
0xffffff81f1b7bda0 : 0xffffff8000c42489 
0xffffff81f1b7be00 : 0xffffff8000c2794b 
0xffffff81f1b7bea0 : 0xffffff8000c27344 
0xffffff81f1b7bec0 : 0xffffff800057d7e5 
0xffffff81f1b7bf40 : 0xffffff800057d311 
0xffffff81f1b7bfa0 : 0xffffff80004e213e 

BSD process name corresponding to current thread: kernel_task
Boot args: f9	g)f)e,d
g$
```

**Screenshots**
> If applicable, add screenshots to help explain your problem.
After reboot, bluetooth can not use.

[image]

Menchen, 2020-04-20:
Duplicate of #23

zjkyz8, 2020-04-21:
close it since duplication

closed 2020-04-21 by zjkyz8