OpenIntelWireless/IntelBluetoothFirmware #99: Invalid Bluetooth Core Spec version

is8EfvcgJV, 2020-04-22:
**Description**
Bluetooth Core Spec and LMP Version are both Unknown in system profiler. I would expect them to be 5.0 based off my card (I don't know if this causes any functional issue as I do not have a Bluetooth 5.0 device to test). 

I notice my firmware version is 10.256, and I have seen issues here with 9.256 where it is fine, and it says 5.0 (0x9) instead of Unknown (0xA).

For example, Issue #66 has 5.0 (0x9) and firmware 9.256. Issue #95 has the same firmware version as me, and the output there is Unknown (0xA) like mine.

**Environment**

- Kext Version: 1.0.3
- WiFi Card Model: Intel Wireless AC 9560 CNVi
- Product ID: 0x0AAA
- macOS Version: 10.15.4

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```
Apple Bluetooth Software Version: 7.0.4f6
Hardware, Features, and Settings:
    Name: iMac
    Address: REDACTED
    Bluetooth Low Energy Supported: Yes
    Handoff Supported: Yes
    Instant Hot Spot Supported: Yes
    Manufacturer: Intel
    Transport: USB
    Firmware Version: 10.256 (10.256)
    Bluetooth Power: On
    Discoverable: On
    Connectable: Yes
    Auto Seek Pointing: On
    Remote wake: On
    Vendor ID: 0x8087
    Product ID: 0x0AAA
    Bluetooth Core Spec: Unknown (0xA)
    HCI Revision: 0x100
    LMP Version: Unknown (0xA)
    LMP Subversion: 0x100
    Device Type (Major): Computer
    Device Type (Complete): Mac Desktop
    Composite Class Of Device: 0x380104
    Device Class (Major): 0x01
    Device Class (Minor): 0x01
    Service Class: 0x1C0
    Auto Seek Keyboard: On
Devices (Paired, Configured, etc.):
    Magic Keyboard 2:
        Address: REDACTED
        Major Type: Miscellaneous
        Minor Type: Unknown
        Services: Magic Keyboard with Numeric Keypad
        Paired: Yes
        Configured: Yes
        Connected: Yes
        Manufacturer: Broadcom (0x5, 0x240C)
        Battery Level: 100%
        Bluetooth Core Spec: 3.0
        Firmware Version: 0x0100
        Vendor ID: 0x004C
        Product ID: 0x026C
        Class of Device: 0x00 0x00 0x0000
        AFH: On
        AFH Map: FEFFFFFEFFFFFFFFFF3F
        RSSI: 0
        Role: Master
        Connection Mode: Active Mode
        Interval: 0 ms
        Host Connectable: Yes
        EDR Supported: Yes
        eSCO Supported: No
        SSP Supported: Yes
    Magic Mouse 2:
        Address: REDACTED
        Major Type: Miscellaneous
        Minor Type: Unknown
        Services: Magic Mouse 2
        Paired: Yes
        Configured: Yes
        Connected: Yes
        Manufacturer: Broadcom (0x5, 0x240C)
        Battery Level: 50%
        Bluetooth Core Spec: 3.0
        Firmware Version: 0x0102
        Vendor ID: 0x004C
        Product ID: 0x0269
        Class of Device: 0x00 0x00 0x0000
        AFH: On
        AFH Map: FEFFFFFEFFFFFFFFFF3F
        RSSI: 0
        Role: Master
        Connection Mode: Active Mode
        Interval: 0 ms
        Host Connectable: Yes
        EDR Supported: Yes
        eSCO Supported: No
        SSP Supported: Yes
Services:
    Bluetooth File Transfer:
        Folder other devices can browse: ~/Public
        When receiving items: Accept all without warning
        State: Disabled
    Bluetooth File Exchange:
        Folder for accepted items: ~/Downloads
        When other items are accepted: Save to location
        When receiving items: Accept all without warning
        State: Disabled
    Bluetooth Internet Sharing:
        State: Disabled
Incoming Serial Ports:
    Bluetooth-Incoming-Port:
    RFCOMM Channel: 3
    Requires Authentication: No
```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

```log
Nothing
```

**Kernel Panic Logs**
> In case there is a kernel panic, add `keepsyms=1` in boot-args then paste the panic info below between ```

```
No kernel panics
```

zxystd, 2020-04-22:
This is not a bug, just because v1.0.3 replace the newest firmware version that Linux Intel released, the display information may not affect functionality. Thanks.

is8EfvcgJV, 2020-04-22:
I see it is just a difference between 9.256 and 10.256. I have been looking at a few firmware releases. Where specifically are you getting your copies of the firmware?

My card is using ibt-17-6-1.sfi and I see there are differences in the file depending on what Linux project I extract it from.

closed 2020-05-13 by zxystd