OpenIntelWireless/IntelBluetoothFirmware #101: Won't respond and won't connect just out of no where

Parthvsquare, 2020-04-29:
**Description**
I was able to pair and scan my bluetooth devices, Today system preference started acting up. Now i am not able to scan nor connect to prephericals

**Environment**

- Kext Version: latest
- WiFi Card Model: 
- Product ID: (i have no idea where to find it)
- macOS Version: 10.15.4

**Info in System Report - Bluetooth**
Apple Bluetooth Software Version:	7.0.4f6
  Hardware, Features, and Settings:
  Name:	MacBook Pro
  Address:	14-4F-8A-7B-BB-63
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	8.256 (8.256)
  Bluetooth Power:	On
  Discoverable:	On
  Connectable:	Yes
  Auto Seek Pointing:	Off
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0A2B
  Bluetooth Core Spec:	4.2 (0x8)
  HCI Revision:	0x100
  LMP Version:	4.2 (0x8)
  LMP Subversion:	0x100
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Portable
  Composite Class Of Device:	0x38010C
  Device Class (Major):	0x01
  Device Class (Minor):	0x03
  Service Class:	0x1C0
  Auto Seek Keyboard:	Off

  Services:
  Bluetooth File Transfer:
  Folder other devices can browse:	~/Public
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth File Exchange:
  Folder for accepted items:	~/Downloads
  When other items are accepted:	Save to location
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth Internet Sharing:
  State:	Disabled
  Incoming Serial Ports:
  Bluetooth-Incoming-Port:
  RFCOMM Channel:	3
  Requires Authentication:	No
  Outgoing Serial Ports:
  ParthsiPad-WirelessiAPv2:
  Address:	00-5B-94-F2-FB-75
  RFCOMM Channel:	1
  Requires Authentication:	No

**Logs**
not able to pull any logs

[image]

**Kernel Panic Logs**
may be

**Screenshots**
[image]

Parthvsquare, 2020-04-29:
Now it says bluetooth unavailable

MiguelCatalan, 2020-05-01:
Same issue here

zxystd, 2020-05-13:
I haven't encountered such a problem. Is it always like this or occasionally?

Parthvsquare, 2020-05-13:
now its occasionally.
I have to restart and there may be a chance of working

closed 2020-07-17 by williambj1