OpenIntelWireless/IntelBluetoothFirmware #103: Bluetooth Not Showing Any Devices

jonshadow12, 2020-05-03:
**Description**
> A clear and concise description of what the bug is

Hi there - I can see the BT module, and the config in the Settings pane, but It doesn't show any devices to connect to. I have tested 3 different known good devices to connect to with none of them showing up.

**Environment**

- Kext Version: 1.0.3
- WiFi Card Model: Internal Intel via Gigabyte X570 AORUS ELITE WIFI - Vendor ID: 0x8087
 - Product ID: 0x0AA7
- macOS Version: 10.15.4

**Info in System Report - Bluetooth**
> Copy and paste all contents down below between ```

```
  Apple Bluetooth Software Version:	7.0.4f6
  Hardware, Features, and Settings:
  Name:	Jon’s Ryzentosh
  Address:	HIDDEN
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	No
  Instant Hot Spot Supported:	No
  Manufacturer:	Intel
  Transport:	USB
  Firmware Version:	8.4352 (8.4352)
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	No
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x8087
  Product ID:	0x0AA7
  Bluetooth Core Spec:	4.2 (0x8)
  HCI Revision:	0x1100
  LMP Version:	4.2 (0x8)
  LMP Subversion:	0x1100
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Desktop
  Composite Class Of Device:	0x380104
  Device Class (Major):	0x01
  Device Class (Minor):	0x01
  Service Class:	0x1C0
  Auto Seek Keyboard:	On
  Services:
  Bluetooth File Transfer:
  Folder other devices can browse:	~/Public
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth File Exchange:
  Folder for accepted items:	~/Downloads
  When other items are accepted:	Save to location
  When receiving items:	Accept all without warning
  State:	Disabled
  Bluetooth Internet Sharing:
  State:	Disabled
  Incoming Serial Ports:
  Bluetooth-Incoming-Port:
  RFCOMM Channel:	3
  Requires Authentication:	No
```

**Logs**
> Use `log show --last boot | grep IntelFir` in terminal to get logs, paste them down between ```log and ```

```log
Jons-Ryzentosh:~ jon$ log show --last boot | grep IntelFirmware
2020-05-03 11:06:32.802133-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-05-03 11:06:32.846473-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-05-03 11:06:32.850169-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0AA7
2020-05-03 11:06:32.914312-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-05-03 11:06:34.423334-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2020-05-03 11:06:34.486256-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-05-03 11:06:34.511047-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-05-03 11:06:34.535662-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-05-03 11:06:34.560240-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了只找到了！！！
2020-05-03 11:06:34.585695-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到中断端点了！！！
2020-05-03 11:06:34.614570-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到Bulk输出端点了！！！
2020-05-03 11:06:34.641852-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到Bulk输入端点了！！！
2020-05-03 11:06:34.669101-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2020-05-03 11:06:34.693901-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-05-03 11:06:34.721251-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-05-03 11:06:34.730232-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-05-03 11:06:34.732306-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-05-03 11:06:34.760229-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-05-03 11:06:34.762230-0400 0x1aa      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Intel device is already patched. patch num: 38
2020-05-03 11:06:34.766073-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-05-03 11:06:34.768152-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-05-03 11:06:34.797232-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-05-03 11:06:34.799226-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-05-03 11:06:34.801317-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-05-03 11:06:34.827449-0400 0x5a3      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
Jons-Ryzentosh:~ jon$ 
```

**Kernel Panic Logs**
> In case there is a kernel panic, add `keepsyms=1` in boot-args then paste the panic info below between ```

```

```

**Screenshots**
> If applicable, add screenshots to help explain your problem.

MrCreeper1008, 2020-05-06:
Same issue here. If I manually connect my Magic Trackpad 2 to the computer via USB then unplug it, it will stay connected via Bluetooth and it will show up on the menu. However, I just can't seem to pair my Sony headphones because it is not showing up.  

One thing to note though, if the device that you want to pair with the computer makes the request to pair first, said device **will suddenly be visible** to the computer, even if it wasn't before making the request.

fudaming, 2020-05-17:
Similar issue. kext loaded and cannot find device sometime, especially when cold boot.
device id 0x8087, 0x0029
[image]

igorkulman, 2020-05-24:
Same issue on AC7260

```bash
❯ log show --last boot | grep IntelFirmware
2020-05-24 14:08:20.849097+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver init()
2020-05-24 14:08:20.849133+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Probe()
2020-05-24 14:08:20.849163+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x07DC
2020-05-24 14:08:20.849258+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Start()
2020-05-24 14:08:22.350242+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: set configuration to 1
2020-05-24 14:08:22.350775+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-05-24 14:08:22.350780+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-05-24 14:08:22.350783+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 忙碌中，别急
2020-05-24 14:08:22.350787+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了只找到了！！！
2020-05-24 14:08:22.350996+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到中断端点了！！！
2020-05-24 14:08:22.351181+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到Bulk输出端点了！！！
2020-05-24 14:08:22.351348+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: 找到了找到了，找到Bulk输入端点了！！！
2020-05-24 14:08:22.351504+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: usb init succeed
2020-05-24 14:08:22.351506+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_RESET
2020-05-24 14:08:22.356046+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-05-24 14:08:22.364411+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-05-24 14:08:22.364416+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_VERSION
2020-05-24 14:08:22.364633+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-05-24 14:08:22.365466+0200 0x11a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Intel device is already patched. patch num: 57
2020-05-24 14:08:22.365472+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-05-24 14:08:22.365475+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: HCI_OP_INTEL_EVENT_MASK
2020-05-24 14:08:22.365645+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt wait
2020-05-24 14:08:22.366425+0200 0x11a      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: can not proceed response
2020-05-24 14:08:22.366437+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: interrupt continue
2020-05-24 14:08:22.366443+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: End download
2020-05-24 14:08:22.366474+0200 0x49d      Default     0x0                  0      0    kernel: (IntelBluetoothFirmware) IntelFirmware: Driver Stop()
```

MrCreeper1008, 2020-06-03:
@fudaming same issue here. kext seems to be loaded properly but there's no log and the computer can't see any devices. not even my previously connected trackpad

zxystd, 2020-06-04:
According to the log, the driver works fine, detect device firmware status successfully.

closed 2020-07-17 by williambj1